espressif/esp-idf #9160: Compilation failure after commit ac4c7d99fea8ab3d3819f95ddfbb1893af059367

bpietsch, 2022-06-15:
- Development Kit: ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROVER-I
- IDF version: v5.0-dev-3486-g82d23c0404
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: macOS
- Using an IDE: No
- Power Supply: USB

## Problem Description

Getting the following unexpected compile failure:

FAILED: esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/soc/esp32/highint_hdl.S.obj 
/Users/brian/Tools/esp/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/Users/brian/Projects/ESP32FactoryImage/FactoryImage/build/config -I/Users/brian/Tools/esp/esp-idf/components/esp_system/include -I/Users/brian/Tools/esp/esp-idf/components/esp_system/port/include -I/Users/brian/Tools/esp/esp-idf/components/esp_system/port/. -I/Users/brian/Tools/esp/esp-idf/components/esp_system/port/soc -I/Users/brian/Tools/esp/esp-idf/components/esp_system/port/include/private -I/Users/brian/Tools/esp/esp-idf/components/newlib/platform_include -I/Users/brian/Tools/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/brian/Tools/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/brian/Tools/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/brian/Tools/esp/esp-idf/components/freertos/esp_additions/include -I/Users/brian/Tools/esp/esp-idf/components/esp_hw_support/include -I/Users/brian/Tools/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/brian/Tools/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/brian/Tools/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/brian/Tools/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/Users/brian/Tools/esp/esp-idf/components/heap/include -I/Users/brian/Tools/esp/esp-idf/components/log/include -I/Users/brian/Tools/esp/esp-idf/components/soc/include -I/Users/brian/Tools/esp/esp-idf/components/soc/esp32/. -I/Users/brian/Tools/esp/esp-idf/components/soc/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/hal/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/hal/include -I/Users/brian/Tools/esp/esp-idf/components/hal/platform_port/include -I/Users/brian/Tools/esp/esp-idf/components/esp_rom/include -I/Users/brian/Tools/esp/esp-idf/components/esp_rom/include/esp32 -I/Users/brian/Tools/esp/esp-idf/components/esp_rom/esp32 -I/Users/brian/Tools/esp/esp-idf/components/esp_common/include -I/Users/brian/Tools/esp/esp-idf/components/xtensa/include -I/Users/brian/Tools/esp/esp-idf/components/xtensa/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/lwip/include -I/Users/brian/Tools/esp/esp-idf/components/lwip/include/apps -I/Users/brian/Tools/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/brian/Tools/esp/esp-idf/components/lwip/lwip/src/include -I/Users/brian/Tools/esp/esp-idf/components/lwip/port/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/brian/Tools/esp/esp-idf/components/vfs/include -I/Users/brian/Tools/esp/esp-idf/components/esp_wifi/include -I/Users/brian/Tools/esp/esp-idf/components/esp_event/include -I/Users/brian/Tools/esp/esp-idf/components/esp_netif/include -I/Users/brian/Tools/esp/esp-idf/components/esp_phy/include -I/Users/brian/Tools/esp/esp-idf/components/esp_phy/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/spi_flash/include -I/Users/brian/Tools/esp/esp-idf/components/esp_timer/include -I/Users/brian/Tools/esp/esp-idf/components/pthread/include -I/Users/brian/Tools/esp/esp-idf/components/bootloader_support/include -I/Users/brian/Tools/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/Users/brian/Tools/esp/esp-idf/components/efuse/include -I/Users/brian/Tools/esp/esp-idf/components/efuse/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/driver/include -I/Users/brian/Tools/esp/esp-idf/components/driver/deprecated -I/Users/brian/Tools/esp/esp-idf/components/driver/esp32/include -I/Users/brian/Tools/esp/esp-idf/components/esp_pm/include -I/Users/brian/Tools/esp/esp-idf/components/esp_ringbuf/include -I/Users/brian/Tools/esp/esp-idf/components/app_update/include -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=format= -Wno-format -Wno-enum-conversion -gdwarf-4 -ggdb -Og -Wwrite-strings -fstack-protector-all -fmacro-prefix-map=/Users/brian/Projects/ESP32FactoryImage/FactoryImage=. -fmacro-prefix-map=/Users/brian/Tools/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-3485-g20f5e180ee\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/soc/esp32/highint_hdl.S.obj -MF esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/soc/esp32/highint_hdl.S.obj.d -o esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/soc/esp32/highint_hdl.S.obj -c /Users/brian/Tools/esp/esp-idf/components/esp_system/port/soc/esp32/highint_hdl.S
/Users/brian/Tools/esp/esp-idf/components/esp_system/port/soc/esp32/highint_hdl.S: Assembler messages:
/Users/brian/Tools/esp/esp-idf/components/esp_system/port/soc/esp32/highint_hdl.S:384: Error: invalid symbolic operand

## Other items if possible

Reverting commit ac4c7d99fea8ab3d3819f95ddfbb1893af059367 resolves the issue.

KonstantinKondrashov, 2022-06-15:
Hi @bpietsch!
Sorry for inconvenience. Please try this fix:
```diff
diff --git a/components/soc/esp32/include/soc/soc_caps.h b/components/soc/esp32/include/soc/soc_caps.h
index 158fc62..a730a17 100644
--- a/components/soc/esp32/include/soc/soc_caps.h
+++ b/components/soc/esp32/include/soc/soc_caps.h
@@ -92,8 +92,8 @@
 
 #if SOC_CAPS_ECO_VER < 2
 #define SOC_DPORT_WORKAROUND                   1
-#define SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL (5U)
 #endif // SOC_CAPS_ECO_VER < 2
+#define SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL (5U)
 
 /*-------------------------- ADC CAPS ----------------------------------------*/
 /**
```
I will fix it on master ASAP.

bpietsch, 2022-06-15:
Thanks Konstantin, that did the trick!

closed 2022-06-21 by espressif-bot