espressif/esp-idf #9161: Add events notification to protocomm or a way to hook into nimble BLE events

bbinet, 2022-06-15:
**Is your feature request related to a problem? Please describe.**

My application needs to get notifications of new BLE connections in the provisioning manager (which uses protocomm with the nimble BLE transport).

**Describe the solution you'd like**

It would be great if protocomm could generate events like `PROTOCOMM_CONNECT`, `PROTOCOMM_DISCONNECT`, ...

**Describe alternatives you've considered**

I also tried to hook directly the nimble `BLE_GAP_EVENT_CONNECT`, `BLE_GAP_EVENT_DISCONNECT` events through the `ble_gap_event_listener_register` nimble api function, but it does not work: I don't know why, but my callback function is never called on nimble BLE GAP events...

Any idea why it does not work?

rahult-github, 2022-06-16:
Hi @bbinet ,

In existing nimble code, if application has invoked advertisement or scan or a connect , the callback function can be registered to receive the gap events.  For e.g. in existing framework for protocomm component, when advertising is started, an event handler ([simple_ble_gap_event](https://github.com/espressif/esp-idf/blob/master/components/protocomm/src/transports/protocomm_nimble.c#L215)) is registered, which then handles all the events.  

If you have your own standalone application, then registering callback would be needed. IDF has blecent and bleprph sample applications which you can refer and based on your use case, you can modify the application to get the callback events. 

Thanks,
 Rahul

bbinet, 2022-06-16:
Hi @rahult-github, yes I know about the `simple_ble_gap_event` callback which is registered in the `ble_gap_adv_start` call, but precisely, I want to use protocomm so I can't register my own callback to be notified of BLE GAP events, that is why I ask for a way in protocomm to notify about connection/disconnection events (maybe using the default event loop?).

Also, do you know why using `ble_gap_event_listener_register` does not work? Does nimble can only notify a single callback for BLE GAP events?

bbinet, 2022-06-17:
Hi @rahult-github

> So if we see, protocomm layer has support for multiple transports ( nimble / bluedroid / httpd ). Having a common event which has support for all transports may not be feasible. Also to think, that if in future a new transport support gets added then it can become more complex. So it makes sense to have transport specific handling done at the individual transport layer (e.g. nimble, in this case).So there is separate nimble specific code that is part of protocomm, which i was referring to in the previous comment, that has the callback registered in current IDF code.

Sure, it is fine to have multiple transports, and if there is no connect/disconnect event common to all transport, we might create specific `PROTOCOM_NIMBLE_CONNECT`/`PROTOCOM_NIMBLE_DISCONNECT` only for the related transports.

> > Also, do you know why using ble_gap_event_listener_register does not work? Does nimble can only notify a single callback for BLE GAP events?
> 
> Can you please share any application code that i can try at my end to reproduce the behaviour you mentioning. This is because, ble_gap_event_listener_register should be working in current nimble code. Please also share the branch you are referring to .

I'm using idf master branch, and here is a commit in my application which tries to receive BLE GAP events from nimble but does not work:
https://gist.github.com/bbinet/915de020dabfaa7d012d19cf2dfe6786

rahult-github, 2022-06-17:
Hi @bbinet ,

> I'm using idf master branch, and here is a commit in my application which tries to receive BLE GAP events from nimble but does not work:
https://gist.github.com/bbinet/915de020dabfaa7d012d19cf2dfe6786

Thanks. I think the code changes done is correct.  To confirm , I quickly tried your logic in one of the existing examples in IDF and i can see the listener getting registered and events getting posted . So code is ok. Attached is patch and console output for the same. 

[listener_bleprph.txt](https://github.com/espressif/esp-idf/files/8926019/listener_bleprph.txt)
[listener_console_log.txt](https://github.com/espressif/esp-idf/files/8926029/listener_console_log.txt)

Can you please confirm indeed all BLE activity is happening . I suspect if for some reason , those events are not posted, then the listener callback function may not get any events .  Debug enabled logs can show what packets are sent / received . 

Thanks,
 Rahul

bbinet, 2022-06-17:
Thank you @rahult-github, I'm now able to successfully register to BLE events using `ble_gap_event_listener_register`, there was 2 issues in my code:
- my event listener callback was registered before `ble_gap_adv_start` was called in protocomm
- my `ble_gap_event_listener` handle need to be defined as static as it generates "Instruction access fault" exception if not static

closed 2022-06-17 by bbinet