espressif/esp-idf #9164: Out of buffer error while using Advanced OTA example in esp-idf

MaazSk, 2022-06-15:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-939-g952b4d848f
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description
I am trying to use the advanced ota example provided by esp-idf but I am getting the following error

> E (9663) HTTP_CLIENT: Out of buffer
> E (9663) esp_https_ota: Failed to open HTTP connection: ESP_FAIL
> E (9663) esp_https_ota: Failed to establish HTTP connection
> E (9673) advanced_https_ota_example: ESP HTTPS OTA Begin failed

How can I change the buffer size or any help how to resolve it can be of better help

Thank you!

mahavirj, 2022-06-15:
Please try setting larger value for  `buffer_size_tx` from HTTP client configuration

 https://github.com/espressif/esp-idf/blob/20f5e180eecccfaff815d707e1fcbba2f4d6a391/components/esp_http_client/include/esp_http_client.h#L128

MaazSk, 2022-06-15:
Larger means what should it be its ideal value as my .bin file would be approximately 1.2MB in size and earlier these code used to work.

Also when I tried to set it 10240 I got the following error.

> E (26957) esp_https_ota: Client error (400)
> E (26967) esp_https_ota: Failed to establish HTTP connection
> E (26977) advanced_https_ota_example: ESP HTTPS OTA Begin failed

The .bin file is been hosted on google drive.

shubhamkulkarni97, 2022-06-15:
Hi @MaazSk,

* You should not use the `sharing link` for downloading the firmware. You can use this tool to generate the download link: https://sites.google.com/site/gdocs2direct/
* Please check if you can download the file on your host machine using curl. This will help to check any issues with the setup.

MaazSk, 2022-06-15:
Hi @shubhamkulkarni97,
Thank you for the reply. I know that I need to create a downloadable link and I have created the same and I am able to directly download the file using the url without any issue.

MaazSk, 2022-06-16:
The issue is been resolved the problem was when I created the link the access given was restricted and not to everyone which should I have been to everyone after changing this setting it is working fine,

Thank you for the help

closed 2022-06-16 by MaazSk