espressif/esp-idf #9165: ESP32S3 OpenThread BR sample fails to advertise dataset

chrkroh, 2022-06-15:
## Environment

- Module or chip used:  ESP32-S3-WROOM-1-N8R8
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3424-gbb23d783c0
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:   Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

Our intention was to setup an OpenThread BorderRouter based on an ESP32-S3-WROOM-1-N8R8 as border router and a Nordic nrf52840 as RCP. We could get the RCP/border router connection to work configured the border router according to https://openthread.io/codelabs/openthread-border-router#2 and displayed this dataset
```
Active Timestamp: 0
Channel: 15
Channel Mask: 0x07fff800
Ext PAN ID: 022f2acc1c143171
Mesh Local Prefix: fd93:7ed5:729a:c086::/64
Network Key: 7add48bae6162281f1006d2a3858b629
Network Name: OpenThread-f0bc
PAN ID: 0xf0bc
PSKc: 5df81cd18ffcd000b4af77bbda7123ee
Security Policy: 672 onrc
```
Afterwards we scanned for available border routers via the OT-CLI of an other Thread device, but the diplayed dataset was missing the network name and had the wrong pan id.

```
uart:~$ ot scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 |                  | 0000000000000000 | f0bc | f2a2a8c164a645f9 | 15 | -71 |  84 |

```
We also tested pairing devices to this setup, but it always failed. 

The RCP firmware is from the Nordic connect sdk v1.9.1 openthread coprocessor sample.

To eliminate any misbehavior of the RCP firmware, we tested the same firmware and hardware working together with an Raspi as border router. When using the raspi the scan shows up the correct network details of the raspi border router.
```
uart:~$ ot scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThreadDemoPi | 1111111122222222 | 1234 | cea816717c326ceb | 15 | -66 | 108 |
```
Apparently something seems wrong with our settings or the ESP openthread lib?! I appended our sdkconfig

### Expected Behavior
Show up border router with configured dataset. 

### Actual Behavior
The network name and pan id is not advertised correctly.
```
uart:~$ ot scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 |                  | 0000000000000000 | f0bc | f2a2a8c164a645f9 | 15 | -71 |  84 |

```

gjc13, 2022-06-16:
Hi. From the behavior looks like the border router is using a corrupt dataset. Would you post the full commands setting up the network?

At the same time we may try the config option `OPENTHREAD_BR_AUTO_START`. The BR will automatically form the network under this option. We need to provide the Wi-Fi ssid and psk in config option `CONFIG_EXAMPLE_WIFI_SSID` and  `CONFIG_EXAMPLE_WIFI_PASSWORD` as well.

I cannot find the sdkconfig file. Would you mind uploading it again?

chrkroh, 2022-06-16:
We already use OPENTHREAD_BR_AUTO_START with our wifi credentials. 
We also assumed a corrupt dataset and tried to init a new one via the ot commands, but the result remained the same.

xieqinan, 2022-06-17:
Hi. I tried to use openthread/ot-nrf528xx whose **version api  is 5**  as the RCP of esp32 obtr, which can realize the network discovery function of thread without the situation described by you. You can refer to the following links. _https://openthread.google.cn/codelabs/openthread-hardware#3_

xieqinan, 2022-06-20:
OK, the version between _esp-idf v5.0-dev-3424-gbb23d783c0_ and _Nordic connect sdk v1.9.1_  may be mismatch,  esp32-h2 will be released soon. if you want to use the OTBR in hurry, you can follow the above steps.

chrkroh, 2022-06-22:
> Hi. I tried to use openthread/ot-nrf528xx as the RCP of esp32 obtr, which can realize the network discovery function of thread without the situation described by you. You can refer to the following links.
> 
> github for ot-nrf528xx : https://github.com/openthread/ot-nrf528xx.git ot-nrf528xx version : 8c508c8b693ce660134e934c967835cb43ffcc31
> 
> esp-idf version: [089d88d](https://github.com/espressif/esp-idf/commit/089d88da7745ef13a71cf3fa9223e75a37af2034)
> 
> operation steps: https://openthread.google.cn/codelabs/openthread-hardware#3
> 
> Hope to help you！

I tried this, but get the same result. I assume there is still something wrong with my sdkconfig.

closed 2022-07-04 by chrkroh