espressif/esp-idf #9169: Assert in TWAI interrupt handler twai_handle_tx_buffer_frame()

zevv, 2022-06-16:
## Environment

chip ESP32-D0WD
idf 4.2
custom board

## Problem Description

Our application occasionaly crashes because of a failed assertion in `twai_handle_tx_buffer_frame()`. 

### Expected Behavior

Program does not crash on assertions in interrupt handlers :)

### Actual Behavior

During normal program operation the application gets aborted on an assertion with the below stack trace.

### Steps to reproduce

We are not sure yet what triggers this situation, as it is hard to reproduce. If I find a way to reproduce I can try to provide a minimal application showing the behavior, but it is likely related to other traffic on the bus.

## Debug Logs

```
0x40089484: panic_abort at /opt/toolchains/esp-idf/components/esp_system/panic.c:356
0x40089f5d: esp_system_abort at /opt/toolchains/esp-idf/components/esp_system/system_api.c:112
0x400904f9: abort at /opt/toolchains/esp-idf/components/newlib/abort.c:46
0x400832ef: lock_acquire_generic at /opt/toolchains/esp-idf/components/newlib/locks.c:138
0x400834fe: _lock_acquire_recursive at /opt/toolchains/esp-idf/components/newlib/locks.c:166
0x4010e99d: _vfiprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)
0x401069e9: fiprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48
0x401068cd: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)
0x40101351: twai_handle_tx_buffer_frame at /opt/toolchains/esp-idf/components/driver/twai.c:189
 (inlined by) twai_intr_handler_main at /opt/toolchains/esp-idf/components/driver/twai.c:232
0x400830e5: _xt_lowint1 at /opt/toolchains/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x400f67ca: cpu_ll_waiti at /opt/toolchains/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /opt/toolchains/esp-idf/components/esp_pm/pm_impl.c:827
0x400d9df6: esp_vApplicationIdleHook at /opt/toolchains/esp-idf/components/esp_common/src/freertos_hooks.c:63
0x4008b803: prvIdleTask at /opt/toolchains/esp-idf/components/freertos/tasks.c:3839 (discriminator 1)
0x4008d6b2: vPortTaskWrapper at /opt/toolchains/esp-idf/components/freertos/port/xtensa/port.c:168

```

zevv, 2022-06-16:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8920895/sdkconfig.txt)

boborjan2, 2022-11-07:
It happens at our setup as well. Assert in twai.c line 189. It happens during busoff quite often.

JensOgorek, 2026-10-29:
### Root cause of this assert in normal operation: the TX-interrupt-lost workaround counts a frame twice (still in master)

We hit this assert on ESP32 robots about once a day, with v5.3.1 and with v5.3.6, without any bus-off. The TX-done check of the errata workaround can report the same frame twice. With one frame in flight, `tx_msg_count` then drops to −1. The check is unchanged on master.

The HAL decides "TX done" with

```c
if ((interrupts & TWAI_LL_INTR_TI || state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED) && status & TWAI_LL_STATUS_TBS)
```

(`components/hal/twai_hal_iram.c:65` in v5.3.6, `components/esp_hal_twai/twai_hal_v1.c:258` on master). It reads the interrupt register, which clears on read, before the status register. If a frame completes between these two reads, TBS is already set, so the workaround reports TX done and clears the OCCUPIED flag. TI is set right after and makes the ISR run again. Because the TI branch does not require OCCUPIED, it reports TX done a second time. In the legacy driver this trips `assert(p_twai_obj->tx_msg_count >= 0)`; the new driver would trip `assert(... p_curr_tx[tx_idx])` in `esp_twai_onchip.c` instead.

The window is a few APB cycles. It is hit when the ISR runs at an arbitrary moment relative to the bus, e.g. after the Bluetooth controller had interrupts masked: both of our core dumps show the TWAI ISR entered from `hli_intr_restore` in `btController`. Turning `CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST` off is not a way out, because then a really lost TI blocks TX for good, and on master the option no longer exists.

A possible fix is to report TX done only while a frame is marked as occupying the buffer:

```c
if ((state_flags & TWAI_HAL_STATE_FLAG_TX_BUFF_OCCUPIED) && (status & TWAI_LL_STATUS_TBS))
```

Every transmit sets the flag, a lost TI is still recovered through TBS at the next interrupt, and a late TI for a frame that was already counted is ignored.

<details>
<summary>Reproduction (ESP32-D0WD-V3, v5.3.6, no transceiver)</summary>

TWAI in NO_ACK mode with TX and RX looped back on one GPIO, 500 kbit/s, one frame in flight at a time.

- The HAL's two register reads in a polling loop, 200 000 frames: 4.8 % of the frames have TBS set while TI only shows up in the next read, i.e. they are counted twice; 0.045 % of the TIs are really lost.
- `twai_hal_get_events()` called directly, 200 000 frames: 8 097 extra TX events. With the condition above: none, and every frame counted once.
- Legacy driver with its ISR: no assert as long as the ISR only runs for our own frames. To let it run at random moments like a delayed ISR, a GPIO interrupt was routed to the TWAI's CPU interrupt and triggered while frames were on the bus. With that, the assert came after 1 to 5 s in 5 of 5 runs. With the condition above, there was no assert and no stall in 15 minutes and 3.3 million frames. With `CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=n` there was no assert, but TX blocked for good after 1 603 frames.

</details>

[※](https://zauberzeug.github.io/colophon/#m=claude-opus-5-5&t=Analysis%2C%20bench%20reproduction%20and%20wording%20by%20the%20model%3B%20the%20decision%20to%20report%20it%20upstream%20by%20the%20author.&a=claude-code "claude-opus-5-5: Analysis, bench reproduction and wording by the model; the decision to report it upstream by the author.")