espressif/esp-idf #9170: Build error with latest docker image

EfanovIvan, 2022-06-16:
docker image
espressif/idf:latest
Digest:sha256:08eb0fb1cf3f9b6560ce73d2a742845d12589c681656f442fc67c3453db43301

```
/opt/esp/idf/components/esp_rom/include/esp32c3/rom/rtc.h: In function 'void rtc_suppress_rom_log()':
/opt/esp/idf/components/soc/esp32c3/include/soc/soc.h:51:40: error: compound assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
   51 |             (*(volatile uint32_t*)(_r) |= (_b));                                                                       \
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/opt/esp/idf/components/esp_rom/include/esp32c3/rom/rtc.h:209:5: note: in expansion of macro 'REG_SET_BIT'
  209 |     REG_SET_BIT(RTC_CNTL_STORE4_REG, RTC_DISABLE_ROM_LOG);
      |     ^~~~~~~~~~~
/opt/esp/idf/components/soc/esp32c3/include/soc/soc.h:51:40: error: using value of simple assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
   51 |             (*(volatile uint32_t*)(_r) |= (_b));                                                                       \
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/esp/idf/components/esp_rom/include/esp32c3/rom/rtc.h:209:5: note: in expansion of macro 'REG_SET_BIT'
  209 |     REG_SET_BIT(RTC_CNTL_STORE4_REG, RTC_DISABLE_ROM_LOG);
      |     ^~~~~~~~~~~
cc1plus: all warnings being treated as errors

```

igrr, 2022-06-16:
Could you please attach your sdkconfig or sdkconfig.defaults file?

EfanovIvan, 2022-06-17:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8923905/sdkconfig.txt)

igrr, 2022-06-17:
Hi @EfanovIvan, I'm sorry but I wasn't able to reproduce the issue so far. I've tried building several IDF examples with the sdkconfig file you have provided, both in the Docker container and locally. Do you observe the same error on your side when building any of the IDF examples, or only when building your application? Is your project or component CMakeLists file adding some additional compiler flags? Which compilation unit (source file) does this error happen in — is it a C or a C++ source file?

EfanovIvan, 2022-06-17:
Hi @igrr Yes, I build my c++ applications. I have additional flags set. I am compiling with -std=c++20
The entire build log is too huge, I post the part with the compiler version.

```
target_compile_options(${COMPONENT_LIB} PRIVATE -Wall -Wextra -Werror)
target_compile_features(${COMPONENT_LIB} PRIVATE cxx_std_20)
```
```
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/riscv32-esp-elf/esp-2022r1-RC1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/esp/tools/riscv32-esp-elf/esp-2022r1-RC1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/esp/tools/riscv32-esp-elf/esp-2022r1-RC1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
```

EfanovIvan, 2022-06-17:
It's ota
```
In file included from /opt/esp/idf/components/soc/esp32c3/include/soc/dport_access.h:11,
                 from /opt/esp/idf/components/hal/esp32c3/include/hal/cpu_ll.h:11,
                 from /opt/esp/idf/components/hal/include/hal/cpu_hal.h:16,
                 from /opt/esp/idf/components/esp_hw_support/include/spinlock.h:11,
                 from /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h:45,
                 from /opt/esp/idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /opt/esp/idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /opt/esp/idf/components/esp_http_client/include/esp_http_client.h:10,
                 from /builds/internal-projects/a138-ble-gateway/firmware/esp32/components/aws/aws-ota.cpp:5:
/opt/esp/idf/components/esp_rom/include/esp32c3/rom/rtc.h: In function 'void rtc_suppress_rom_log()':
/opt/esp/idf/components/soc/esp32c3/include/soc/soc.h:51:40: error: compound assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
   51 |             (*(volatile uint32_t*)(_r) |= (_b));                                                                       \
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/opt/esp/idf/components/esp_rom/include/esp32c3/rom/rtc.h:209:5: note: in expansion of macro 'REG_SET_BIT'
  209 |     REG_SET_BIT(RTC_CNTL_STORE4_REG, RTC_DISABLE_ROM_LOG);
      |     ^~~~~~~~~~~
/opt/esp/idf/components/soc/esp32c3/include/soc/soc.h:51:40: error: using value of simple assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
   51 |             (*(volatile uint32_t*)(_r) |= (_b));                                                                       \
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/esp/idf/components/esp_rom/include/esp32c3/rom/rtc.h:209:5: note: in expansion of macro 'REG_SET_BIT'
  209 |     REG_SET_BIT(RTC_CNTL_STORE4_REG, RTC_DISABLE_ROM_LOG);
      |     ^~~~~~~~~~~
```

igrr, 2022-06-17:
@EfanovIvan Thank you, the issue is clear now, we will update REG_SET_BIT and other macros to be compatible with C++20.

igrr, 2022-06-17:
Until the fix is merged, you can use `target_compile_options(${COMPONENT_LIB} PRIVATE -Wall -Wextra -Wno-volatile -Werror)` as a workaround

closed 2022-07-07 by espressif-bot