espressif/esp-idf #9174: Memory leak when saving values in NVS

kotyara12, 2022-06-17:
Good time.
I accidentally found a memory leak while saving values in NVS. My devices every night at 00:00 "lose" a little bit of free memory in the heap. Little by little, but "it's a shame." At this time, the statistics from the sensors are saved in NVS (extremum values for different periods). The very first suspicion was on NVS. I checked my code. I turned on memory debugging to the maximum depth (10) and my guess was confirmed:
```
c:\PlatformIO\village_pc\ota>c:\Users\kotyara12\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\PlatformIO\village_pc\ota\firmware\firmware_1.elf 0x40143550 0x400fad89 0x400fb526 0x400fa330 0x400fa874 0x400f972d 0x401442e4 0x400d78f4 0x400d79db 0x400dd712

0x40143550: operator new(unsigned int, std::nothrow_t const&) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_opnt.cc:43

0x400fad89: nvs::HashList::insert(nvs::Item const&, unsigned int) at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_item_hash_list.cpp:57

0x400fb526: nvs::Page::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char) at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_page.cpp:227

0x400fa330: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_storage.cpp:355
 (inlined by) nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_storage.cpp:272

0x400fa874: nvs::NVSHandleSimple::set_typed_item(nvs::ItemType, char const*, void const*, unsigned int) at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_handle_simple.cpp:30

0x400f972d: int nvs::NVSHandle::set_item<unsigned long long>(char const*, unsigned long long) at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\include/nvs_handle.hpp:270
 (inlined by) nvs_set<long long unsigned int> at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_api.cpp:344
 (inlined by) nvs_set_u64 at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\nvs_flash\src/nvs_api.cpp:384

0x401442e4: nvs_set_time at C:/PlatformIO/libs/system/reNvs/src/reNvs.cpp:66

0x400d78f4: rSensorItem::nvsStoreExtremums(char const*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:1021

0x400d79db: rSensorX2::nvsStoreExtremums(char const*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:2155
 (inlined by) rSensorX2::nvsStoreExtremums(char const*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:2142

0x400dd712: sensorsStoreData() at c:\PlatformIO\village_pc/lib/sensors/sensors.cpp:397
```

kotyara12, 2022-06-17:
My suspicion falls on nvs::HashList::insert.
This block of memory has been hanging for more than 24 hours.

negativekelvin, 2022-06-18:
I can see why you may think nvs flash should not consume ram because it is storing data in flash but it keeps an index in ram and will need to add blocks to that index as the number of stored items grows.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#item-hash-list

kotyara12, 2022-06-18:
Yes, but the number of stored elements DOES NOT GROW!!! The data is replaced by others, but the amount actually stored does not grow. If you store indexes for all obsolete data, then this is a big nonsense.

negativekelvin, 2022-06-18:
The index for a page is only deleted when the whole page is empty so yes you could end up with a lot of sparse pages with indices and inefficient use of memory.

igrr, 2022-06-18:
@kotyara12 could you please check, does the memory get released when nvs_flash_deinit function is called? If yes, then @negativekelvin's comment most likely explains the behavior you are observing. When a particular page becomes empty, the page index will be freed up. There is an upper bound on how much RAM can be allocated for the index, per one flash sector. For this reason, we don't consider this to be a leak.

By the way, could you please mention the IDF version (commit ID or tag) which you are using? This information is requested in the issue template. Thanks!

chipweinberger, 2022-06-20:
@igrr just my 2 cents, but as long as the ram is bounded, and it's reasonably small, IMO its preferable to allocate it up front.

It's unlikely that I'd *want* my program to take advantage of they few extra KB nvs is saving by not allocating up front. IMO It's better to know my embedded program wont crash, and that I know I have all the ram I need up front. Slow growing ram is particularly hard to account for and test.

kotyara12, 2022-06-20:
> By the way, could you please mention the IDF version

 framework-espidf @ 3.40302.0 (4.3.2)

kotyara12, 2022-06-20:
> IMO its preferable to allocate it up front

As far as I understand, ESP-IDF is almost entirely heap oriented. An exception is the allocation of static buffers for tasks and queues. Everything else "inside" the ESP-IDF is allocated on the heap. So don't get away with it. Yes, and I do not see much point, in general.

On my devices, the heap is used very, very actively, especially when huge JSONs are collected piece by piece to be sent to external services. And I tested static buffer allocation for data and dynamic allocation. Static allocation gave MUCH WORSE results, alas. I gave it up. The exception is all tasks and queues that I can statically allocate. All data is on the heap.

With proper use of the heap, my devices run for months without failure, and I use the heap very, very heavily.

That is why I drew attention to such a small leak, because in a year it can already give significant problems.

chipweinberger, 2022-06-20:
AFAICT, ESP-IDF does most of its biggest allocations when you call `init` for some particular feature (with heap), which is IMO the preferred approach. But of course there are subtleties.

kotyara12, 2022-06-22:
Good afternoon. No, I don't have a "test" device yet. All available devices work in "combat mode" and I don't want to break their work once again. I monitor the amount of free memory remotely, with graphs that update every few minutes.

kotyara12, 2022-08-23:
Hello, @0xjakob

Apparently, everything is so. Because on August 22, the free memory on one of the devices increased instead of decreasing. As I understand from your explanations, the page is full and the index for it has been deleted...
[image]

closed 2022-08-23 by kotyara12

0xjakob, 2022-08-23:
@kotyara12 Thanks for the update. Still, a further improvement could be to initialization NVS only if you need it and deinitialize it afterwards, so you only have the additional memory consumption caused by NVS for that moment.

KaeLL, 2022-08-23:
@chipweinberger is right and imo this warrants mention in the docs, be it under NVS section or perf RAM Usage's.