espressif/esp-idf #9175: SDSPI API broken when used in C++ project and won't compile

sukesh-ak, 2022-06-18:
## Environment

- Development Kit:      [Unexpected Maker FeatherS3 (ESP32-S3)]
- Module or chip used:  [ESP32-S3]
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:   CMake / idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes] but no difference from command line as well
- Power Supply:         [USB|external 5V]

## Problem Description
SDSPI API broken when used in C++ project and won't compile with compiler errors.

### Expected Behavior
Compile without errors

### Actual Behavior
Compiler errors as below. Works well in `C` without any code/API changes.

```C++
../main/sd_card_example_main.cpp: In function 'void app_main()':
../main/sd_card_example_main.cpp:97:5: warning: missing initializer for member 'spi_bus_config_t::data4_io_num' [-Wmissing-field-initializers]
     };
     ^
../main/sd_card_example_main.cpp:97:5: warning: missing initializer for member 'spi_bus_config_t::data5_io_num' [-Wmissing-field-initializers]
../main/sd_card_example_main.cpp:97:5: warning: missing initializer for member 'spi_bus_config_t::data6_io_num' [-Wmissing-field-initializers]
../main/sd_card_example_main.cpp:97:5: warning: missing initializer for member 'spi_bus_config_t::data7_io_num' [-Wmissing-field-initializers]
../main/sd_card_example_main.cpp:97:5: warning: missing initializer for member 'spi_bus_config_t::flags' [-Wmissing-field-initializers]
../main/sd_card_example_main.cpp:97:5: warning: missing initializer for member 'spi_bus_config_t::intr_flags' [-Wmissing-field-initializers]
../main/sd_card_example_main.cpp:100:35: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive]
     ret = spi_bus_initialize(host.slot, &bus_cfg, SPI_DMA_CH_AUTO);
                              ~~~~~^~~~
In file included from D:/esp/esp-idf/components/driver/include/driver/spi_master.h:12,
                 from D:/esp/esp-idf/components/driver/include/driver/sdspi_host.h:14,
                 from D:/esp/esp-idf/components/fatfs/vfs/esp_vfs_fat.h:20,
                 from ../main/sd_card_example_main.cpp:14:
D:/esp/esp-idf/components/driver/include/driver/spi_common.h:156:48: note:   initializing argument 1 of 'esp_err_t spi_bus_initialize(spi_host_device_t, const spi_bus_config_t*, spi_dma_chan_t)'
 esp_err_t spi_bus_initialize(spi_host_device_t host_id, const spi_bus_config_t *bus_config, spi_dma_chan_t dma_chan);
                              ~~~~~~~~~~~~~~~~~~^~~~~~~
../main/sd_card_example_main.cpp:47:26: error: invalid conversion from 'int' to 'gpio_num_t' [-fpermissive]
     #define PIN_NUM_CS   33//5
                          ^~
../main/sd_card_example_main.cpp:109:27: note: in expansion of macro 'PIN_NUM_CS'
     slot_config.gpio_cs = PIN_NUM_CS;
                           ^~~~~~~~~~
../main/sd_card_example_main.cpp:110:32: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive]
     slot_config.host_id = host.slot;
                           ~~~~~^~~~
../main/sd_card_example_main.cpp:186:23: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive]
     spi_bus_free(host.slot);
                  ~~~~~^~~~
In file included from D:/esp/esp-idf/components/driver/include/driver/spi_master.h:12,
                 from D:/esp/esp-idf/components/driver/include/driver/sdspi_host.h:14,
                 from D:/esp/esp-idf/components/fatfs/vfs/esp_vfs_fat.h:20,
                 from ../main/sd_card_example_main.cpp:14:
D:/esp/esp-idf/components/driver/include/driver/spi_common.h:169:42: note:   initializing argument 1 of 'esp_err_t spi_bus_free(spi_host_device_t)'
 esp_err_t spi_bus_free(spi_host_device_t host_id);
                        ~~~~~~~~~~~~~~~~~~^~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
### Steps to reproduce

1. Copy the sample `https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdspi`
2. rename `sd_card_example_main.c` to `sd_card_example_main.cpp` 
3. change `CMakeLists.txt` for the file name from .c to .cpp
4. add `extern "C" { void app_main(void); }`
5. Set the target `idf.py set-target esp32s3` 
6. Compile `idf.py menuconfig` and `idf.py build`

### Code to reproduce this issue
No code changes required from the sample, other than the above steps for the compiler errors. 
Sample works well if its just `C`.

0xjakob, 2022-06-20:
@sukesh-ak Unfortunately C++ isn't a strict superset of C. Hence, you may encounter difficulties when compiling the same code with C++. There is a [good FAQ page](https://isocpp.org/wiki/faq/mixing-c-and-cpp) for that topic. The examples weren't written to work with C++ right away. We'll work on this in the future but for the moment I hope you can understand that we can't change the entire set of examples in IDF. The main point is usually the initialization which is incompatible in C-style and that's also what you mainly see in your compiler output above. This become tricky to fix for IDF because some established initialization macros rely on that C-style mechanism.

sukesh-ak, 2022-06-20:
@0xjakob Thanks for the response.

Just to add some context. I have a larger project (a future product) and lot of code written in C++.
I added support for SD SPI for sd card support and started getting compile errors. In order to isolate the issue, I went back to a simple sample and found that it's broken due to type changes in the implementation rather than the sample code.

So I need a solution 💯 

The errors are not pointing to C/C++ but new type change introduced for example from `int` to `spi_host_device_t` and from `int` to `gpio_num_t` for example.

```C++
../main/sd_card_example_main.cpp:100:35: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive]
     ret = spi_bus_initialize(host.slot, &bus_cfg, SPI_DMA_CH_AUTO);
                              ~~~~~^~~~
```

```C++
../main/sd_card_example_main.cpp:47:26: error: invalid conversion from 'int' to 'gpio_num_t' [-fpermissive]
     #define PIN_NUM_CS   33
                          ^~
```

```C++
../main/sd_card_example_main.cpp:110:32: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive]
     slot_config.host_id = host.slot;
                           ~~~~~^~~~
../main/sd_card_example_main.cpp:186:23: error: invalid conversion from 'int' to 'spi_host_device_t' [-fpermissive]
     spi_bus_free(host.slot);
                  ~~~~~^~~~
```

Looks like a known issue from these experimental private routines (from you)

https://github.com/espressif/esp-idf/blob/5f0c8f57c66538121f6225350b34e24b0b1f4ff4/examples/cxx/experimental/experimental_cpp_component/private_include/spi_host_private_cxx.hpp#L29

sukesh-ak, 2022-06-20:
Work around for now is to change `int` to `gpio_num_t` for PIN definitions.

```c++
    #define PIN_NUM_MISO GPIO_NUM_37 
    #define PIN_NUM_MOSI GPIO_NUM_35
    #define PIN_NUM_CLK  GPIO_NUM_36
    #define PIN_NUM_CS   GPIO_NUM_33
```

And change `host.slot` to `SPI2_HOST` for fixing the type change.

Just to make sure I am clear, these are issues due to type changes rather than to do with just C/C++ differences. 

**These are breaking changes!** since its common to use `#define` for PIN definitions in ESP-IDF.

With the above changes the exact same code compiles and works as expected.

0xjakob, 2022-06-27:
@sukesh-ak The type changes are certainly not a mistake. They are to avoid mistakes when confusing arguments. User are advised to use these types instead of raw integers. My private routine is there because to convert from the C++ type (which is even safer) to the C type.

If the change happened not during a minor version upgrade, then I agree, it is a breaking change. But even then, I don't think we would change it back afterwards. 

Please use the dedicated types instead of the standard data types. The dedicated types make the software safer. If you have to convert types, you can use a cast (`static_cast<>()` in C++).

sukesh-ak, 2022-07-06:
> @sukesh-ak The type changes are certainly not a mistake. They are to avoid mistakes when confusing arguments. User are advised to use these types instead of raw integers. My private routine is there [...]

I understand type change from `int` to `gpio_num_t` is a positive change.
Just that, this change started throwing errors because sdkconfig values were expected to be `int`.
Casting can be easy in this case using `int pin = (gpio_num_t)33;`

Overall issue is sorted now, so we can close this one.

closed 2022-07-06 by sukesh-ak

0xjakob, 2022-07-07:
> Just that, this change started throwing errors because sdkconfig values were expected to be int.

Oh, now I understand this! Yes, it's inconvenient. I encountered this problem myself before. But I think as long as we have sdkconfig, we will continue having this problem in C++. Sorry.

sukesh-ak, 2022-07-08:
> > Just that, this change started throwing errors because sdkconfig values were expected to be int.
> 
> Oh, now I understand this! Yes, it's inconvenient. I encountered this problem myself before. But I think as long as we have sdkconfig, we will continue having this problem in C++. Sorry.

Not sure what you mean. Type changes and its impact through sdkconfig (throwing compiler errors) has nothing to do with C++

I am no newbie and been using C/C++ for about 34yrs. New to electronics for sure.

Done with replies 😊