espressif/esp-idf #9176: When closing an open usb-cdc connection, the timer svc stops executing some periodic functions, but calls others

moefear85, 2022-06-18:
## Environment

- Development Kit:   Ai-Thinker module  esp32-s2
- IDF version (run ``git describe --tags`` to find it): master branch, cloned on 27-05-2022
- Build System:   idf.py
- Operating System:  Ubuntu 22.04
- Using an IDE?: VSCode
- Power Supply:  external 3.3V

## Problem Description

I have 5 functions periodically executed by the timer service, 3 of them about every 1-3 seconds, the other 2 every 10ms and 40ms, but they only start running once the usb-cdc port is opened on the pc side.

Everything works right before and during the pc connection. But once I close the cdc port from the pc side, the two fast functions continue getting called by the timer, while the slower 3 stop getting called. I set up log statements at the very start and very end of each function to determine the moments each enters and exists. I am sure no function is hanging mid-way. This isn't even possible, as it would have to stall the entire timer service to my knowledge, causing the whole timer thread to hang. But the timer thread doesn't hang, as evident by the continued calling of the fast functions. Yet the other 3 are not entered again at all until the cdc port is opened again.

I tried twiddling the tinyusb and timer priorities, making each higher than the other. this has no effect on the problematic behavior. The only way this scenario is even possible, to my knowledge, is if the timer service explicitly decides to skip executing some functions. I'm not aware it can do this, nor why it would do this for the slow functions. Either way, I don't see why the choice would depend on whether usb-cdc is open or not. All functions, expecially the fast executing ones, do read and/or write operations to usb-cdc, so I don't see why some would behave differently than others.

0xjakob, 2022-06-20:
@moefear85 Just to clarify: with "timer svc" or "timer service", you mean the [`esp_timer` component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html) and not the [General purpose timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gptimer.html), correct?

igrr, 2022-06-20:
Or the FreeRTOS timers? (https://github.com/espressif/esp-idf/blob/8b902739ac6ef46c0f5c6f25491aa8ed79a4e898/components/freertos/FreeRTOS-Kernel-SMP/timers.c#L63)

moefear85, 2022-06-20:
hi, i mean the freertos timer task

tore-espressif, 2025-04-14:
Is there any update about this issue? We could not reproduce the problem based on the information provided

closed 2025-04-17 by moefear85