espressif/esp-idf #9177: WiFi logging strings are linked in final binary regardless of log level

boarchuz, 2022-06-18:
## Environment

- Module or chip used:  ESP32
- IDF version: v5.0-dev-3481-g16a4ee7c36
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- IDE: VS Code + IDF extension

## Problem Description

Approximately **26kB** of apparent debug strings from WiFi libs may be linked regardless of configured maximum log level, bloating binary size for no benefit.

### Expected Behavior
Reduce binary size as much as reasonably possible by discarding unused WiFi logging symbols when configuration options allow.

### Actual Behavior
Many `rodata_wlog_[verbose/debug/info/warning/error]` entries from WiFi libraries (libnet80211.a, etc) are not discarded, according to the map file, even with logging level set to 'None'. A simple WiFi station connection includes approximately 20kB of this data. When I tell the linker to discard them, functionality is unaffected; they seem unnecessary.

### Steps to reproduce

Build examples/wifi/getting_started/station with CONFIG_LOG_DEFAULT_LEVEL_NONE=y, and note all the `rodata_wlog_[verbose/debug/info/warning/error]`s in map file

MaxwellAlan, 2022-06-20:
Hi @boarchuz 
Thanks for your feature request, we will evaluate this.

zhangyanjiaoesp, 2022-07-20:
@boarchuz 
We have checked the log, although there are `rodata_wlog_[verbose/debug/info/warning/error`s in the map file, the final compiled bin file does not have. So the `CONFIG_LOG_DEFAULT_LEVEL_NONE=y` does works.

boarchuz, 2022-07-20:
Somebody replied that this isn't the case but it looks like they've deleted their comment.

I had another look and they're right. The ".flash.rodata_noload" where these symbols end up means they are not in fact included in the binary, however unfortunately idf_size.py *does* include this section, resulting in incorrect rodata and binary sizes which led me astray.
Sorry for the inconvenience.

igrr, 2022-07-20:
Let's keep this open until the issue is fixed in idf_size.py.

AxelLin, 2022-09-21:
> Let's keep this open until the issue is fixed in idf_size.py.

So how is the status? Is anyone working on the fix for idf_size.py?
@zhangyanjiaoesp

AxelLin, 2022-12-17:
> Let's keep this open until the issue is fixed in idf_size.py.

So how is the status now? Is this fixed?

igrr, 2022-12-17:
We haven't started working on a fix for this issue yet. It is still in the backlog.

closed 2023-08-15 by espressif-bot