espressif/esp-idf #9184: 'GPIO' undeclared using esp-idf ver5

nopnop2002, 2022-06-19:
## Environment
```
$ idf.py --version
ESP-IDF v5.0-dev-3202-ga2d5041492-dirty
```

## Problem Description
'GPIO' undeclared using esp-idf ver5

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "driver/gpio.h"
#include "esp_log.h"

#define GPIO_PIN 2

#define TAG "MAIN"

void app_main()
{
  gpio_reset_pin( GPIO_PIN );
  gpio_set_direction( GPIO_PIN, GPIO_MODE_OUTPUT );
  gpio_set_level( GPIO_PIN, 0 );

  gpio_set_level( GPIO_PIN, 1 );
  vTaskDelay(100);
  gpio_set_level( GPIO_PIN, 0 );
  vTaskDelay(100);

  GPIO.out_w1ts = (1 << GPIO_PIN);
  vTaskDelay(100);
  GPIO.out_w1tc = (1 << GPIO_PIN);
  vTaskDelay(100);
}
```

## Debug Logs

```
../main/main.c: In function 'app_main':
../main/main.c:27:2: error: 'GPIO' undeclared (first use in this function)
  GPIO.out_w1ts = (1 << GPIO_PIN);
  ^~~~
../main/main.c:27:2: note: each undeclared identifier is reported only once for each function it appears in
[818/825] Building C object esp-idf/wi...ifi_provisioning.dir/src/manager.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

moefear85, 2022-06-19:
I don't think there is any official GPIO struct or class exposed as part of the interface in esp-idf. So even if such a structure existed in previous versions, this is subject to change across major version changes

nopnop2002, 2022-06-19:
@moefear85 

It's a shame that you can't use the GPIO class because you can do IO about 5 times faster.

ESP-Marius, 2022-06-20:
Hi,

As @moefear85 mentions, the GPIO struct was never intended to be a part of the public interface of `gpio.h`, but previously it might have been exposed indirectly through some of the header includes. 

If you want to use it you can still include `hal/gpio_ll.h`, which exposes this struct, as well as functions for accessing it.

Note that the HAL layer is not part of the stable API, so updates might contain breaking changes, even between non-major version updates of IDF.

nopnop2002, 2022-06-20:
Thank you for your quick answer.
It worked with Ver5.

closed 2022-06-20 by nopnop2002