espressif/esp-idf #9186: Deep Sleep failed to work completely on ESP-S3-EYE

qqice233, 2022-06-19:
## Environment

- Development Kit:      ESP-S3-EYE
- IDF version (run ``git describe --tags`` to find it):
    v4.4.1
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply:         USB|Battery

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
The ESP32 core should be shut down. All the peripherals should be powered off, and the current meter should show a very low number.
### Actual Behavior
The ESP32 core WAS shut down, but the Green LED and LCD remains powered on, and the current meter shows 40mA.

// If possible, attach a picture of your setup/wiring here.
[image]

### Code to reproduce this issue

```cpp
/* Deep sleep wake up example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <time.h>
#include <sys/time.h>
#include "sdkconfig.h"
#include "soc/soc_caps.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_sleep.h"
#include "esp_log.h"
#include "driver/adc.h"
#include "driver/rtc_io.h"
#include "soc/rtc.h"

#if CONFIG_IDF_TARGET_ESP32
#include "esp32/ulp.h"
#endif

#if SOC_TOUCH_SENSOR_NUM > 0
#include "soc/sens_periph.h"
#include "driver/touch_pad.h"
#endif

#ifdef CONFIG_IDF_TARGET_ESP32C3
#define DEFAULT_WAKEUP_PIN      CONFIG_EXAMPLE_GPIO_WAKEUP_PIN
#ifdef CONFIG_EXAMPLE_GPIO_WAKEUP_HIGH_LEVEL
#define DEFAULT_WAKEUP_LEVEL    ESP_GPIO_WAKEUP_GPIO_HIGH
#else
#define DEFAULT_WAKEUP_LEVEL    ESP_GPIO_WAKEUP_GPIO_LOW
#endif
#endif

[... 435 lines cut ...]
Entering deep sleep
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.......................................
I (226) sleep: Configure to isolate all GPIO pins in sleeEnabling timer wakeup, 20s
Entering deep sleep
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.......................................
I (226) sleep: Configure to isolate all GPIO pins in sleeEnabling timer wakeup, 20s
Entering deep sleep
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.......................................
Enabling timer wakeup, 20s
Entering deep sleep
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.......................................
I (228) sleep: Configure to isolate all GPIO pins in sleeEnabling timer wakeup, 20s
Entering deep sleep
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.......................................
Enabling timer wakeup, 20s
Entering deep sleep
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect....
```
I wonder if there is a method to shut them down completely? Otherwise the battery will be drilled out quickly.

moefear85, 2022-06-19:
if you are using a module/board, then any components besides the esp chip itself, don't necessarily power off. so the power consumption you are measuring is those other components (such as the leds). you have to create your own board, or modify the existing one, adding whatever electronic components necessary to switch them off. But usually there are easy ways. For the leds, you need to explicitly set the gpios connected to them to the proper level that would turn them off. for other ICs, they usually have an ENable pin that you can connect to a gpio to disable/ENable them, thus also putting them to sleep.
Either way, I think the module has a power-indicating led that is always on when there is power, and that will always consume 10 to 20mA no matter what, unless you modify the pcb and add a transistor to control them, whose gate you would attach then to a gpio. there are alot of tutorials on the internet.

qqice233, 2022-06-20:
> if you are using a module/board, then any components besides the esp chip itself, don't necessarily power off. so the power consumption you are measuring is those other components (such as the [...]
Sure, I did try to set gpio_3 (according to the PCB layout, which is controlling the Green LED) to 0 before esp_deep_sleep_start(), but the LED turns on after the esp_deep_sleep_start().
> Either way, I think the module has a power-indicating led that is always on when there is power.
No, the Green LED is controlable.

igrr, 2022-06-20:
On ESP32-S3-EYE board, to keep the green LED off, GPIO3 must be actively driven low. However in this case, the current will flow through R80 and R83 (3.3V/2kOhm = 1.6mA).
I'm afraid that aside from desoldering either the D10 LED or R80, there's no way to prevent the LED circuit from drawing the current during deep sleep.

Here is the link to the board schematic, for reference: https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-EYE-MB_20211201_V2.2.pdf.

You may open an issue at https://github.com/espressif/esp-who/issues with a suggestion, the team will consider an enhancement to this board in the future.

I'm going to close this issue as it is not something that is related to the ESP-IDF project.