espressif/esp-idf #9187: [ESP32-S3][SD Card] sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109

chipweinberger, 2022-06-19:
**Esp-Idf:** v4.4.1
**ESP32-S3**

During SD card use, I occasionally see this error in the log:

**E (925049) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109**

This is a showstopper. All subsequent SD card operations fail, and my program usually crashes or gets hung somewhere (still need to debug this further).

**What can I do about this? Is this recoverable? Is it a sign of some other issue?**

**Details:** 
- error does not occur often.
- happens more often when quickly reading from SD card, using many calls to fread()
- subsequent fflush & fsync calls return errors  (rv != 0)
- the task responsible for SD writing usually stops working (haven't debugged this further)
- at least once, a stack overflow happened in my main task shortly after this error occured
- I have 10k pullups on my SD card lines

Schematic, for reference:
[image]

igrr, 2022-06-20:
Hi @chipweinberger, this seems more like a hardware design problem (0x109 - data CRC error, brownout,..). The voltage regulator on the devkit might not offer sufficient peak current or bulk capacitance to handle multiple external peripherals in addition to the S3 itself.

However, there is also one outstanding issue with SDMMC on S3 which may also result in a CRC error. Could you please check if either of the patches suggested in https://github.com/espressif/esp-idf/issues/8257#issuecomment-1117738699 or https://github.com/espressif/esp-idf/issues/8521#issuecomment-1065672367 change the outcome?

As for the stack overflow, I recommend enabling `CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y` in sdkconfig, then running the program again to check that you are definitely not overflowing the stack.

chipweinberger, 2022-06-20:
Thanks for the recommendations! Your suggestions are *tremendously* useful.

I'll try those and report back!

chipweinberger, 2022-06-24:
I changed my SD card access pattern, and remarkably am no longer repro'ing this anymore.

I used to be fread-ing 1 byte at a time, very quickly, *going in the reverse direction* (it was not cached and super slow). 

Now I'm fread-ing backwards in aligned 512 byte chunks, (which is much faster), and the issue no longer repro's. Or at least, I haven't repro'd it in 10+ minutes, when it used to repro in a few seconds.

I'll keep this issue open for now though in case hit it again & I might have time revert and try your patch in a few days.

chipweinberger, 2022-08-11:
Just want to add, I am hitting this error sometimes in v4.4.2. Still need to debug it further.

`E (22424) sdmmc_req: sdmmc_host_wait_for_event returned 0x107`

chipweinberger, 2022-09-03:
Also want to add, I'm beginning to think `sdmmc_req: sdmmc_host_wait_for_event returned 0x107` might be a power supply issue.

ESP32S3 Devkit-C + SD Card + plugging / unplugging USB device often results in 0x107.  Maybe 0x109 is related to power as well.

planevina, 2022-10-11:
> I changed my SD card access pattern, and remarkably am no longer repro'ing this anymore.
> 
> I used to be fread-ing 1 byte at a time, very quickly, _going in the reverse direction_ (it was therefore not cached and super slow).
> 
> Now I'm fread-ing backwards in aligned 512 byte chunks, (which is much faster), and the issue no longer repro's. Or at least, I haven't repro'd it in 10+ minutes - I used to repro in a few seconds.
> 
> I'll keep this issue open for now though in case hit it again & I might have time revert and try your patch in a few days.

I got the same error.

Can you tell me how to resove it ?thanks

chipweinberger, 2022-10-12:
@planevina my theory is that it is power related.

Try adding a much bigger capacitor across the sd card power and ground wires.

planevina, 2022-10-12:
SD_MMC.begin("/root", true, false, SDMMC_FREQ_DEFAULT)

I use this to change the sdmmc freq  from 40mhz to 20mhz
It's stable now.

MATTYGILO, 2023-09-15:
Was a solution found?

chipweinberger, 2023-09-16:
yes

- try using bigger capacitors across the power lines 
- try using a bigger linear regulator for the power supply 
- try using 20mhz instead of 40mhz
- try adding sleeps between reads (to reduce power draw)

chipweinberger, 2024-02-24:
closing. I havent seen this in awhile. solutions above^

closed 2024-02-24 by chipweinberger