espressif/esp-idf #9189: about asio.hpp

KOOKOKOK, 2022-06-19:
----------------------------- Delete below -----------------------------
vscode + esp-idf plugin

consle log:
C:/Espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include/asio/associated_allocator.hpp:19:10: fatal error: memory: No such file or directory
 #include <memory>

----------------------------- Delete above -----------------------------

```cpp
#include"asio.hpp"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

const ip::udp::endpoint ep( ip::address::from_string("127.0.0.1"), 8001);
static ip::udp::socket sock(ser,ep);
void udp_send(){
    // char buff[1024];
    string s = "hello world!";
    while ( true) {
        ip::udp::socket sock(service, ip::udp::endpoint(ip::udp::v4(), 0));
        sock.send_to(asio::buffer(s.c_str()), ep);
        vTaskDelay(pdMS_TO_TICKS(10));//10ms
    } 
}
.....

void app_main()
{
......
       xTaskCreate(udp_send,"udp_send",2048,NULL,1,NULL);

}

```
There is no problem with the WiFi connection. The code has been omitted.

moefear85, 2022-06-19:
Is your main file definately a C++ file? because:

> void app_main()

to my knowledge would only work if the file were C and has the extension ".c". But the rest of your code is C++. The header "asio.hpp" is also a C++ header. You must compile your main file as C++ for this to work, meaning you have to give it a ".cpp" extension, and that would mean everything has to be enclosed in:

> extern "C" {
> ...
> }

or atleast your main function would have to read:
> extern "C" void app_main()

0xjakob, 2022-06-20:
@KOOKOKOK @moefear85 

> You must compile your main file as C++ for this to work

I agree. Please make sure you are compiling with C++ (i.e., give the file where the above code is located the ending `.cpp` or `.cc`).

> everything has to be enclosed in: extern "C" {...

I'm afraid that's not entirely correct. If you do that with C++ code, you won't be able to use certain C++ features, in particular features that rely on name mangling (function overloads, templates). So declaring `extern "C" app_main()` should be enough.

KOOKOKOK, 2022-06-20:
@0xjakob @moefear85 Hi ,friends,thanks for your advice, I have encountered a new problem. I hope I can get some advice.
```

typedef union {
wifi_ap_config_t ap; /< configuration of AP */
wifi_sta_config_t sta; /< configuration of STA */
} wifi_config_t;

typedef struct {
uint8_t ssid[32]; /< SSID of target AP. */
uint8_t password[64]; /< Password of target AP. /
wifi_scan_method_t scan_method; /< do all channel scan or fast scan */
bool bssid_set; /< whether set MAC address of target AP or not. Generally, station_config.bssid_set needs to be 0; and it needs to be 1 only when users need to check the MAC address of the AP./
uint8_t bssid[6]; /< MAC address of target AP*/
uint8_t channel; /< channel of target AP. Set to 1~13 to scan starting from the specified channel before connecting to AP. If the channel of AP is unknown, set it to 0.*/
uint16_t listen_interval; /< Listen interval for ESP32 station to receive beacon when WIFI_PS_MAX_MODEM is set. Units: AP beacon intervals. Defaults to 3 if set to 0. */
wifi_sort_method_t sort_method; /< sort the connect AP in the list by rssi or security mode */
wifi_scan_threshold_t threshold; /< When sort_method is set, only APs which have an auth mode that is more secure than the selected auth mode and a signal stronger than the minimum RSSI will be used. */
wifi_pmf_config_t pmf_cfg; /< Configuration for Protected Management Frame. Will be advertized in RSN Capabilities in RSN IE. */
uint32_t rm_enabled:1; /< Whether Radio Measurements are enabled for the connection */
uint32_t btm_enabled:1; /< Whether BSS Transition Management is enabled for the connection */
uint32_t mbo_enabled:1; /< Whether MBO is enabled for the connection */
uint32_t reserved:29; /< Reserved for future feature set */
} wifi_sta_config_t;

wifi_config_t wifi_config = {
.sta = {
.ssid = "123",
.password = "123",
}
};

```

When I was in ' There will be no error when compiling under the '.c' file, when I am in' Errors will occur when compiling under the '.cpp' file.

The following is the output error message：
“
C99 designator 'ssid' outside aggregate initializer
};

”

KOOKOKOK, 2022-06-20:
Here is the code I used when compiling：

``` c
#include <esp_wifi_types.h>

   wifi_config_t wifi_config = {
        .sta = {
            .ssid = "123",
            .password = "123",
        }
    };

 void app_main(void)
{
}
```

KOOKOKOK, 2022-06-20:
@moefear85 
This is the code I mentioned above that produces different results when compiled in C and c++.

Here is the code I used when compiling:
```c
#include <esp_wifi_types.h>

   wifi_config_t wifi_config = {
        .sta = {
            .ssid = "123",
            .password = "123",
        }
    };

 void app_main(void)
{
}
```

moefear85, 2022-06-20:
@KOOKOKOK 

I can't reproduce the problem. The code you provided compiles in both C and C++ (although it throws warnings in C++ about the uninitialized other members). But it compiles and is flashable. Perhaps you made a typo elsewhere?

KOOKOKOK, 2022-06-20:
@moefear85 
I use vscode + esp-idf plugin v4.4.1.

When the compilation reaches this step：
Error occurred：

[978/983] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj 

 error: C99 designator 'ssid' outside aggregate initializer
     };
     ^
../main/main.cpp:9:5: error: C99 designator 'password' outside aggregate initializer
../main/main.cpp:9:5: warning: missing initializer for member 'wifi_sta_config_t::scan_method' [-Wmissing-field-initializers]
../main/main.cpp:9:5: warning: missing initializer for member 'wifi_sta_config_t::bssid_set' [-Wmissing-field-initializers]
../main/main.cpp:9:5: warning: missing initializer for member 'wifi_sta_config_t::bssid' [-Wmissing-field-initializers]
../main/main.cpp:9:5: warning: missing initializer for member 'wifi_sta_config_t::channel' [-Wmissing-field-initializers]

0xjakob, 2022-06-20:
@KOOKOKOK 

In C++11 (which is the standard in IDF 4.4), you should be able to do something like:
```c++
wifi_config_t wifi_config = {}; // zero-initialize the entire struct
wifi_config.sta.ssid = ... // setting required values
wifi_config.XXX = ...
...
```

moefear85, 2022-06-20:
@KOOKOKOK 

that's strange, if you're main function truly is exactly identical to the way you posted above, because I can't see any syntax errors in it, and it compiles in my case, I'm using VSCode and esp-idf v5, but I don't think the compiler or the standard changed in any relevant way here.
These errors can be hard to debug, it might even be some subtletly in another file in your project. But if you start a new project, paste the above into it, does it compile too? If so, you have something start your investigation with. If not, then there is something deeper wrong perhaps with your setup.

But if acceptable, please try 0xjakob's solution, it is easier.

KOOKOKOK, 2022-06-20:
@0xjakob 

The following is a link to my complete code. I am not sure whether other configuration files will also cause problems：

https://github.com/KOOKOKOK/esp32.git

KOOKOKOK, 2022-06-20:
@moefear85 

The following is a link to my complete code. I am not sure whether other configuration files will also cause problems：

https://github.com/KOOKOKOK/esp32.git

0xjakob, 2022-06-21:
@KOOKOKOK The link you sent is unreachable to me.

xiaguangbo, 2022-06-21:
like this:
```
extern "C"
{
 void app_main()
{
...
}
}
```

KOOKOKOK, 2022-06-21:
> like this:
> 
> ```
> extern "C"
> {
>  void app_main()
> {
> ...
> }
> }
> ```

hi，Simply using this change has no effect

xiaguangbo, 2022-06-21:
main.cpp、void app_main(){}：
[image]

main.cpp、extern "C"{void app_main(){}}：
Compiled. The warning means that there are still uninitialized members, you need to initialize all the listed ones
[image]

KOOKOKOK, 2022-06-21:
@xiaguangbo 

You can see the official example code. The official code is in ' Because I want to use the ASIO Library (a CPP Library) later, I must use the '.cpp' file. My code can be compiled successfully under '.c' file.

offical:
https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c

mine:
https://github.com/KOOKOKOK/esp32.git

xiaguangbo, 2022-06-21:
Can you restate the problem and the desired effect again, completely? If you think the picture is helpful to describe the problem, please attach the picture

KOOKOKOK, 2022-06-21:
@xiaguangbo 
At present, I need to compile in c++ environment, but there will be errors in c++ environment.

This is the result of my compilation failure in c++：
[image]

This is the result of my successful compilation under c：
[image]
[image]

xiaguangbo, 2022-06-21:
wifi_config_t sta_config = {};

//Assign ssid & password strings
strcpy((char*)sta_config.sta.ssid, "ssid");
strcpy((char*)sta_config.sta.password, "password");
sta_config.sta.bssid_set = false;

https://esp32.com/viewtopic.php?f=13&t=1317&p=5942&hilit=c+initializer#p5942

The master branch of esp-idf uses a new compiler, and this problem does not occur

closed 2022-06-23 by KOOKOKOK