espressif/esp-idf #9190: (SPIFFS)    comparison is always true due to limited range of data type [-Wtype-limits]

vinothkrstatiq, 2022-06-19:
I am trying to read the number of lines in the file, but i get warning like comparison is always true due to limited range of data type [-Wtype-limits] and my code get stuck in infinite loop but in other compiler it works file  only in esp idf i get this problem 
Can anyone help me with this?
 FILE *f2;
    	int count = 0; // Line counter (result)
    	char filename[100]="/spiffs/hello.txt";
    	 char c; // To store a character read from file

f2 = fopen(filename, "r");

    	// Check if file exists
    	if (f2 == NULL)
    	{
    		printf("Could not open file %s", filename);

    	}

    	// Extract characters from file and store in character c
    	for (c = getc(f2); c != EOF; c = getc(f2)){
    		if (c == '\n'){ // Increment count if this character is newline
    			count = count + 1;
    	     }
    	}
    	// Close the file
    	fclose(f2);
    	printf("The file %s has %d lines\n ", filename, count);
    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen("/spiffs/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fprintf(f, "Hello World!\n");
    fclose(f);

igrr, 2022-06-19:
`getc` function returns an `int`, so you need to change the type of `char c` to `int c`. Otherwise the value of EOF (-1) can't be represented — `char` is an unsigned type in Xtensa GCC.

vinothkrstatiq, 2022-06-19:
Thanks for the immediate response
got it

closed 2022-06-19 by vinothkrstatiq