espressif/esp-idf #9192: A2DP Sink Demo - IDF4.4.1

mitchjs, 2022-06-19:
ESP-IDF v4.4.1

im so confused.... i see other issues mentioning the audio packet loss during AVRC commands received
mostly volume controls
i set up latest release 4.4.1 and used the example A2DP.sink
all i did was, comment out the simulated volume changes (why is that even in the demo)
and build the project, and send it to my dev board
it works... then you change volume on host (iPhone) and boom audio gets choppy

i see posts dating over year ago with issues like this...
it looks like some commits make to IDF v4.1, which i would assume are there in IDF v4.4.1
is the ESP32 not up to the ability to do a simple A2DP Sink?
oddly it works for few seconds of playback... then all bad

```
I (130768) RC_TG: Volume is set by remote controller to: 50%
**W (130778) BT_APPL: Pkt dropped**
I (130778) RC_TG: AVRC set absolute volume: 46%
I (130778) RC_TG: Volume is set by remote controller to: 46%
**W (130788) BT_APPL: Pkt dropped**
```

ringbuffer problem, that has never been solved? is this a deeper issue at the controller level?

mitch

xiongweichao, 2022-06-20:
Hi @mitchjs ,

Sorry, I didn't reproduce the issue. Several questions to know.

1. Are you just running the sink demo?
2. What is the phone model used? Can this issue be reproduced with other phones?

Thanks

mitchjs, 2022-06-20:
yes, just the sink demo,. I commented out the simulated volume control change task
in bt_app_av.c
testing in ESP32_WROOM-32
with iPhone 11, with latest iOS

i just got out a Samsung TAB A  (2019) SM-T290
and its playing smooth!!!! NO PKT drops when change volume...
i got a iPad i can try too
and retry (for sanity sakes the iPhone 11)
ok that works too... 

i  know i had pks dropped, i show it in the logging... but not now...
thank you for looking at my issue, which is none issue now

prob can close this... ill reopen if it happens again and i can provide more info

mitchjs, 2022-06-20:
i went to sleep, woke up and just tested again... it happens
once it starts... then any volume control causes it to happen
(using iPhone)

then i reset the ESP32, reconnect the ESP-SPEAKER
and no pkt dropped messages

and move the onscreen slider fast, up.... down
and suddenly pkt dropped messages
it seems rapid changes in volume
so rapid incoming AVRCP messages?
once i get them to drop, even the slower volume changes
by the physical side button on iPhone gets a log of pkt dropped
(and of course audio cuts out)
so it doesnt recover
i can reproduce this at will on the iPhone
BUT NOT on the android tablet... i move the slider back and forth fast
and no issues
do you have an iphone to test with? and use the ON SCREEN touch slider in the MUSIC app
and once it starts, then any volume change will start showing these messages
i now try the youtube app, and move the slider on screen (which is tricky to show)
and its good, then go to the ios control panel volume and move its slider
and pkt dropped
and i have to disconnect to clear it up

```
I (18837494) RC_TG: AVRC set absolute volume: 30%
I (18837504) RC_TG: Volume is set by remote controller to: 30%
I (18837504) RC_TG: AVRC set absolute volume: 31%
I (18837514) RC_TG: Volume is set by remote controller to: 31%
W (18837524) BT_APPL: Pkt dropped

W (18837534) BT_APPL: Pkt dropped

I (18837534) RC_TG: AVRC set absolute volume: 32%
I (18837534) RC_TG: Volume is set by remote controller to: 32%
I (18837544) RC_TG: AVRC set absolute volume: 33%
I (18837544) RC_TG: Volume is set by remote controller to: 33%
W (18837554) BT_APPL: Pkt dropped

I (18837554) RC_TG: AVRC set absolute volume: 33%
I (18837564) RC_TG: Volume is set by remote controller to: 33%
W (18837574) BT_APPL: Pkt dropped

I (18837574) RC_TG: AVRC set absolute volume: 33%
I (18837574) RC_TG: Volume is set by remote controller to: 33%
I (18837584) RC_TG: AVRC set absolute volume: 33%
I (18837584) RC_TG: Volume is set by remote controller to: 33%
W (18837594) BT_APPL: Pkt dropped

I (18837604) RC_TG: AVRC set absolute volume: 52%
I (18837604) RC_TG: Volume is set by remote controller to: 52%
I (18837614) RC_TG: AVRC set absolute volume: 53%
I (18837614) RC_TG: Volume is set by remote controller to: 53%
W (18837624) BT_APPL: Pkt dropped

I (18837634) RC_TG: AVRC set absolute volume: 54%
I (18837634) RC_TG: Volume is set by remote controller to: 54%
W (18837644) BT_APPL: Pkt dropped

I (18837644) RC_TG: AVRC set absolute volume: 54%
I (18837644) RC_TG: Volume is set by remote controller to: 54%
I (18837654) RC_TG: AVRC set absolute volume: 55%
I (18837664) RC_TG: Volume is set by remote controller to: 55%
W (18837664) BT_APPL: Pkt dropped

I (18837674) RC_TG: AVRC set absolute volume: 55%
I (18837674) RC_TG: Volume is set by remote controller to: 55%
I (18837684) RC_TG: AVRC set absolute volume: 55%
I (18837684) RC_TG: Volume is set by remote controller to: 55%
W (18837694) BT_APPL: Pkt dropped

I (18837704) RC_TG: AVRC set absolute volume: 55%
I (18837704) RC_TG: Volume is set by remote controller to: 55%
W (18837714) BT_APPL: Pkt dropped

I (18837714) RC_TG: AVRC set absolute volume: 55%
I (18837724) RC_TG: Volume is set by remote controller to: 55%
I (18837734) RC_TG: AVRC set absolute volume: 55%
I (18837734) RC_TG: Volume is set by remote controller to: 55%
W (18837744) BT_APPL: Pkt dropped
```

mitchjs, 2022-06-20:
@xiongweichao , just in case you dont see that i reopend... as i found a way to make it happen consistently

mitchjs, 2022-06-22:
@BetterJincheng ,

my inital tests, it WORKED! no more audio breakup, no PKT Dropps
I want to test it some more... i need to wait till i have full access to my development area
tonight.

could you explain the simple change? I'm guessing making the task a lower priority?

i see "BtI2STask" task had/has same -3 priority, so now the "btAppTask" has a lower priority than the I2S task
(makes sense that the I2S Task is high)

thanks
mitch

BetterJincheng, 2022-06-23:
@mitchjs Our Bluetooth Stack handles the incoming audio packet in `BTC_task` which has the priority of `max-3`. We log `Volume is set by remote controller` in `btAppTask`. When we receive lots of AVRC command, the log will always be printed and the `BTC_Task` can be starved as it has lower priority than `btAppTask`. `BtI2STask` should have high priority because I2S is responsible for playing audio, which requires high real-time performance.
Turn to us anytime. Thanks!

BetterJincheng, 2022-06-24:
@mitchjs Yes, you are right. ESP_LOGx will block the task.

closed 2022-07-07 by espressif-bot