espressif/esp-idf #9193: LoadStoreAlignment-Error with ESP-WIFI-MESH and external PSRAM

IH303, 2022-06-19:
## Environment

- Development Kit:        custom
- Module or chip used:  ESP32S2-WROVER
- IDF version:                v5.0-dev-3567-ge8d172eb87
- Build System:             CMake
- Compiler version:       xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Linux
- IDE:                            vscode
- Power Supply:           external 5V

## Problem Description
I am using the ESP32S2-WROVER which comes with an external PSRAM of 2MB and I want this external PSRAM to be allocatable using malloc() as well. When I enable `Support for external SPI-connected RAM` and set `Maximum malloc() size, in bytes, to always put in internal memory` to 1024 or less (in sdkconfig) my app crashes in the moment the device tries to connect to the router via ESP-WIFI-MESH.

### Actual Behavior
When I set `Maximum malloc() size` larger than 2048 the app works fine, but I want to relieve the internal RAM as much as possible so I set the size to a rather small value (512 to 1024 bytes). The ESP-WIFI-MESH-API seems to allocate stuff in external RAM in this case, which leads to a crash (LoadStoreAlignment) for some reason. 

### Expected Behavior
The ESP-WIFI-MESH-API should work regardless of `Maximum malloc() size` like most of the other idf-components.
Maybe the API should force allocation in internal RAM with heap_caps_malloc(..,MALLOC_CAP_INTERNAL) , idk..

### Steps to reproduce

I dont know if it is specific to ESP32S2 but try to use a ESP32S2 with external PSRAM connected.

1. Open the _examples/mesh/ip_internal_network_ example
2. Set following configs in sdkconfig:
  - **Enable** `Support for external SPI-connected RAM` \
  (CONFIG_SPIRAM=y)
  - Set `Maximum malloc() size, in bytes, to always put in internal memory` to **1024** \
  (CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=1024)
  - Set `Example Configuration / Router SSID` to your SSID\
  (CONFIG_MESH_ROUTER_SSID="\<Your SSID\>")
  - Set `Example Configuration / Router password` to your router password\
  (CONFIG_MESH_ROUTER_PASSWD="\<Your password\>")

### Code to reproduce this issue
Keep the example code (_ip_internal_network_) as is

## Debug Logs

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1760
load:0x4004c000,len:0xab4
load:0x40050000,len:0x310c
entry 0x4004c1bc
I (21) boot: ESP-IDF v5.0-dev-3567-ge8d172eb87-dirty 2nd stage bootloader
I (21) boot: compile time 22:21:11
I (22) boot: chip revision: 0
I (26) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
...
...
...
I (966) phy_init: phy_version 2300,d67cf06,Feb 10 2022,10:03:07
I (1016) wifi:mode : sta (7c:df:a1:00:d6:aa)
I (1016) wifi:enable tsf
I (1016) wifi:mode : sta (7c:df:a1:00:d6:aa) + softAP (7c:df:a1:00:d6:ab)
I (1026) wifi:Total power save buffer number: 8
I (1026) wifi:Init max length of beacon: 752/752
I (1026) wifi:Init max length of beacon: 752/752
I (1046) mesh: <nvs>read layer:1
I (1046) mesh: <nvs>read assoc:0, err:0x1102
I (1046) wifi:Total power save buffer number: 8
I (2746) wifi:mode : sta (7c:df:a1:00:d6:aa)
I (2756) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:1
I (2756) mesh_main: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
I (2756) mesh_main: mesh starts successfully, heap:2102656, root not fixed

Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400de7d8  PS      : 0x00060030  A0      : 0x800f5f99  A1      : 0x3ffd8f10  
0x400de7d8: esp_mesh_check_vnd_ie at ??:?

A2      : 0x3f512d28  A3      : 0x3f512c7e  A4      : 0x3f512d6b  A5      : 0x00000040  
A6      : 0x00000045  A7      : 0x00000008  A8      : 0x800de7d8  A9      : 0x3ffd8ef0  
A10     : 0xca8cc441  A11     : 0xffffffff  A12     : 0x3f512d6b  A13     : 0x400957ac  
0x400957ac: aes_128_cbc_decrypt at /home/ibrahim/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/crypto_mbedtls.c:471

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3f512d6b  LBEG    : 0x3f512d6b  LEND    : 0x400957ac  LCOUNT  : 0x40026b04  
0x400957ac: aes_128_cbc_decrypt at /home/ibrahim/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/crypto/crypto_mbedtls.c:471

0x40026b04: _xt_user_exc at /home/ibrahim/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:643

Backtrace:0x400de7d5:0x3ffd8f100x400f5f96:0x3ffd8f30 0x40101ce5:0x3ffd8f80 0x40032cde:0x3ffd9070 0x40032d41:0x3ffd90c0 0x40036291:0x3ffd90e0 0x4003468f:0x3ffd9100 0x4002ef81:0x3ffd9130 
0x400de7d5: esp_mesh_check_vnd_ie at ??:?

0x400f5f96: ieee80211_parse_beacon at ??:?

0x40101ce5: sta_recv_mgmt at ??:?

0x40032cde: sta_input at ??:?

0x40032d41: sta_rx_cb at ??:?

0x40036291: ppRxPkt at ??:?

0x4003468f: ppTask at ??:?

0x4002ef81: vPortTaskWrapper at /home/ibrahim/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

ELF file SHA256: 95ad784491ef7d0a

CPU halted.
```

igrr, 2022-06-20:
Linking https://github.com/espressif/esp-idf/issues/8295 which seems to have the same cause.

Alvin1Zhang, 2022-09-23:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/29ae23884541a01e6dd0e73e348ed6b11e8ade0f, feel free to reopen.

closed 2022-09-23 by Alvin1Zhang