espressif/esp-idf #9196: Separate linker sections, bss, and attribute macros for RTC memory blocks

boarchuz, 2022-06-20:
**Is your feature request related to a problem? Please describe.**

- RTC slow and fast memory differ in capabilities, and are limited in size. It is often necessary to specify placement of a variable in one or the other.
- The only way to force a variable into a particular RTC memory is via the data segment  (eg. using RTC_FAST_ATTR or RTC_SLOW_ATTR), leading to the unexpected and inefficient outcome that [these](https://github.com/espressif/esp-idf/blob/e8d172eb87e606c52c6a995d118c1eb0f4810abc/components/esp_hw_support/esp_clk.c#L52) or `RTC_FAST_ATTR uint8_t foo[0x1000];`, for example, will _not_ be in bss but will rather bloat the binary with empty data to read and copy into memory
- Whatever magic normally allows the linker to put things in the appropriate sections doesn't exist for these custom RTC sections, so there should be a way to do this manually instead (eg. RTC_SLOW_BSS_ATTR)

**Describe the solution you'd like**

- Provide separate sections for each RTC memory block (eg. rtc_slow.noinit + rtc_fast.noinit) instead of treating them as a single configurable block (as with rtc.noinit)
- esp_attr.h macros for easy usage (eg. RTC_SLOW_BSS_ATTR, RTC_FAST_NOINIT_ATTR)
- Intialise _both_ rtc_slow.bss and rtc_fast.bss on restart [here](https://github.com/espressif/esp-idf/blob/e8d172eb87e606c52c6a995d118c1eb0f4810abc/components/esp_system/port/cpu_start.c#L313-L315)

**Describe alternatives you've considered**

- There are 'general' rtc.bss, rtc.noinit and rtc.data sections (eg. with RTC_DATA_ATTR), however a configuration option determines whether they are in RTC slow or fast memory, making this very fragile and limiting, and the RTC fast memory option is only available in single core mode

**Additional**

I've had a go at making the necessary changes here: https://github.com/boarchuz/esp-idf/tree/rtc_mem_sections
It seems to work well but there's a lot I'm unable to test. Please let me know if a PR would be worthwhile to review/discuss.

igrr, 2022-06-21:
Hi @boarchuz, thank you for the suggestion.

Regarding the issue that zero-initialized variables which could be in .bss are placed into .data, we have considered this in the past and decided to keep it this way. The reason is that while RTC_BSS_ATTR attribute is easy to introduce, it is also very easy to forget to change the attribute when adding or changing the variable initializer. It is very easy to overlook "BSS" when changing the initializer from 0, to, say, 1:
```c++
static RTC_BSS_ATTR int s_myval = 1;
```
in this case, no error will be produced, but the program will not work correctly. The increase to the binary size can not be very significant — typical IDF apps are on the order of 1 MB in size, but the RTC data variables can add at most a few kB. This means that the binary grows by less than 1% and the load time (from flash to RAM) also grows by that much. On balance, a less than 1% increase in app size and load time is a minor issue compared to the potential for introducing a bug by placing the variable into the wrong section.

boarchuz, 2022-06-21:
Hi @igrr,

Thanks for your feedback. I know it's not a huge difference but I think an opportunity to save a few kilobytes for 'free' is always worth considering. It all adds up.

The lack of error could be addressed by starting the section name with ".bss" (for example, by changing from `.rtc.slow.bss` to `.bss.rtc.slow`). The build will then fail with `error: only zero initializers are allowed in section 'bss.rtc.slow'`. EXT_RAM_BSS is also affected by this, I think.

Is that worth pursuing at all? As always, no issues closing this if it's still not worthwhile.

igrr, 2022-06-21:
> The build will then fail with error: only zero initializers are allowed in section 'bss.rtc.slow'

I actually didn't know about this, thank you for the tip. This does address my concern about accidentally using the wrong section name.

The only other issue is the generic problem that the code (linker script, startup code) becomes more complex, more difficult to test properly (because of the larger number of possible sections/attributes). If only a few customers end up using this and we don't test this adequately, this functionality may end up in a non-working state and be broken until someone notices. Newer chips (starting from ESP32-C3) don't make distinction between "slow" and "fast" RTC memory, so that's another factor against adding such distinction in the code.

If you have some suggestion how to test this functionality, and what to do with these attributes on chips which don't have a distinction between slow and fast RTC memory, that would really help.