espressif/esp-idf #9198: Problem with UART, if UART is placed in IRAM [problem found]

vavroa, 2022-06-21:
Problem found in IDF 4.4.1

If the SPI function that accesses the external flash is being executed and the ISR is called from the UART at this point, the entire program will crash.

The problem is that the ISR from UART uses RINGBUFFER, which has some functions located in external memory.

I am attaching a patch that I applied to this problem.

[0001-Ringbuffer-to-iram.zip](https://github.com/espressif/esp-idf/files/8949117/0001-Ringbuffer-to-iram.zip)
.

meltdown03, 2022-06-22:
You should make a Pull Request instead of attaching the patch here. That way it can be tested and edited by the Espressif team

igrr, 2022-06-27:
Hi @vavroa,

Could you please attach your sdkconfig file?

In v4.4.1 release, FreeRTOS and ringbuffer functions are already placed into IRAM, unless CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH configuration option is enabled. I'd like to check if you are seeing this issue with this option is enabled or not.

vavroa, 2022-06-27:
Indeed, this option was enabled, probably for this reason, the ringbuffer functions were placed in FLASH, which caused problems in their access from ISR.

Now I've tested it with option CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH turned off and it looks like it works.

igrr, 2022-06-27:
Thanks for confirming!

Looks like the action on our side would be to either prevent `CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH` from being used together with UART-ISR-in-IRAM, or somehow automatically move ringbuffer functions back to IRAM in that case. Another option would be to create a new option just for ringbuffers, `CONFIG_ESP_RINGBUFFER_PLACE_FUNCTIONS_INTO_FLASH`, and make sure it is disabled when UART IRAM interrupts are enabled.

vavroa, 2022-06-27:
Both options are a good idea.

In our case, we are at the limit of IRAM capacity, so we tried these different options.

closed 2022-07-19 by espressif-bot