espressif/esp-idf #9200: Bluedroid ble server example doesn't compile

sch0bert, 2022-06-21:
## Environment

Development Kit: ESP-WROOM-32 generic dev kit
Kit version (for WroverKit/PicoKit/DevKitC): v1
Module or chip used: ESP32-WROOM-32
IDF version: v4.4.1
Build System: idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
Operating System: Windows
(Windows only) environment type: ESP Command Prompt
Using an IDE?: No
Power Supply: USB

## Problem Description

I tried to disable the BLE Security module from Bluedroid and compile it but it throws an error in the linking process. I faced this issue in the example given in the IDF folder "examples\bluetooth\bluedroid\ble\gatt_security_server"

### Expected Behavior

Being able to compile.

### Actual Behavior

Error during compile.

### Steps to reproduce

1. idf.py menuconfig
2. Enter "Component config -> Bluetooth -> Bluedroid options -> (Disable) Include BLE Security module (SMP)
3. Save the configuration and exit menuconfig
4. idf.py build

Menuconfig option:
[image]

## Debug Logs

This is the error that is being thrown:

[image]

esp-cjh, 2024-02-27:
@sch0bert The APIs and data structures of SMP module are used in the gatt_security_server example. So If you disabled SMP module a compile error will occur.
If you want a BLE server example without SMP, refer to `examples\bluetooth\bluedroid\ble\gatt_server`.

sch0bert, 2024-02-27:
@esp-cjh Unfortunately I'm not longer following into this issue as it was seen in a previews company I worked and that was more than I year ago. If you want to close it, be my guest.

esp-cjh, 2024-02-27:
@sch0bert OK, thanks for your report.

closed 2024-02-27 by Alvin1Zhang