espressif/esp-idf #9201: Option to disable WIFI Prov BLE force encryption for Bluedroid

sch0bert, 2022-06-21:
**Is your feature request related to a problem? Please describe.**

In the latest implementations made to the framework, it was implemented an option to disable WIFI Provisioning BLE Force encryption for NimBLE in this commit: https://github.com/espressif/esp-idf/commit/916533ef92ce8cf50ef53fa94262cf94a5124bb5 to avoid displaying the request of pairing while connecting a device to your smart phone. This works great when you are using NimBLE in your project but they forgot to add it to Bluedroid.  So since IDF v4.4, in applications where Blueroid is used, this pairing request is displayed all the time and could cause me some issues with my users as it also ask you to provide access to your contact list.

**Describe the solution you'd like**

Not limit it to NimBLE.

**Additional context**

You can have more context based on this issue that I once opened: https://github.com/espressif/esp-idf/issues/8880

sch0bert, 2022-06-23:
@rahult-github why is this marked as done?

rahult-github, 2022-06-23:
Hi @sch0bert  , it is part of automated process. Change is done and pushed on tree internally. It will be synced on github branch soon. The bot has updated this issue.

sch0bert, 2022-06-23:
Thanks! Could you share the commit? I would like to do a cherry-pick

sch0bert, 2022-06-24:
@rahult-github Will it be included in v4.4.1 or is it going to be released as part of another version?

rahult-github, 2022-06-24:
Hi @sch0bert ,

Please find patch for the changes done attached which you can use currently, till the github tree gets synced. Please let me know if any issue is still observed. [0001-WifiProv-Extended-support-for-configuring-if-link-en.txt](https://github.com/espressif/esp-idf/files/8976077/0001-WifiProv-Extended-support-for-configuring-if-link-en.txt) 

The change needs to be backported to respective release branches , which i have initiated. There is a internal procedure to be followed to make this change part of next release. 

Thanks,
 Rahul

sch0bert, 2022-06-24:
I applied the changes in the framework, it works. Thanks!

sch0bert, 2022-06-24:
@rahult-github is v4.4.1 pairing with Bluedroid not working in iOS?

rahult-github, 2022-06-24:
Hi @sch0bert , you mean to say that with attached patch included, on v4.4.1 , you still observe issue ?

sch0bert, 2022-06-27:
Hello @rahult-github

Seems like it hasn't been integrated this fix into v4.4.1 yet

https://github.com/espressif/esp-idf/issues/7647

Could you confirm?

rahult-github, 2022-06-27:
Hi @sch0bert ,

Sorry, but I am not clear, if you query for merging in 4.4.1 is for current issue , or #7647 

The current issue #9201 :  So, as mentioned above,  we are in process of backporting change to relevant release branches. Once they are done, they would be available in the respective branch releases.
 
Issue #7647 :  The change you provided was picked and merged after release 4.4.1 was made. Hence you won't see it in that release version.
 
 Also, i think, we should not cross connect the 2 issues on same page, Update on respective issue page is also easy for tracking purpose.

Thanks,
 Rahul

sch0bert, 2022-06-27:
Sure, sorry for the confusion. We can then close this issue.

closed 2022-06-27 by sch0bert