espressif/esp-idf #9205: esp_ble_gap_update_whitelist is limited by esp_ble_gap_get_whitelist_size

GKoSon, 2022-06-22:
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.1/api-reference/bluetooth/esp_gap_ble.html?highlight=ble_scan_filter_allow_only_wlst#_CPPv430esp_ble_gap_get_whitelist_sizeP8uint16_t

I used esp_ble_gap_get_whitelist_size in ESP32S3

code
uint16_t length=100;
esp_ble_gap_get_whitelist_size(&length);
ESP_LOGE(GATTC_TAGA, "esp_ble_gap_get_whitelist_size=%d",length);

i find the length=11

when i put many MAC in whitelist

ESP_LOGE("GAP", "[%d]esp_ble_gap_update_whitelist =%d ",wlcount,esp_ble_gap_update_whitelist(true, whitelist_address, BLE_WL_ADDR_TYPE_RANDOM));//BLE_ADDR_TYPE_PUBLIC

i find the max is 12

so the MAX is 12

can i change the whitelist size ?like esp_ble_gap_set_whitelist_size(100)

zhp0406, 2024-02-26:
Yes, currently the ESP32-C3 chip whitelist can only support up to **12** entries, while the ESP32-C6 can support up to 15. The whitelist quantity for ESP32-C3 cannot be changed at the moment. What is your use case?

Alvin1Zhang, 2024-03-10:
Thanks for reporting, sorry for slow turnaround, feel free to reopen with more updates.

closed 2024-03-10 by Alvin1Zhang