espressif/esp-idf #9206: Not getting an IP when running the basic ethernet example

Henrikastro, 2022-06-22:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.3.3-30-g5be64c3111
- Build System:         [idf.py]
- Compiler version : (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [GitBash].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I am trying to get the [basic ethernet example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic) to work on my ESP-32, I'm using an W5500 to connect the ethernet cable to my ESP. I'm able to compile and run the basic example no problem, but the program is not setting an IP for my connection, stopping on doing anything after the Ethernet Started message. I've been trying to get it to work for a while but at this point I have no clue of what am I doing wrong. I am using the following pinouts to connect the ESP with the W5500:

| GPIO  | W5500 |
| ------------- | ------------- |
| GPIO18  | SCLK  |
| GPIO23  | MOSI  |
| GPIO19  | MISO  |
| GPIO5  | SCS  |
| GPIO4  | INT  |
| NC | RESET  |

The power source is also connected to the ESP and I have grounded the W5500 to the ESP.

### Expected Behavior

From the basic example page:
> I (394) eth_example: Ethernet Started
> I (3934) eth_example: Ethernet Link Up
> I (3934) eth_example: Ethernet HW Addr 30:ae:a4:c6:87:5b
> I (5864) esp_netif_handlers: eth ip: 192.168.2.151, mask: 255.255.255.0, gw: 192.168.2.2
> I (5864) eth_example: Ethernet Got IP Address
> I (5864) eth_example: ~~~~~~~~~~~
> I (5864) eth_example: ETHIP:192.168.2.151
> I (5874) eth_example: ETHMASK:255.255.255.0
> I (5874) eth_example: ETHGW:192.168.2.2
> I (5884) eth_example: ~~~~~~~~~~~

### Actual Behavior

> I (0) cpu_start: App cpu up.
> I (277) cpu_start: Pro cpu start user code
> I (277) cpu_start: cpu freq: 160000000
> I (277) cpu_start: Application information:
> I (282) cpu_start: Project name:     ethernet_basic
> I (287) cpu_start: App version:      v4.3.3-30-g5be64c3111-dirty
> I (294) cpu_start: Compile time:     Jun 22 2022 08:05:01
> I (300) cpu_start: ELF file SHA256:  72e97e7054665783...
> I (306) cpu_start: ESP-IDF:          v4.3.3-30-g5be64c3111-dirty
> I (313) heap_init: Initializing. RAM available for dynamic allocation:
> I (320) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (326) heap_init: At 3FFB4510 len 0002BAF0 (174 KiB): DRAM
> I (332) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (339) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (345) heap_init: At 4008D704 len 000128FC (74 KiB): IRAM
> I (353) spi_flash: detected chip: generic
> I (356) spi_flash: flash io: dio
> W (360) spi_flash: Detected size(4096k) larger than the size in the binary image
>  header(2048k). Using the size in the binary image header.
> I (374) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (384) w5500-mac: version=e0
> I (404) esp_eth.netif.glue: 02:00:00:12:34:56
> I (404) esp_eth.netif.glue: ethernet attached to netif
> I (404) eth_example: Ethernet Started
> I (170404) eth_example: Ethernet Link Up
> I (170404) eth_example: Ethernet HW Addr 02:00:00:12:34:56

and it stops right here, with some Ethernet Link Down and Link Up messages if I let it running for a while.

### Steps to reproduce

1. Use `. ./export.sh` on the esp-idf folder using git bash
2. cd to the example page
3. setup the sdkconfig with the right pins and SPI
4. build and flash using idf.py

## Other items if possible
I used the VSCode interface to setup the sdkconfig files.
Here's the [sdkconfig.txt](https://github.com/espressif/esp-idf/files/8957142/sdkconfig.txt), also I am using this pinouts as reference to connect to the W5500 to my board:

[image]

kostaond, 2022-06-23:
Hi @Henrikastro, thanks for nice issue report. It seems there is problem with SPI communication. How is your ESP32 connected to W5500? Via jump wires? If so, try to lower the SPI CLK. Let's say to 10 MHz.

Henrikastro, 2022-06-23:
Hi @kostaond, thanks for replying. Yes I'm using soldered jump wires. I've set the SPI CLK as follows 
[image]
even as low as 5 to no avail. Anything else I could be doing wrong? When using the Arduino Ethernet examples it also can't adress me an DHCP IP and it ends up giving me an static IP.

Henrikastro, 2022-06-23:
I think I left out something important. I'm trying to run the ESP connecting it directly to my computer, so it can act as an DHCP server and I can send commands to it via Ethernet cable.

kostaond, 2022-06-23:
So you want to be your computer DHCP server or ESP32?

Henrikastro, 2022-06-23:
The ESP must attribute an IP to itself when I connect it to the ethernet cable, so the ESP32 should be the DHCP I think? Or should my computer be doing this work? I'm really not sure

kostaond, 2022-06-23:
Well, it depends on your actual use case. Usually, DHCP server is running on router to which all devices are connected. In your case you have several options of how to make your setup working:
1) Configure static IP address on your computer and configure different static address in the same subnet on the ESP32. 
2) Configure and run DHCP server on your computer and make the ESP32 to be DHCP client (it will get the address from the server). Are you on Linux or Windows? Configuring DHCP server on Linux is quite simple, there is bunch of guides on the internet (e.g. https://www.linuxfordevices.com/tutorials/ubuntu/dhcp-server-on-ubuntu). However, it might be more tricky on Windows...
3) Make ESP32 as DHCP server and configure your computer as DHCP client.

Above is just a general description, you needed to to help with something specific, just let me know.

kostaond, 2022-06-23:
> Is there an ready example of a DHCP server in the esp-idf I can check or is it even possible to set it up myself using the library?

There is not ready to use example but check [this](https://github.com/espressif/esp-idf/issues/4086#issuecomment-559530841).

Henrikastro, 2022-06-23:
You're a lifesaver, thank you so much. I had the wrong idea on what the example was doing and having this answered made me see what was my mistake. 

> There is not ready to use example but check this

Thank you so much for this as well, I'll definetly be using this. I was trying to make the ESP32 the DHCP server and my computer the client. One last thing, since you mentioned especifically configuring my computer as DHCP client, let's say am able to set up the ESP to be the server. Do I need to setup anything else on my computer? Theoretically Windows should be able to work as client as if it was connected to a regular network right? Or since I'm doing it on the ESP32 I have to do some more tweaking?

kostaond, 2022-06-23:
Good to hear it helped you! Windows network interfaces are set as DHCP client by default.

Please close this issue if there is no problem related to ESP-IDF. If you have any other questions, don't hesitate to ask but please rather consider using ESP Forum at https://www.esp32.com/

closed 2022-06-23 by Henrikastro