espressif/esp-idf #9208: Problem esp-idf/components/driver/include/driver/i2s_pdm.h

remyhx, 2022-06-22:
Using ESP-IDF v5.0-dev-3639-gc2ccc383da (just updated), I noticed I2S.H is deprecated, so I ported my code to I2S_PDM.H

There is a mistake in the help file https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html (in the example "i2s_new_channel(&chan_cfg, &rx_handle, NULL);" should be " i2s_new_channel(&chan_cfg, NULL, &rx_handle);". But it's not the issue now.

For the PDM slot i can only choose I2S_SLOT_MODE_STEREO or I2S_SLOT_MODE_MONO. In mono there is data imported, but just a quiet channel, I need to change it to stereo to hear the sound. I guess the wrong slot (like left/right) is chosen, but I can't change that.

Like the example:

/* Allocate an I2S rx channel */
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER);
    i2s_new_channel(&chan_cfg, NULL, &rx_handle);

    /* Init the channel into PDM RX mode */
    i2s_pdm_rx_config_t pdm_rx_cfg = {
        .clk_cfg = I2S_PDM_RX_CLK_DEFAULT_CONFIG(SAMPLE_RATE),
        .slot_cfg = I2S_PDM_RX_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO),
        .gpio_cfg = {
            .clk = WS_PIN,
            .din = DATA_IN_PIN,
            .invert_flags = {
                .clk_inv = false,
            },
        },
    };
    
    i2s_channel_init_pdm_rx_mode(rx_handle, &pdm_rx_cfg);

i2s_channel_enable(rx_handle); is called once and for reading i2s_channel_read(rx_handle, (char *)buf, AUDIO_BUF, &bytes_read, portMAX_DELAY);
i2s_channel_disable(rx_handle); at the end.

So in stereo I can hear myself, in Mono I can't,

L-KAYA, 2022-06-23:
Hi remyhx, it's glad to hear your feedback! And the new driver currently indeed has some bugs in PDM mode. The issue you mentioned is a known issue and it's already in progress, there will be another update for the I2S soon, thanks for you reporting!

closed 2022-07-22 by espressif-bot

silelis, 2023-08-25:
Hello,

I use componentr compatible with 4.4.1 API and it use i2s.h which is deprecioned. Therefor I have compilation errot (fatal error: driver/i2s.h no sutch file or directory).
Is there some way to use legacy drivers in 5.0 api?

L-KAYA, 2023-08-29:
Hi @silelis , if you're using IDF v5.0, the legacy I2S driver should still available, you can refer to the [test app](https://github.com/espressif/esp-idf/blob/release/v5.0/components/driver/test_apps/i2s_test_apps/legacy_i2s_driver/main/test_legacy_i2s.c) of the legacy driver.

If there are still any problems, please open an new Github issue, thanks!