espressif/esp-idf #9211: panic in `esp_apptrace_riscv_down_buffer_get`

driftregion, 2022-06-23:
## Environment

- Development Kit:    None
- Module or chip used:  ESP32-C3-WROOM
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3424-gbb23d783c0
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:  USB
## Problem Description

A: Increasing `CONFIG_FREERTOS_HZ` to 1000 while 
B: SysView apptrace is enabled and  
C: `nvs_flash_init()` is called 
results in a panic. 

### Expected Behavior

apptrace works normally

### Actual Behavior

panic

### Steps to reproduce

Using `examples/storage/nvs_rw_value`, 

1. Apply the following defconfig
```
CONFIG_APPTRACE_DEST_JTAG=y
CONFIG_APPTRACE_SV_ENABLE=y
CONFIG_FREERTOS_HZ=1000
```
2. Observe panic

## Debug Logs

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806bc
0x403806bc: esp_restart_noos_dig at /home/user/esp/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6108,len:0x1994
load:0x403ce000,len:0x8f8
load:0x403cf600,len:0x2e28
entry 0x403ce000
I (35) boot: ESP-IDF v5.0-dev-3424-gbb23d783c0-dirty 2nd stage bootloader
I (35) boot: compile time 11:00:51
I (35) boot: chip revision: 3
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (44) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08df8h ( 36344) map
I (110) esp_image: segment 1: paddr=00018e20 vaddr=3fc8ca00 size=01764h (  5988) load
I (114) esp_image: segment 2: paddr=0001a58c vaddr=40380000 size=05a8ch ( 23180) load
I (126) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1bad0h (113360) map
I (147) esp_image: segment 4: paddr=0003baf8 vaddr=40385a8c size=06ef8h ( 28408) load
I (152) esp_image: segment 5: paddr=000429f8 vaddr=50000010 size=00010h (    16) load
I (156) boot: Loaded app from partition at offset 0x10000
I (159) boot: Disabling RNG early entropy source...
I (176) cpu_start: Pro cpu up.
I (184) cpu_start: Pro cpu start user code
I (184) cpu_start: cpu freq: 160000000 Hz
I (184) cpu_start: Application information:
I (187) cpu_start: Project name:     nvs_rw_value
I (193) cpu_start: App version:      v5.0-dev-3424-gbb23d783c0-dirty
I (200) cpu_start: Compile time:     Jun 23 2022 11:00:47
I (206) cpu_start: ELF file SHA256:  7a836ad226316854...
I (212) cpu_start: ESP-IDF:          v5.0-dev-3424-gbb23d783c0-dirty
I (219) heap_init: Initializing. RAM available for dynamic allocation:
I (226) heap_init: At 3FC977F0 len 00028810 (162 KiB): DRAM
I (232) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (239) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (246) spi_flash: detected chip: generic
I (250) spi_flash: flash io: dio
W (254) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (267) sleep: Configure to isolate all GPIO pins in sleep state
I (274) sleep: Enable automatic switching of GPIO sleep configuration
I (281) esp_apptrace: Apptrace initialized on CPU0. Tracing control block @ 0x3fc96cb8.
I (290) cpu_start: Starting scheduler.
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x42004b2e  RA      : 0x403849ae  SP      : 0x3fc8e7c0  GP      : 0x3fc8d200  
0x42004b2e: esp_apptrace_riscv_down_buffer_get at /home/user/esp/esp-idf/components/app_trace/port/riscv/port.c:211

0x403849ae: esp_apptrace_read at /home/user/esp/esp-idf/components/app_trace/app_trace.c:190

TP      : 0x3fc923f8  T0      : 0x7fffffff  T1      : 0x600c2130  T2      : 0x00000003  
S0/FP   : 0x00000001  S1      : 0x00000000  A0      : 0x3fc8cb6c  A1      : 0x3fc8e7c4  
A2      : 0x3fc8e7c8  A3      : 0x00000000  A4      : 0x42004b2e  A5      : 0x3fc8eca8  
0x42004b2e: esp_apptrace_riscv_down_buffer_get at /home/user/esp/esp-idf/components/app_trace/port/riscv/port.c:211

A6      : 0x00000000  A7      : 0xffffffff  S2      : 0x3fc8e80c  S3      : 0x3fc8e82f  
S4      : 0x3fc9b740  S5      : 0x00000040  S6      : 0x3fc8cda0  S7      : 0x00000100  
S8      : 0x00000000  S9      : 0x3fc9b350  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x3fc97698  T4      : 0x00000000  T5      : 0x00000001  T6      : 0xffffffff  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

esp_apptrace_riscv_down_buffer_get (hw_data=0x3fc8cb6c <s_trace_hw_data>, size=0x3fc8e7c4 <xIsrStack+1316>, tmo=0x3fc8e7c8 <xIsrStack+1320>) at /home/user/esp/esp-idf/components/app_trace/port/riscv/port.c:211
211     {
#0  esp_apptrace_riscv_down_buffer_get (hw_data=0x3fc8cb6c <s_trace_hw_data>, size=0x3fc8e7c4 <xIsrStack+1316>, tmo=0x3fc8e7c8 <xIsrStack+1320>) at /home/user/esp/esp-idf/components/app_trace/port/riscv/port.c:211
#1  0x403849ae in esp_apptrace_read (dest=dest@entry=ESP_APPTRACE_DEST_JTAG, buf=buf@entry=0x3fc8e82f <xIsrStack+1423>, size=size@entry=0x3fc8e80c <xIsrStack+1388>, user_tmo=user_tmo@entry=0) at /home/user/esp/esp-idf/components/app_trace/app_trace.c:190
#2  0x40383702 in SEGGER_RTT_ReadNoLock (BufferIndex=BufferIndex@entry=1, pData=pData@entry=0x3fc8e82f <xIsrStack+1423>, BufferSize=BufferSize@entry=1) at /home/user/esp/esp-idf/components/app_trace/sys_view/esp/SEGGER_RTT_esp.c:124
#3  0x40383b3e in _HandleIncomingPacket () at /home/user/esp/esp-idf/components/app_trace/sys_view/SEGGER/SEGGER_SYSVIEW.c:466
#4  0x40383d18 in _SendPacket (pStartPacket=<optimized out>, pEndPacket=<optimized out>, EventId=EventId@entry=2) at /home/user/esp/esp-idf/components/app_trace/sys_view/SEGGER/SEGGER_SYSVIEW.c:744
#5  0x4038425c in SEGGER_SYSVIEW_RecordEnterISR (IrqId=<optimized out>) at /home/user/esp/esp-idf/components/app_trace/sys_view/SEGGER/SEGGER_SYSVIEW.c:1846
#6  0x40380f38 in non_shared_intr_isr (arg=arg@entry=0x3fc9b204) at /home/user/esp/esp-idf/components/esp_hw_support/intr_alloc.c:426
#7  0x4038b636 in _global_interrupt_handler (sp=<optimized out>, mcause=<optimized out>) at /home/user/esp/esp-idf/components/riscv/interrupt.c:64
#8  0x403801ee in _interrupt_handler ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
ELF file SHA256: 7a836ad226316854
```

ESP-Marius, 2022-06-23:
This seems to be the same issue as https://github.com/espressif/esp-idf/issues/9131

If you try the fix I suggested there does it solve the issue for you?

driftregion, 2022-06-23:
@ESP-Marius , yes it does. Thank you. Please close this issue if you believe it is duplicate.

ESP-Marius, 2022-06-23:
@driftregion Great! 

Yeah seems like a duplicate. I'll close this one then, feel free to subscribe to the other issue to get notified when the fix is merged.

closed 2022-06-23 by ESP-Marius