espressif/esp-idf #9212: Timeout when there's an erasure of the flash chip

filgra96, 2022-06-23:
## Environment

- Development Kit: none
- Module or chip used: ESP32-S2
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32s2-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes - VSCode
- Power Supply: USB

## Problem Description

I tried to perform an `esp_flash_erase_chip()` of a flash memory chip that is located outside of the ESP32-S2 module. It is a 16MB Winbond flash 25Q128JVFQ.
When I try to do the operation, I obtain a ESP_ERR_TIMEOUT.
Reading [this topic](https://esp32.com/viewtopic.php?t=23425), I tried to check the *Flash timeout checkout disabled* option in menuconfig.
Doing so, I confirm the absence of the timeout error, however I find multiple WDT triggers before it finishes the job (see the log).

### Expected Behavior

I think it should erase the chip without problems. I can't understand if there is something missing in the steps I followed.

### Actual Behavior

```
E (40496) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (40496) task_wdt:  - IDLE (CPU 0)
E (40496) task_wdt: Tasks currently running:
E (40496) task_wdt: CPU 0: main
E (40496) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4009D11F:0x3FFC5E200x40024552:0x3FFC5E40 0x4002D543:0x3FFD1EB0 0x4002D6F9:0x3FFD1EF0 0x40034456:0x3FFD1F30 0x40034F17:0x3FFD1F70 0x40034F45:0x3FFD1F90 0x40034D5D:0x3FFD1FC0 0x4002585A:0x3FFD1FE0 0x4008BACE:0x3FFD2010 0x40087AF7:0x3FFD2030 0x4010C747:0x3FFD2060
0x4009d11f: task_wdt_isr at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40024552: _xt_lowint1 at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4002d543: gpspi_flash_ll_set_dummy at C:/Espressif/frameworks/esp-idf-v4.4/components/hal/esp32s2/include/hal/gpspi_flash_ll.h:361
 (inlined by) spi_flash_hal_gpspi_configure_host_io_mode at C:/Espressif/frameworks/esp-idf-v4.4/components/hal/spi_flash_hal_common.inc:124

0x4002d6f9: spi_flash_hal_gpspi_common_command at C:/Espressif/frameworks/esp-idf-v4.4/components/hal/spi_flash_hal_common.inc:150

0x40034456: memspi_host_read_status_hs at C:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash/memspi_host_driver.c:123

0x40034f17: spi_flash_chip_generic_read_reg at C:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:376

0x40034f45: spi_flash_chip_generic_wait_idle at C:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:426

0x40034d5d: spi_flash_chip_generic_erase_chip at C:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash/spi_flash_chip_generic.c:136

0x4002585a: esp_flash_erase_chip at C:/Espressif/frameworks/esp-idf-v4.4/components/spi_flash/esp_flash_api.c:526

0x4008bace: spi_ext_flash_erase_partition at C:\Users\fgraziani\Documents\Lavoro\27b1con\build/../components/spi_manager_component/spi_manage.c:460

0x40087af7: app_main at C:\Users\fgraziani\Documents\Lavoro\27b1con\build/../main/main.c:141 (discriminator 3)

0x4010c747: main_task at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/port_common.c:129
```

igrr, 2022-06-23:
@filgra96 About the task watchdog, please check the related section of the Programming Guide: https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32/api-reference/system/wdts.html#task-watchdog-timer. The task watchdog triggers because your task is running a long operation without yielding to the idle task.

You can try using `esp_flash_erase_region` function instead of `esp_flash_erase_chip` in this case — this will allow the driver to yield once in a while, preventing the watchdog timeout.

One possible improvement for us to make is to let `esp_flash_erase_chip` function yield while waiting for the erase operation to complete (via check_yield in the driver).

filgra96, 2022-06-23:
Yes, I know I could either increase the timer of the WDT or feed the WDT. My main question was why the WDT gets triggered using an official library, but you answered. Thank you!

filgra96, 2022-07-01:
I tried to erase the single sectors of the flash memory, however the WDT get triggered again..
```
E (65492) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (65492) task_wdt:  - IDLE (CPU 0)
E (65492) task_wdt: Tasks currently running:
E (65492) task_wdt: CPU 0: main
E (65492) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4009DD83:0x3FFC5A300x400245DE:0x3FFC5A50 0x4001AA8D:0x3FFD3630 0x40026049:0x3FFD3650 0x40034B76:0x3FFD3670 0x40034E5D:0x3FFD36A0 0x40025ADA:0x3FFD36E0 0x4008BDB8:0x3FFD3730 0x40087B3D:0x3FFD3750 0x4010D9CF:0x3FFD3790
0x4009dd83: task_wdt_isr at C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_system/task_wdt.c:183 (discriminator 3)

0x400245de: _xt_lowint1 at C:/Espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40026049: delay_us at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/spi_flash_os_func_app.c:148

0x40034b76: spi_flash_chip_generic_wait_idle at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/spi_flash_chip_generic.c:443

0x40034e5d: spi_flash_chip_winbond_erase_sector at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/spi_flash_chip_winbond.c:118
 (inlined by) spi_flash_chip_winbond_erase_sector at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/spi_flash_chip_winbond.c:99

0x40025ada: esp_flash_erase_region at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/esp_flash_api.c:616

0x4008bdb8: spi_ext_flash_erase_partition at C:\Users\fgraziani.ELFA\Documents\Lavoro\Wurth\27b1con\build/../components/spi_manager_component/spi_manage.c:468

0x40087b3d: app_main at C:\Users\fgraziani.ELFA\Documents\Lavoro\Wurth\27b1con\build/../main/main.c:150 (discriminator 2)

0x4010d9cf: main_task at C:/Espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/port_common.c:129
```

here my code (for a 16MB of flash and 4KB of sector):
```C
esp_err_t
spi_ext_flash_erase_partition (esp_flash_t * flash_chip)
{
    esp_err_t ret = ESP_FAIL;
    uint32_t iteration = 0;
    uint32_t start_addr = 0;
    const uint32_t sector_size = 4096;
    const uint32_t number_of_sect = 4096;

    for (iteration = 0; iteration < number_of_sect; ++iteration)
    {
        ret = esp_flash_erase_region(flash_chip, start_addr, sector_size);
        if (ESP_OK != ret)
        {
            break;
        }
        else
        {
            start_addr += sector_size;
        }
    }

    if (ESP_OK == ret)
    {
        ESP_LOGI(g_p_tag, "Memoria flash esterna cancellata con successo");
    }
    else
    {
        ESP_LOGE(g_p_tag,
                 "Tentativo di cancellazione memoria esterna con errore %s",
                 esp_err_to_name(ret));
    }

    return ret;
}
```

igrr, 2022-07-01:
Could you please attach your sdkconfig file? I'm interested in the values of options which control yielding during flash operations:

CONFIG_SPI_FLASH_YIELD_DURING_ERASE
CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS
CONFIG_SPI_FLASH_ERASE_YIELD_TICKS

filgra96, 2022-07-01:
I manipulated the parameters according to [the offical documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#os-functions), i started with the following parameters (it triggers the wdt using `esp_flash_erase_chip()` or `esp_flash_erase_region()`):
- `# CONFIG_ESP_TASK_WDT_PANIC is not set`
- `CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y`
- `CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=20`
- `CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=1`
- `CONFIG_ESP_TASK_WDT_TIMEOUT_S=5`

and I switched to these (also here it triggers with both instructions):
- `# CONFIG_ESP_TASK_WDT_PANIC is not set`
- `CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y`
- `CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=2`
- `CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=20`
- `CONFIG_ESP_TASK_WDT_TIMEOUT_S=5`

filgra96, 2022-07-05:
@igrr This is my complete sdkconfig if you need it [sdkconfig.txt](https://github.com/espressif/esp-idf/files/9043538/sdkconfig.txt)

yehj0107, 2022-07-06:
I got the same issue. Is there any solution or temporary workaround?

filgra96, 2022-07-12:
Also, I can't understand why the automatic formatting (after a failed mount operation) requires1/2 seconds, while erasing the memory requires a minute or two. Is there a way to manually perform the same formatting operation that happens when mounting? In this case there aren't timeouts

igrr, 2022-07-12:
@filgra96 I'm sorry, I have missed the point in your original issue report that you are working with an external flash chip.

For some reason we only implement the automatic "yield" operation during flash erase only for the main flash chip:
https://github.com/espressif/esp-idf/blob/c9140caf8cb501f17e4e0703a43a5693a6c28f6d/components/spi_flash/spi_flash_os_func_app.c#L190-L211

(the "yield" and "check_yield" functions are not implemented for external flash)

I'll check internally to see if there was any underlying reason for this. If not, we will add support for automatic yield for external flash.

In the meantime, you can add logic like this to your implementation of `spi_ext_flash_erase_partition` function:
```c++
    for (iteration = 0; iteration < number_of_sect; ++iteration)
    {
        if (iteration % 16 == 0) vTaskDelay(1);  // Yield for 1 tick every 16 sectors
        ...
```

> Also, I can't understand why the automatic formatting (after a failed mount operation) requires1/2 seconds, while erasing the memory requires a minute or two. Is there a way to manually perform the same formatting operation that happens when mounting? In this case there aren't timeouts

I think the reason is that FAT FS library only initializes part of the sectors when formatting the filesystem, not all the sectors.

At what point do you want to erase the filesystem — before calling the esp_vfs_fat_spiflash_mount or after?

filgra96, 2022-07-13:
Thank you for your help @igrr 
> At what point do you want to erase the filesystem — before calling the esp_vfs_fat_spiflash_mount or after?

My idea is: the first time I use the flash, it will be validated and mounted (it will fail so the system will format and mount again the flash device).
After that, in case of some unwanted behavior (detected by a read of a known content in flash) I want to erase the chip and I unmount it. Then I mount it again. The mount operation will fail again and the chip will be formatted.
I experimented this last part and I'm pretty sure it works in this way.

At the moment of speaking, because of it was impossible to erase without the WDT trigger, I created a function that invalidate (i.e. it erases only 32KB block of memory and then stops) the flash and it seems sufficient for a re-format operation performed when I mount it.
It is not a 100% correct way, infact I think your logic mentioned
> In the meantime, you can add logic like this to your implementation of spi_ext_flash_erase_partition function:

will be ok. But I think it will require a lot of time to perform the erasure, right?

chipweinberger, 2022-10-10:
Another "temporary" solution for anyone else that still hits this for any reason.

It's simple to disable the WDT, and re-enable it.

Code:
```
esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(0));
esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(1));

// call format code here

esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(0));
esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(1));
```

chipweinberger, 2022-10-10:
@igrr, also I want to add, **I've hit the WDT during SD card mounting as well.** I think my WDT was 5 seconds. Ideally, mounting should yield as well.

closed 2023-01-13 by espressif-bot