espressif/esp-idf #9219: CS pin for SD card by SPI using callback

ElieSKickMaker, 2022-06-23:
Hello, 

I've got a SD card connected to my esp32 but I'm using a gpio expander to drive the CS pin. Today I can't use the sd driver because CS pin must be a pin driver. 

Is it possible to replace [cs_high](https://github.com/espressif/esp-idf/blob/c2ccc383dae2a47c2c2dc8c7ad78175a3fd11361/components/driver/sdspi_host.c#L121) and [cs_low ](https://github.com/espressif/esp-idf/blob/c2ccc383dae2a47c2c2dc8c7ad78175a3fd11361/components/driver/sdspi_host.c#L127)
by a custom callback ? 
Or something else if you know a workaround.

Thanks for your help

igrr, 2022-06-23:
Hi @ElieSKickMaker,

There is no functionality yet for overriding the cs related functions. You can either patch the source code of IDF on your side, or use the [linker wrap feature](https://github.com/espressif/esp-idf/issues/7681#issuecomment-941957783) to override GPIO related functions.
For example, you can wrap `gpio_config` and `gpio_set_level` functions, using roughly the following logic:
```c++
void __wrap_gpio_set_level(int gpio, int level) {
    if (gpio > 100) {  /* GPIO number 100+n corresponds to expander pin n */
        my_gpio_expander_set_level(gpio - 100, level);
    } else {
        __real_gpio_set_level(gpio, level);
    }
}
```

Then you can set CS pin number to such a "virtual" number (100+n).

ElieSKickMaker, 2022-06-23:
Hello , 

Thanks for your help, I don't know the linker wrap feature. I will try to wrap  the cs_high() and cs_low() !