espressif/esp-idf #9221: ESP-LCD-RGB gitlab solution repo - Compilation error

tushar-rs, 2022-06-23:
Hi there, 

we have been given access for ESP-LCD-RGB gitlab solution repo for some early experiments on RGB display. The link for the same is [esp-lcd-rgb](https://glab.espressif.cn/esp-solution/esp-lcd-rgb/esp-lcd-rgb).
We have not found issue filing option in this repo/gitlab may be due to limited access. 

There are compilation error while compiling this gitlab repo examples. Below is the detail. 

We tried on two different IDFs. IDF4.4.1 and latest master i.e IDF5 (https://github.com/espressif/esp-idf) on window 10. 
On IDF5: 
1.	Open “ESP-IDF5 CMD” 
2.	Change IDF_PATH to %path_new_gitlab%/esp-lcd-rgb-master/esp-idf
3.	Change directory to %path_new_gitlab%/esp-lcd-rgb-master/example/lvgl_demo
4.	Idf.py fullclean and then idf.py –set-target esp32s3
It gives toolchain compatibility error: refer attachment **IDF5_error.txt**

On IDF4.4 
1.	Open “ESP-IDF4.4 CMD” 
2.	Change IDF_PATH to %path_new_gitlab%/esp-lcd-rgb-master/esp-idf
3.	Change directory to %path_new_gitlab%/esp-lcd-rgb-master/example/lvgl_demo
4.	Idf.py fullclean; idf.py –set-target esp32s3; idf.py menuconfig and set SPIRAM enable with quad mode 
5.	Idf.py build give error related to spiram.c. Refer attached **IDF4_4_error.txt**

Can you please help on, 
1.	Help to resolve this component build error.  OR  2 (below). 
2.	Provide the list of source file that is changed for PSRAM optimization so that we can replace the same in IDF 4.4 and try to build the application. 
3.	Which “esp_lcd” component is latest out of  “%path_new_gitlab%/esp-lcd-rgb\components\esp_lcd”   and    “%path_new_gitlab%/esp-lcd-rgb\esp-idf\components\esp_lcd”? 
4.	When these changes will be part of the public master repo i.e idf5?

## Environment

- Development Kit: [ESP32-S3-RGB] ([url](https://www.aliexpress.com/item/3256803826937918.html?UTABTest=aliabtest286446_400232&_randl_currency=USD&_randl_shipto=US&src=google&memo1=freelisting&src=google&albch=shopping&acnt=708-803-3821&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=15083225414&albag=137595356308&trgt=303059982125&crea=en1005004013252670&netw=u&device=c&albpg=303059982125&albpd=en1005004013252670&gclid=Cj0KCQjwzLCVBhD3ARIsAPKYTcQBIysLS7M9aNKE73h4ysO8-LmuAElvlALweX12VcseheKuInquCvgaAp0REALw_wcB&gclsrc=aw.ds&aff_fcid=46a315ff17d4489598f4b92c66a1851f-1655480479529-03751-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=46a315ff17d4489598f4b92c66a1851f-1655480479529-03751-UneMJZVf&terminal_id=3c3697fcd23e4264ad641684ceaa5c79&OLP=1082700508_f_group0&o_s_id=1082700508&afSmartRedirect=y&gatewayAdapt=4itemAdapt))
- MCU: ESP32-S3R8
- ESP-IDF: v4.4.1
- Chip revision: 0
- Host: windows 10

[IDF4_4_error.txt](https://github.com/espressif/esp-idf/files/8968504/IDF4_4_error.txt)
[IDF5_error.txt](https://github.com/espressif/esp-idf/files/8968505/IDF5_error.txt)

igrr, 2022-06-23:
Hi @tushar-rs, could you please raise this issue in that example project repo, instead? The error stem from the fact that the project seems to be using a customized esp-idf copy, so it's out of scope for ESP-IDF Github project.

wujiangang, 2022-06-24:
@tushar-rs Now you can raise your issue in gitlab, since the URL has some private info, so not share here. Please raise the issue under project issue-submit.

tushar-rs, 2022-06-24:
I have added the issue in GitLab, closing this one here.

closed 2022-06-24 by tushar-rs

weilian1977, 2023-01-26:
关注一下