espressif/esp-idf #9223: websocket client not able to receive data after some interval

vinothkrstatiq, 2022-06-24:
I am using Espressif-IDE Version: 2.5.0
Operating System: windows 10
Java Runtime Version: 11.0.14.1+10-LTS
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.5.0.202206080313
ESP-IDF v4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.10.4
I am using websocket client 
without closing the websocket connection I am trying to continuously read JSON message sent from cloud 
it is able to read the data from cloud ,but after some interval it is not able receive the data.

if I close the websocket connection and again reconnect, it is able to read but without that I am not able to read the data.

without closing the websocket connection, I want to continuously read the json data which is coming from cloud.
can anyone help me here ?

vinothkrstatiq, 2022-07-06:
@gabsuren  
hi gabsuren could you treat this issue as priority ?
our project get stuck here because of this issue .
immediate help would be highly appreciated 
Thanks in advance

gabsuren, 2022-07-06:
@vinothkrstatiq sure, we are looking into this and will come back to you soon!

gabsuren, 2022-07-06:
@vinothkrstatiq could you please also provide the logs, it can help us more easily to  reproduce the issue?

vinothkrstatiq, 2022-07-06:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 18:03:42
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 002f0000
I (89) boot:  3 storage          Unknown data     01 82 00300000 00100000
I (97) boot: End of partition table
I (101) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=219dch (137692) map
I (167) esp_image: segment 1: paddr=00031a04 vaddr=3ffbdb60 size=05890h ( 22672) load
I (176) esp_image: segment 2: paddr=0003729c vaddr=40080000 size=08d7ch ( 36220) load
I (191) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=c2880h (796800) map
I (479) esp_image: segment 4: paddr=001028a8 vaddr=40088d7c size=15b5ch ( 88924) load
I (516) esp_image: segment 5: paddr=0011840c vaddr=50000000 size=00010h (    16) load
I (531) boot: Loaded app from partition at offset 0x10000
I (531) boot: Disabling RNG early entropy source...
I (543) cpu_start: Pro cpu up.
I (543) cpu_start: Starting app cpu, entry point is 0x4008131c
0x4008131c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (559) cpu_start: Pro cpu start user code
I (560) cpu_start: cpu freq: 160000000
I (560) cpu_start: Application information:
I (564) cpu_start: Project name:     blemesh
I (569) cpu_start: App version:      1
I (573) cpu_start: Compile time:     Jul  6 2022 18:03:23
I (580) cpu_start: ELF file SHA256:  d263bf47420ed6fd...
I (585) cpu_start: ESP-IDF:          v4.4-dirty
I (591) heap_init: Initializing. RAM available for dynamic allocation:
I (598) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (604) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (610) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (616) heap_init: At 3FFC95B0 len 00016A50 (90 KiB): DRAM
I (622) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (629) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (635) heap_init: At 4009E8D8 len 00001728 (5 KiB): IRAM
I (643) spi_flash: detected chip: generic
I (646) spi_flash: flash io: dio
I (651) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (721) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (721) wifi station: Initializing SPIFFS
I (851) wifi station: Partition size: total: 956561, used: 1004
I (851) wifi station: Reading file
I (851) wifi station: Read from file: 'Tree'
data Tree
I (851) wifi station: Reading file
I (861) wifi station: Read from file: 'qqqqqqqq'
data qqqqqqqq
[... 488 lines cut ...]
W (32261) wifi station: Received=

variable stored data =
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data
waiting for  RemoteStopTransaction
 data

vinothkrstatiq, 2022-07-06:
@gabsuren 
from above log after waiting for RemoteStopTransaction
I suppose to receive JSON data 
I highlighted that in bold above, but it is showing nothing 
**W (32261) wifi station: Received=
variable stored data =**

in Arduino same project working fine
in ESP-IDF this problem happening ?
I am continually sending and receiving JSON message from WebSocket

vinothkrstatiq, 2022-07-07:
@gabsuren 
i would suggest you to try send and receive JSON data in websocket client. keep it in infinite loop and check after some interval weather we are able to receive any data from cloud.
example JSON data :
["abc", 123, true, false, null, {
                "version":      "v5.0-dev-3202-ga2d5041492-dirty",
                "cores":        2
        }]

gabsuren, 2022-07-07:
@vinothkrstatiq thank you for the attached info. 
I tried to use internal [esp_websocket_client/examples](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client/examples) as client and [ws_echo_server](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/ws_echo_server) as server, I tried to run it for an hour but connection wasn't drop during that time. 
In order test it you need to configure _esp_websocket_client_  to use _ws_echo_server_ server, please run both of them in separate boards and set 

> CONFIG_WEBSOCKET_URI_FROM_STRING = ws://IP_address_of_esp_board:80/ws

One of my guess it can be related with cloud server, as it drops the connection if there is no data for some time and server waits PONG data when it sends PING request, similar to https://github.com/espressif/esp-idf/issues/8557

You can try the solution which user described, make _esp_websocket_client_send_with_opcode_ public and send PONG message when there is PING, meanwhile we are adding new API to have it public.

If both of suggestions didn't work or you have any questions please let us know.

Thank you

vinothkrstatiq, 2022-07-08:
@gabsuren ,

Thanks for detailed explanation 
as you mentioned problem was from server side 
I found the solution
highly appreciated for your immidiate help

closed 2022-07-08 by vinothkrstatiq

hitecSmartHome, 2024-06-05:
I'm experiencing a similar issue. 
Websocket client just stops.
Sometimes it is after half a day sometimes it is after 5 days!
The rest of the app is working fine but the websocket does not receive any data.