espressif/esp-idf #9224: ESP32 HTTPS + mDNS server

kaizoku-oh, 2022-06-24:
I'm using ESP32 HTTPS server (internal component) and its working fine when I go to https://<IP ADDRESS>/
When I added the mDNS server code I expected the following to work: https://<HOSTNAME>/

here is my mDNS configuration:

```c
static void _start_mdns_server(void)
{
  esp_err_t s32RetVal;
  mdns_txt_item_t tstMdnsTxItems[2] =
  {
    {"device name", "esp32-c3"         },
    {"mac address", "00:00:00:00:00:00"},
  };

  s32RetVal = mdns_init();
  if(ESP_OK == s32RetVal)
  {
    s32RetVal =  mdns_hostname_set("esp32server");
    if(ESP_OK == s32RetVal)
    {
      s32RetVal =  mdns_instance_name_set("mDNS service");
      if(ESP_OK == s32RetVal)
      {
        s32RetVal = mdns_service_add(NULL, "_http", "_tcp", 80, tstMdnsTxItems, 2);
        if(ESP_OK == s32RetVal)
        {
          ESP_LOGI(TAG, "mDNS server started successfully");
        }
        else
        {
          ESP_LOGE(TAG, "Failed to add mDNS service (%d)", s32RetVal);
        }
      }
      else
      {
        ESP_LOGE(TAG, "Failed to set mDNS instance name (%d)", s32RetVal);
      }
    }
    else
    {
      ESP_LOGE(TAG, "Failed to set mDNS hostname (%d)", s32RetVal);
    }
  }
  else
  {
    ESP_LOGE(TAG, "Failed to initialize mDNS server (%d)", s32RetVal);
  }
}
```

My question is: is there something wrong in my config like for example the port 80 needs to change to 443...

nopnop2002, 2022-06-26:
@kaizoku-oh 

I think it simply works with:

```
static void _start_mdns_server(void)
{
  ESP_ERROR_CHECK( mdns_init() );
  ESP_ERROR_CHECK( mdns_hostname_set("esp32server") );
}
```

kaizoku-oh, 2022-06-27:
It still doesn't work here is my code for starting HTTPS server:

```c
static void _start_https_server(void)
{
  esp_err_t s32RetVal;
  httpd_handle_t pvServer;
  httpd_ssl_config_t stConf = HTTPD_SSL_CONFIG_DEFAULT();

  stConf.cacert_pem = ca_cert_pem_start;
  stConf.cacert_len = ca_cert_pem_end - ca_cert_pem_start;
  stConf.prvtkey_pem = prvtkey_pem_start;
  stConf.prvtkey_len = prvtkey_pem_end - prvtkey_pem_start;
#ifndef USE_HTTPS
  stConf.transport_mode = HTTPD_SSL_TRANSPORT_INSECURE;
#endif /* USE_HTTPS */
  s32RetVal = httpd_ssl_start(&pvServer, &stConf);
  if(ESP_OK == s32RetVal)
  {
    s32RetVal = httpd_register_uri_handler(pvServer, &stRootEndpoint);
    if(ESP_OK == s32RetVal)
    {
      ESP_LOGI(TAG, "HTTPS server started successfully");
    }
    else
    {
      ESP_LOGE(TAG, "Failed to register web server URI handler");
    }
  }
  else
  {
    ESP_LOGE(TAG, "Failed to start web server");
  }
}

```

nopnop2002, 2022-06-27:
I don't think the reason it doesn't work has anything to do with mDNS.

Will it work correctly if you specify an IP address?