espressif/esp-idf #9225: Guru Meditation Error (Load Prohibited) when calling esp_eth_driver_install

Henrikastro, 2022-06-24:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version : [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.3.3-30-g5be64c3111
- Build System:         [idf.py]
- Compiler version : (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

So I'm trying to setup my ESP32 to run with an W5500 instead of using an regular USB cable. I've managed to assign an static IP to my ESP tweaking the [basic Ethernet example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic) and I was able to ping it just fine. Now I'm trying to make this piece of code run on GrblHAL so I can control my CNC using the Ethernet cable. I managed to put it in the project and call it just fine but when it gets to the `ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle)); ` I get the Guru Meditation Error. Here's the log:

### Error Log

> --- idf_monitor on \\.\COM3 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1548
> ho 0 tail 12 room 4
> load:0x40078000,len:12820
> load:0x40080400,len:3020
> 0x40080400: _init at ??:?
> 
> entry 0x400805e4
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandle
> d.
> 
> Core  1 register dump:
> PC      : 0x4011a553  PS      : 0x00060633  A0      : 0x800d5424  A1      : 0x3f
> fb6540
> 0x4011a553: esp_intr_get_cpu at C:/Users/User/Desktop/GrblHAL2/esp-idf/component
> s/esp_system/intr_alloc.c:712
> 
> A2      : 0x00000000  A3      : 0x0000000f  A4      : 0x00000004  A5      : 0x3f
> fb9c60
> A6      : 0x00000017  A7      : 0x00000100  A8      : 0x00000000  A9      : 0x3f
> fb6510
> A10     : 0x3ffbdd94  A11     : 0x00000004  A12     : 0x00000004  A13     : 0x00
> 060623
> A14     : 0x00000001  A15     : 0x3ffaed3c  SAR     : 0x0000001c  EXCCAUSE: 0x00
> 00001c
> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00
> 000000
> 
> Backtrace:0x4011a550:0x3ffb6540 0x400d5421:0x3ffb6560 0x400e849e:0x3ffb6580 0x40
> 0e7e8c:0x3ffb65b0 0x400e5303:0x3ffb65e0 0x400e3dc5:0x3ffb66c0 0x400e3a31:0x3ffb6
> 720 0x400d6536:0x3ffb6780 0x400d639f:0x3ffb67a0
> 0x4011a550: esp_intr_get_cpu at C:/Users/User/Desktop/GrblHAL2/esp-idf/component
> s/esp_system/intr_alloc.c:711
> 
> 0x400d5421: gpio_isr_handler_add at C:/Users/User/Desktop/GrblHAL2/esp-idf/compo
> nents/driver/gpio.c:484
> 
> 0x400e849e: emac_w5500_init at C:/Users/User/Desktop/GrblHAL2/esp-idf/components
> /esp_eth/src/esp_eth_mac_w5500.c:607
> 
> 0x400e7e8c: esp_eth_driver_install at C:/Users/User/Desktop/GrblHAL2/esp-idf/com
> ponents/esp_eth/src/esp_eth.c:204 (discriminator 2)
> 
> 0x400e5303: enet_start at c:\users\user\desktop\grblhal2\esp-idf\esp32\build/../
> main/enet.c:307 (discriminator 2)
> 
> 0x400e3dc5: settings_changed at c:\users\user\desktop\grblhal2\esp-idf\esp32\bui
> ld/../main/driver.c:1433
>  (inlined by) settings_changed at c:\users\user\desktop\grblhal2\esp-idf\esp32\b
> uild/../main/driver.c:1416
> 
> 0x400e3a31: driver_setup at c:\users\user\desktop\grblhal2\esp-idf\esp32\build/.
> ./main/driver.c:1906
> 
> 0x400d6536: grbl_enter at c:\users\user\desktop\grblhal2\esp-idf\esp32\build/../
> main/grbl/grbllib.c:185
> 
> 0x400d639f: vGrblTask at c:\users\user\desktop\grblhal2\esp-idf\esp32\build/../m
> ain/main.c:50
> 
> 
> 
> ELF file SHA256: d4b99a96558a4446
> 
> CPU halted.
> 
### Code snippet

```cpp
// Initialize TCP/IP network interface (should be called only once in application)
    ESP_ERROR_CHECK(esp_netif_init());
    // Create default event loop that running in background
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg);
    // Set default handlers to process TCP/IP stuffs
    ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));
    // Register user defined event handers
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = 1;
    phy_config.reset_gpio_num = -1;

    char* ip= "192.168.5.241";
    char* gateway = "192.168.5.1";
    char* netmask = "255.255.255.0";
    esp_netif_ip_info_t info_t;
    memset(&info_t, 0, sizeof(esp_netif_ip_info_t));

    if (eth_netif)
    {
        ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif));
        info_t.ip.addr = esp_ip4addr_aton((const char *)ip);
        info_t.netmask.addr = esp_ip4addr_aton((const char *)netmask);
        info_t.gw.addr = esp_ip4addr_aton((const char *)gateway);
        esp_netif_set_ip_info(eth_netif, &info_t);
    }
    gpio_install_isr_service(0);
    spi_device_handle_t spi_handle = NULL;
    spi_bus_config_t buscfg = {
        .miso_io_num = 19,
        .mosi_io_num = 23,
        .sclk_io_num = 18,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(1, &buscfg, 1));
    spi_device_interface_config_t devcfg = {
        .command_bits = 16, // Actually it's the address phase in W5500 SPI frame
        .address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
        .mode = 0,
        .clock_speed_hz = 18 * 1000 * 1000,
        .spics_io_num = 5,
        .queue_size = 20
    };
    ESP_ERROR_CHECK(spi_bus_add_device(1, &devcfg, &spi_handle));
    /* w5500 ethernet driver is based on spi driver */
    eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
    w5500_config.int_gpio_num = 4;
    esp_eth_mac_t *mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
    esp_eth_phy_t *phy = esp_eth_phy_new_w5500(&phy_config);
    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
    esp_eth_handle_t eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));
    ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_MAC_ADDR, (uint8_t[]) {
        0x02, 0x00, 0x00, 0x12, 0x34, 0x56
    }));
    /* attach Ethernet driver to TCP/IP stack */
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
    /* start Ethernet driver state machine */
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
```

Henrikastro, 2022-06-24:
It was the `gpio_install_isr_service(0);` that was causing the error. Turns out already uses a similar function and that was the cause of the error.

closed 2022-06-24 by Henrikastro

KaeLL, 2022-06-24:
> It was the gpio_install_isr_service(0)

Are you sure? That function checks if it has been called previously, which means either it's not the source of your problem, or there's a bug in the code.

Henrikastro, 2022-06-25:
> Are you sure? That function checks if it has been called previously, which means either it's not the source of your problem, or there's a bug in IDF.

Well, commenting it out made the Guru Meditation error stop ocurring. Maye the IDF one was being called first and the GrblHAL was causing the error afterwards? By the way, `gpio_isr_register()` is the one that HAL calls.