espressif/esp-idf #9229: pthread_rwlock_t implementation is missing try-lock functions

Spacechild1, 2022-06-24:
ESP-IDF has recently added support for `pthread_rwlock_t` (https://github.com/espressif/esp-idf/commit/32e34447018539522c8ffae612408ef34d395783), but strangely the try-lock functions `pthread_wrlock_tryrdlock` and `pthread_wrlock_trywrlock` are missing... I'm wondering why they have been omitted?

0xjakob, 2022-06-27:
@Spacechild1 

> I'm wondering why they have been omitted?

They simply haven't been implemented. The original feature request came from the Rust-community who needed only the simple functionality. I'll take a look at the code within the next few days.

Spacechild1, 2022-06-27:
Thanks!

> to see if the implementation would be easy.

They are pretty easy to implement: instead of waiting on the condition variable, you would just unlock the resource mutex and return `EBUSY`. I can also make a PR if necessary.

svenstaro, 2023-01-19:
Has there been any progress on this? I think this functionality would be pretty neat to have. Speaking from a Rust perspective, it's actually required to make full use of the tracing ecosystem.

elotom, 2023-02-02:
I also encountered this problem. It would be nice to have this for the Rust ecosystem.

closed 2023-04-18 by espressif-bot

0xjakob, 2023-04-19:
The try-lock functions have been merged into master. Feel free to take a look.

svenstaro, 2023-10-30:
Looks like https://github.com/espressif/esp-idf/commit/e7312cccea37c12bea6b9a0e6ba5c9f39762391a is still not in any release. What's the policy here? What future release would this be contained in?

0xjakob, 2023-10-31:
Hi @svenstaro, it has been merged to master. This means it will go into the next release (v5.2), as well as all the following releases.