espressif/esp-idf #9231: MQTT connection reset by peer over GSM PPPoS

inimin, 2022-06-25:
## Environment
- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.4.1-405-g6c5fb29c2c
- Build System:    [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]
- GSM Module : MC60

## Problem Description
There is problem with MQTT over GSM PPPoS, since I used MQTT over wifi without any problem. I want to use GSM MC60 as 2G modem connection and write a program base on PPPOS_CLIENT example; The MQTT connection get closed after a period of time without any reason when using GSM and pppos.

Tcpdump on mqtt server also show that 
- "Syn" packet from client ,
- "Syn-Ack" packet from server,
- "Ack" packet from client  
- "RST" packet from client
(Client is ESP board)

### Expected Behavior

I expect establishing MQTT connection over GSM with PPPoS.

### Actual Behavior

Connection get closed without any reason over GSM with PPPoS but it work fine when connecting over WIFI.Also another client over smartphone connect to server without any problem. 

### Steps to reproduce

I use pppos_client example with Quectel MC60 PPPoS startup flowchart which set power key pin to start module then send to pppos mode. Event in pppos_client example the connection is not established.

### Code to reproduce this issue
 
I Use pppos_client example, I just add the startup power key followed by MC60 Quectel PPP flowchart.

## Debug Logs

```

I (17846) esp-netif_lwip-ppp: Connected
I (17846) esp-netif_lwip-ppp: Name Server1: *****
I (17856) esp-netif_lwip-ppp: Name Server2: *****
local  IP address *****
D (17866) event: running post IP_EVENT:6 with handler 0x400d85c0 and context 0x3ffb9fd4 on loop 0x3ffb9084
0x400d85c0: on_ip_event at /*****/workspace/esp-idf-piod/build/../components/esp-idf-gsm/src/gsm.c:142

remote IP address 192.168.254.254
D (17876) esp_gsm: IP event! 6
primary   DNS address *****
I (17876) esp_gsm: Modem Connect to PPP Server
secondary DNS address *****
I (17886) esp_gsm: ~~~~~~~~~~~~~~
ppp phase changed[1]: phase=10
I (17896) esp_gsm: IP          : *****
D (17896) esp-netif_lwip-ppp: Phase Running
I (17906) esp_gsm: Netmask     : 255.255.255.255
pbuf_free(0x3ffbd7a8)
I (17916) esp_gsm: Gateway     : 192.168.254.254
pbuf_free: deallocating 0x3ffbd7a8
I (17916) esp_gsm: Name Server1: *****
pbuf_free(0x3ffbe3cc)
I (17926) esp_gsm: Name Server2: *****
pbuf_free: deallocating 0x3ffbe3cc
I (17936) esp_gsm: ~~~~~~~~~~~~~~
I (17946) esp_gsm: GOT ip event!!!
D (17946) event: running post IP_EVENT:6 with handler 0x400e307c and context 0x3ffbd600 on loop 0x3ffb9084
0x400e307c: esp_netif_action_connected at /*****/esp/esp-idf/components/esp_netif/esp_netif_handlers.c:44

D (17956) esp_netif_handlers: esp_netif action connected with netif0x3ffbbf7c from event_id=6
[... 964 lines cut ...]
|    30841      |000|       0   | (id, flags, offset)
+-------------------------------+
|  251  |    6  |    0xcd77     | (ttl, proto, chksum)
+-------------------------------+
|    *****  | (src)
+-------------------------------+
|   *****  | (dest)
+-------------------------------+
ip4_input: p->len 40 p->tot_len 40
pbuf_remove_header: old 0x3ffc0bee new 0x3ffc0c02 (20)
pbuf_remove_header: old 0x3ffc0c02 new 0x3ffc0c16 (20)
tcp_pcb_purge
tcp_pcb_purge: data left on ->unacked
pbuf_free(0x3ffb5d38)
pbuf_free: deallocating 0x3ffb5d38
pbuf_free(0x3ffc0bd4)
pbuf_free: deallocating 0x3ffc0bd4
pbuf_free(0x3ffb6318)
pbuf_free: deallocating 0x3ffb6318
E (104676) TRANSPORT_BASE: poll_read select error 104, errno = Connection reset by peer, fd = 54
E (104686) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
D (104696) event: running post MQTT_EVENTS:0 with handler 0x400d7e00 and context 0x3ffbee9c on loop 0x3ffbdc84
0x400d7e00: mqtt_event_handler at /*****/workspace/esp-idf-piod/build/../components/esp-idf-mqtt/src/mqtt.c:199

```

inimin, 2022-06-25:
Sorry, but I found that SIM Card was not charged.

closed 2022-06-25 by inimin

abhi-ratnman, 2022-08-25:
Which example code did you use for this? I can not find an example code for PPPoS in the current examples

StarOfSlytherin, 2022-09-19:
> Which example code did you use for this? I can not find an example code for PPPoS in the current examples

It's available on v4.4 branch. For master branch you have to use it `esp-modem` component.