espressif/esp-idf #9233: ESP32 S3 Serial Console Example doesn't work

SinanAkkoyun, 2022-06-25:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Board:      [TinyS3 Unexpected Maker]
- IDF version latest
- Build System:   [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [WSL2].
- Using an IDE?: [Yes] VSCode and WSL2
- Power Supply:         [USB]

## Problem Description

The advanced serial console example doesn't work with an ESP32-S3. It totally works with an ESP32-PICO-D4 in the exact same environment

### Expected Behavior
S3 advanced console example should work

### Actual Behavior
Serial Connection dies right when the command prompt should appear.

### Steps to reproduce
1. Clone the advanced console example
2. Set Target to ESP32 S3 Internal UART
3. flash the advanced example onto an S3 board and test via interal UART

### Code to reproduce this issue
The stock advanced console example code.

Thank you very much for all the awesome examples. I hope this will soon work on S3!

igrr, 2022-06-26:
If you are connecting the USB cable to tinyS3 board's USB port, this means that you are most likely using the built-in USB port, not the UART ([schematic](https://esp32s3.com/files/schematic-tinys3.pdf)).
You can use the serial console if you attach an external USB-Serial converter to the TX/RX/GND pins of the board, though.

If that doesn't help, could you please explain the "actual behavior" in a bit more detail? What does "dies" mean in this context?

SinanAkkoyun, 2022-06-26:
Hi igrr, thanks for replying!
Yes, exactly, I am then using the internal USB controller. But given that I can use the internal USB just fine for flashing and normal serial IO, why shouldn't it work with the console example?

igrr, 2022-06-26:
The example you are trying to compile (console/advanced) is written to work specifically with the UART driver. It's not impossible to modify the example to work with USB, however supporting both in the same example will probably make the code too complex. Perhaps we should rename it to `advanced_uart` to make the purpose of the example clear.

There is another example, console/basic, which does support 3 interfaces: UART, USB_OTG and USB_SERIAL_JTAG. You could try that one, see its README for details. You probably need to use USB_SERIAL_JTAG peripheral since USB_CDC console isn't supported on ESP32-S3 yet.

aircable, 2023-10-20:
I have been trying to get the system/console/advanced_usb_cdc to work and I can't.
Flashing on the internal port and monitor.
The output does not appear and the input gives: Writing to serial is timing out.

This is the unchanged example on the S3, USB only.
Any idea what's wrong?