espressif/esp-idf #9234: Not able to include esp_websocket_client library inside component

vinothkannan369, 2022-06-26:
I am using Espressif-IDE Version: 2.5.0
Operating System: windows 10
Java Runtime Version: 11.0.14.1+10-LTS
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.5.0.202206080313
ESP-IDF v4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.10.4

I have created component in the project, when I include esp_websocket_client.h inside the component   it shows error message like Unresolved inclusion: "cJSON.h"
can anyone help me here?

vinothkannan369, 2022-06-27:
It is not only above library even if I include cJSON also same problem comes like
Unresolved inclusion: "cJSON.h"
I am trying to include these library's in component

gabsuren, 2022-07-04:
@vinothkannan369 thank you for your question.

Not sure if it is the case, but starting from `ESP_IDF V5.0`, `esp_websocket_client` library is moved to [esp-protocols](https://github.com/espressif/esp-protocols). In order to add the component please run
`idf.py add-dependency espressif/esp_websocket_client==0.0.1`

As you mentioned you use `ESP-IDF v4.4`, please check if `esp_websocket_client.h` is available in `$ESP-IDF/components/esp_websocket_client/include/`, if it is there then the problem is not related with upper solution, let us know  and we will discuss the next steps

vinothkannan369, 2022-07-05:
@gabsuren 
Thanks for the help

closed 2022-07-05 by vinothkannan369