espressif/esp-idf #9237: SDMMC returns card status indicates an error after write operation error message after read/write and erase operations

muratdemirtas, 2022-06-26:
## Environment
- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used: ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    // v4.4
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB external 5V|]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
TEST ok, without warning message,
### Actual Behavior
TEST passed but warning message exists.
 
### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

uint8_t rx_buffer[512];
uint8_t tx_buffer[512];
void app_test(sdmmc_card_t *sdCard) {
    sdCardInstance = sdCard;
    
    for(int i = 0 ; i < 20000; i++)
    {
        memset(rx_buffer,0,512);
        memset(tx_buffer,0,512);
        for(int i = 0; i < 512; i++)
            rx_buffer[i] = i%255;
        sdmmc_write_sectors(sdCardInstance, rx_buffer,i, 1 );
        sdmmc_read_sectors(sdCardInstance, tx_buffer, i, 1);
        if(memcmp(tx_buffer, rx_buffer, 512) == 0){
            printf("TEST OK, index %d\n", i);
        }else
            printf("TEST FAIL\n");
    }
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.
[15:19:05:260] TEST OK, index 1500␍␊
[15:19:05:260] <0x1b>[0;31mE (19473) sdmmc_cmd: sdmmc_write_sectors_dma: card status indicates an error after write operation: r2=0x00ff<0x1b>[0m␍␊
[15:19:05:313] TEST OK, index 1501␍␊
[15:19:05:313] <0x1b>[0;31mE (19483) sdmmc_cmd: sdmmc_write_sectors_dma: card status indicates an error after write operation: r2=0x00ff<0x1b>[0m␍␊
[15:19:05:313] TEST OK, index 1502␍␊
[15:19:05:313] <0x1b>[0;31mE (19503) sdmmc_cmd: sdmmc_write_sectors_dma: card status indicates an error after write operation: r2=0x00ff<0x1b>[0m␍␊
[15:19:05:313] TEST OK, index 1503␍␊
[15:19:05:452] <0x1b>[0;31mE (19623) sdmmc_cmd: sdmmc_write_sectors_dma: card status indicates an error after write operation: r2=0x00ff<0x1b>[0m␍␊
[15:19:05:452] TEST OK, index 1504␍␊
[15:19:05:452] <0x1b>[0;31mE (19633) sdmmc_cmd: sdmmc_write_sectors_dma: card status indicates an error after write operation: r2=0x00ff<0x1b>[0m␍␊
[15:19:05:452] TEST OK, index 1505␍␊
[15:19:05:452] <0x1b>[0;31mE (19643) sdmmc_cmd: sdmmc_write_sectors_dma: card status indicates an error after write operation: r2=0x00ff<0x1b>[0m␍␊
[15:19:05:452] TEST OK, index 1506␍␊

igrr, 2022-06-27:
@muratdemirtas thanks for reporting the issue! Do you see this issue with every SD card, or just some specific one? If it happens with a specific SD card, do you mind writing which manufacturer and card model that is?

muratdemirtas, 2022-06-27:
@igrr I will perform the test asap with new sd cards. thanks.

muratdemirtas, 2022-06-27:
[15:57:12:231] rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)␍␊
[15:57:12:231] configsip: 0, SPIWP:0xee␍␊
[15:57:12:231] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[15:57:12:231] mode:DIO, clock div:2␍␊
[15:57:12:231] load:0x3fff0030,len:6936␍␊
[15:57:12:231] load:0x40078000,len:15356␍␊
[15:57:12:231] ho 0 tail 12 room 4␍␊
[15:57:12:231] load:0x40080400,len:3840␍␊
[15:57:12:231] entry 0x4008064c␍␊
[15:57:12:231] <0x1b>[0;32mI (29) boot: ESP-IDF v5.0-dev-3654-gc2ccc383da-dirty 2nd stage bootloader<0x1b>[0m␍␊
[15:57:12:231] <0x1b>[0;32mI (29) boot: compile time 15:56:13<0x1b>[0m␍␊
[15:57:12:231] <0x1b>[0;32mI (30) boot: chip revision: 3<0x1b>[0m␍␊
[15:57:12:231] <0x1b>[0;32mI (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (41) boot.esp32: SPI Speed      : 40MHz<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (46) boot.esp32: SPI Mode       : DIO<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (50) boot.esp32: SPI Flash Size : 2MB<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (55) boot: Enabling RNG early entropy source...<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (60) boot: Partition Table:<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000<0x1b>[0m␍␊
[15:57:12:274] <0x1b>[0;32mI (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000<0x1b>[0m␍␊
[15:57:12:323] <0x1b>[0;32mI (86) boot:  2 factory          factory app      00 00 00010000 00100000<0x1b>[0m␍␊
[15:57:12:323] <0x1b>[0;32mI (94) boot: End of partition table<0x1b>[0m␍␊
[15:57:12:323] <0x1b>[0;32mI (98) boot_comm: chip revision: 3, min. application chip revision: 0<0x1b>[0m␍␊
[15:57:12:323] <0x1b>[0;32mI (105) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b87ch ( 47228) map<0x1b>[0m␍␊
[15:57:12:323] <0x1b>[0;32mI (130) esp_image: segment 1: paddr=0001b8a4 vaddr=3ffb0000 size=027b8h ( 10168) load<0x1b>[0m␍␊
[15:57:12:323] <0x1b>[0;32mI (135) esp_image: segment 2: paddr=0001e064 vaddr=40080000 size=01fb4h (  8116) load<0x1b>[0m␍␊
[15:57:12:355] <0x1b>[0;32mI (140) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1c868h (116840) map<0x1b>[0m␍␊
[15:57:12:392] <0x1b>[0;32mI (187) esp_image: segment 4: paddr=0003c890 vaddr=40081fb4 size=0b194h ( 45460) load<0x1b>[0m␍␊
[15:57:12:392] <0x1b>[0;32mI (206) esp_image: segment 5: paddr=00047a2c vaddr=50000000 size=00010h (    16) load<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (213) boot: Loaded app from partition at offset 0x10000<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (213) boot: Disabling RNG early entropy source...<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (226) cpu_start: Pro cpu up.<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (226) cpu_start: Starting app cpu, entry point is 0x40081160<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (0) cpu_start: App cpu up.<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (240) cpu_start: Pro cpu start user code<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (240) cpu_start: cpu freq: 160000000 Hz<0x1b>[0m␍␊
[15:57:12:437] <0x1b>[0;32mI (240) cpu_start: Application information:<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (245) cpu_start: Project name:     sd_card<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (250) cpu_start: App version:      v5.0-dev-3654-gc2ccc383da-dirty<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (257) cpu_start: Compile time:     Jun 27 2022 15:54:37<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (263) cpu_start: ELF file SHA256:  e84b254811d613ed...<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (269) cpu_start: ESP-IDF:          v5.0-dev-3654-gc2ccc383da-dirty<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (276) heap_init: Initializing. RAM available for dynamic allocation:<0x1b>[0m␍␊
[15:57:12:480] <0x1b>[0;32mI (284) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;32mI (290) heap_init: At 3FFB34D8 len 0002CB28 (178 KiB): DRAM<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;32mI (296) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;32mI (302) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;32mI (309) heap_init: At 4008D148 len 00012EB8 (75 KiB): IRAM<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;32mI (316) spi_flash: detected chip: generic<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;32mI (319) spi_flash: flash io: dio<0x1b>[0m␍␊
[15:57:12:523] <0x1b>[0;33mW (323) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.<0x1b>[0m␍␊
[15:57:12:566] <0x1b>[0;32mI (337) cpu_start: Starting scheduler on PRO CPU.<0x1b>[0m␍␊
[15:57:12:566] <0x1b>[0;32mI (0) cpu_start: Starting scheduler on APP CPU.<0x1b>[0m␍␊
[15:57:12:566] <0x1b>[0;32mI (349) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 <0x1b>[0m␍␊
[15:57:12:566] ESP host init ok␍␊
[15:57:12:604] <0x1b>[0;32mI (399) sdspi_transaction: cmd=52, R1 response: command not supported<0x1b>[0m␍␊
[15:57:12:779] <0x1b>[0;32mI (599) sdspi_transaction: cmd=5, R1 response: command not supported<0x1b>[0m␍␊
[15:57:13:421] Name: SD01G␍␊
[15:57:13:421] Type: SDSC␍␊
[15:57:13:421] Speed: 20 MHz␍␊
[15:57:13:421] Size: 922MB␍␊
[15:57:13:421] CSD: ver=1, sector_size=512, capacity=1888256 read_bl_len=9␍␊
[15:57:13:421] SSR: bus_width=1␍␊
[... cut ...]
[15:57:14:145] TEST OK, index 5␍␊
[15:57:14:273] TEST OK, index 6␍␊
[15:57:14:416] TEST OK, index 7␍␊
[15:57:14:543] TEST OK, index 8␍␊
[15:57:14:671] TEST OK, index 9␍␊
[15:57:14:782] TEST OK, index 10␍␊
[15:57:14:910] TEST OK, index 11␍␊
[15:57:15:037] TEST OK, index 12␍␊
[15:57:15:165] TEST OK, index 13␍␊
[15:57:15:276] TEST OK, index 14␍␊
[15:57:15:404] TEST OK, index 15␍␊
[15:57:15:531] TEST OK, index 16␍␊
[15:57:15:658] TEST OK, index 17␍␊
[15:57:15:770] TEST OK, index 18␍␊
[15:57:15:897] TEST OK, index 19␍␊

muratdemirtas, 2022-06-27:
@igrr thanks problem resolved after fresh installation of current master branch(v5.0). SD card is same.

closed 2022-06-27 by muratdemirtas

igrr, 2022-06-27:
@muratdemirtas Interesting, thanks for the update!

Indeed, this error message was added recently in the master branch. The code in v4.4 release didn't check the full card status after write operation at all. So it's entirely possible that the logic for this new error message is not correct in some cases. However, so far we haven't met an SD card which exhibited this issue.

If you notice this issue again, please feel free to reopen!