espressif/esp-idf #9238: ESP32-S3: `assert sch_arb.c 890, param 00000000 00000000` when using `ble_mesh_wifi_coexist`

mrninhvn, 2022-06-27:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-S3-WROOM-1]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

Using `ble_mesh_wifi_coexist` example.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

JiaLWang, 2022-07-05:
Hi @mrninhvn ，

Could you please provide the detailed error log?
And steps to reproduce it.

mrninhvn, 2022-07-05:
> 

> Hi @mrninhvn ，
> 
> Could you please provide the detailed error log? And steps to reproduce it.

I'm using `[ble_mesh_wifi_coexist](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_wifi_coexist)` example with ESP32-S3-WROOM-1, I didn't edit anything, just changed the target, the above error appeared randomly.

Cleidiana, 2022-07-05:
I have the same problem.

JiaLWang, 2022-07-06:
Thanks for your information.
We have located the problem, and we will fix and update it ASAP.

Weijian-Espressif, 2022-07-12:
[esp32s3_lib.zip](https://github.com/espressif/esp-idf/files/9089321/esp32s3_lib.zip)
Please test the attached S3 lib and provide the log file if you still have problems.

Weijian-Espressif, 2022-07-12:
@mrninhvn @Cleidiana

mrninhvn, 2022-07-12:
Thank you @Weijian-Espressif.
Can you tell me where to use this file?

mrninhvn, 2022-07-12:
Thank you @Weijian-Espressif, It's ok now (y)

Alvin1Zhang, 2023-01-12:
Thanks for reporting, sorry for slow turnaround, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/41839b69bda176cc29a2467b8d759f53cc31f77a, feel free to reopen.

closed 2023-01-12 by Alvin1Zhang