espressif/esp-idf #9240: sdmmc_read_sectors reads 512 byte sector size in 40ms - 50ms

muratdemirtas, 2022-06-27:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

/**
 * LittleFS disk io read function
 * @param c littlefs config structure
 * @param block block address
 * @param off offset address
 * @param buffer read buffer
 * @param size read buffer size
 * @return return ok on success
 */
static int lfs_deskio_read(const struct lfs_config *c,
                           lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size)
{
#ifdef CONFIG_LITTLE_FS_IO_DEBUG
    uint32_t current_tick = xTaskGetTickCount();
    printf("[READ]block addr %d, offset %d, size %d\n",block, off,size);
#endif
    esp_err_t ret = sdmmc_read_sectors(sdCardInstance, buffer, block, 1);
    if(ret != ESP_OK){
#ifdef CONFIG_LITTLE_FS_IO_DEBUG
        printf("[READ]Error on read, code: %d\n", ret);
#endif
        return LFS_ERR_IO;
    }
#ifdef CONFIG_LITTLE_FS_IO_DEBUG
    uint32_t elapsed = xTaskGetTickCount() - current_tick;
    elapsed = elapsed * portTICK_PERIOD_MS;
    printf("[READ]512 byte reading in %d ms\n", elapsed);
#endif
    return LFS_ERR_OK;
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
[17:53:28:418] [READ]512 byte reading time 50 ms␍␊
[17:53:28:466] [READ]512 byte reading time 40 ms␍␊
[17:53:28:753] [READ]512 byte reading time 40 ms␍␊
[17:53:28:785] [READ]512 byte reading time 40 ms␍␊
[17:53:28:833] [READ]512 byte reading time 40 ms␍␊
[17:53:29:025] [READ]512 byte reading time 50 ms␍␊
[17:53:29:152] [READ]512 byte reading time 40 ms␍␊
[17:53:29:360] [READ]512 byte reading time 40 ms␍␊
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

muratdemirtas, 2022-06-27:
<script src="https://gist.github.com/muratdemirtas/bc7039a2bbb0bf54488a9a724339720d.js"></script>

muratdemirtas, 2022-06-27:
Changed SD card frequency to 40MHZ but no luck :(

[18:37:30:266] Name: SD␍␊
[18:37:30:266] Type: SDHC/SDXC␍␊
[18:37:30:266] Speed: 40 MHz␍␊
[18:37:30:266] Size: 14922MB␍␊
[18:37:30:266] CSD: ver=2, sector_size=512, capacity=30560256 read_bl_len=9␍␊
[18:37:30:266] SSR: bus_width=1␍␊
[18:37:30:266] SD Card block size 512, read len 9, capacity 30560256␍␊
[18:37:30:282] [READ]512 byte reading in 40 ms␍␊
[18:37:30:330] [READ]512 byte reading in 50 ms␍␊
[18:37:30:378] [READ]512 byte reading in 40 ms␍␊
[18:37:30:409] [READ]512 byte reading in 40 ms␍␊

muratdemirtas, 2022-06-27:
Allocated dma buffers for sd card but no luck:(
void* tmp_buf = heap_caps_malloc(512, MALLOC_CAP_DMA);

[18:48:37:310] [READ]512 byte reading in 40 ms␍␊
[18:48:37:358] dma capable␍␊
[18:48:37:358] [READ]512 byte reading in 40 ms␍␊
[18:48:37:646] dma capable␍␊

[image]

muratdemirtas, 2022-06-27:
#3249

muratdemirtas, 2022-06-27:
```
[19:31:47:979] Name: SD␍␊
[19:31:47:979] Type: SDHC/SDXC␍␊
[19:31:47:979] Speed: 40 MHz␍␊
[19:31:47:979] Size: 14922MB␍␊
[19:31:48:022] CSD: ver=2, sector_size=512, capacity=30560256 read_bl_len=9␍␊
[19:31:48:022] SSR: bus_width=1␍␊
[19:31:48:022] SD Card block size 512, read len 9, capacity 30560256␍␊
[19:31:48:022] <0x1b>[0;32mI (1293) example: SD card info:<0x1b>[0m␍␊
[19:31:48:022] <0x1b>[0;32mI (1303) example: ⇥	Bus width (log2): 0<0x1b>[0m␍␊
[19:31:48:022] <0x1b>[0;32mI (1303) example: ⇥	Freq (kHz): 40000<0x1b>[0m␍␊
[19:31:48:022] <0x1b>[0;32mI (1303) example: ⇥	DDR: 0<0x1b>[0m␍␊
[19:31:48:022] <0x1b>[0;32mI (1313) example: ⇥	CID: Date 338, MFG_ID 18, Name SD, OEM ID 13398, Rev 0, Serial 3271<0x1b>[0m␍␊
[19:31:48:022] <0x1b>[0;32mI (1323) example: ⇥	CSD: Capacity 30560256, Card Common Class 1461, CSD version 1, MMC version 0, read block len 9, sector size 512, tr speed 50000000<0x1b>[0m␍␊
[19:31:48:070] <0x1b>[0;32mI (1333) example: ⇥	CSD: Ease mem state 0, Power class 0, Revision 0, Sec feature 0<0x1b>[0m␍␊
[19:31:48:070] <0x1b>[0;32mI (1343) example: ⇥	SCR: bus width 5, erase mem state 1, reserved 0, rsvd_mnf 0, sd_spec 2<0x1b>[0m␍␊
[19:31:48:070] <0x1b>[0;32mI (1353) example: ⇥	SSR: cur_bus_width 0, discard_support 0, fule_support 0, reserved 0<0x1b>[0m␍␊
```

muratdemirtas, 2022-06-28:
i have ported littlefs over sd card but i have to improve that read speed.

example code for anyone.

https://github.com/muratdemirtas/sqlite3-esp32-littlefs-sdcard-sdspi

muratdemirtas, 2022-06-29:
in esp-idf-4.4 everything is ok.
`21:02:26:874] [SQLite3]File position set ok, pos 2227200␍␊
[21:02:26:874] [SQLite3]Current offset 2227200, required amount 512␍␊
[21:02:26:939] [SQLite3]Readed 512 bytes␍␊
[21:02:26:939] esp32_Read: 3r JanStore.db 512 512 OK␍␊
[21:02:26:939] esp32_Read: 1r JanStore.db 512 2227712[2227712] ␍␊
[21:02:26:939] [SQLite3]File position set ok, pos 2227712␍␊
[21:02:26:939] [SQLite3]Current offset 2227712, required amount 512␍␊
[21:02:26:939] [SQLite3]Readed 512 bytes␍␊
[21:02:26:939] esp32_Read: 3r JanStore.db 512 512 OK␍␊
[21:02:26:939] esp32_Read: 1r JanStore.db 512 2228224[2228224] ␍␊
[21:02:26:939] [SQLite3]File position set ok, pos 2228224␍␊
[21:02:26:939] [SQLite3]Current offset 2228224, required amount 512␍␊
[21:02:26:939] [SQLite3]Readed 512 bytes␍␊
[21:02:26:939] esp32_Read: 3r JanStore.db 512 512 OK␍␊
[21:02:26:939] esp32_Read: 1r JanStore.db 512 2228736[2228736] ␍␊
[21:02:26:987] [SQLite3]File position set ok, pos 2228736␍␊
[21:02:26:987] [SQLite3]Current offset 2228736, required amount 512␍␊
[21:02:26:987] [SQLite3]Readed 512 bytes␍␊
[21:02:26:987] esp32_Read: 3r JanStore.db 512 512 OK␍␊
[21:02:26:987] esp32_Read: 1r JanStore.db 512 2229248[2229248] ␍␊
[21:02:26:987] [SQLite3]File position set ok, pos 2229248␍␊
[21:02:26:987] [SQLite3]Current offset 2229248, required amount 512␍␊
[21:02:26:987] [SQLite3]Readed 512 bytes␍␊`

muratdemirtas, 2022-06-30:
it was a spi ram allocation issue. The mmc driver using dma memory for communication with sd card. If your device allocates dma from spi ram then the esp first creating a memory block in spi ram( THIS IS THE ISSUE). 2x spi communication makes communication slower.  Thanks.

closed 2022-06-30 by muratdemirtas

xdevelnet, 2023-03-17:
> it was a spi ram allocation issue. The mmc driver using dma memory for communication with sd card. If your device allocates dma from spi ram then the esp first creating a memory block in spi ram( THIS IS THE ISSUE) which 2x spi communication makes communication slower. Thanks.

So how exactly you solved this issue? I assume that you somehow disabled using memory block in spi ram?