espressif/esp-idf #9241: WiFi and ble run for a period of time, and rwble.c 261 error

xueren-git, 2022-06-28:
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version DevKitC: v1
- Module or chip used:  ESP32-C3
- IDF version : esp-idf-v4.4.1
- Build System:   idf.py
- Compiler version:  (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: No
- Power Supply:         USB|external 3.3V|

## Problem Description

Run ble gattc and wifi sta mode, Bluetooth is scanning all the time. A socket connects to the server and sends a packet of data every 10 seconds。

### Expected Behavior

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs

I (2022-06-28 07:37:21.287) Health: recv size=8
I (2022-06-28 07:37:31.277) Health: send packet (keep alive)
I (2022-06-28 07:37:31.278) Health: send packet size=8
I (2022-06-28 07:37:31.278) Health: send packet end 1
I (2022-06-28 07:37:31.310) Health: recv size=8
I (2022-06-28 07:37:41.277) Health: send packet (keep alive)
I (2022-06-28 07:37:41.278) Health: send packet size=8
I (2022-06-28 07:37:41.279) Health: send packet end 1
I (2022-06-28 07:37:41.327) Health: recv size=8
I (2022-06-28 07:37:51.277) Health: send packet (keep alive)
I (2022-06-28 07:37:51.278) Health: send packet size=8
I (2022-06-28 07:37:51.279) Health: send packet end 1
I (2022-06-28 07:37:51.319) Health: recv size=8
DIAG0 120be3a0
DIAG1 10000000
BB DIAG0: 000a4004
                            BB DIAG1: bc002602
                            BB DIAG2: 00000000
                            BB DIAG3: 8e89bed6
                            BB DIAG4: 00000000
                            BB DIAG5: 00000000
assert rwble.c 261, param 00020000 00000000
E (12800280) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (12800280) task_wdt:  - IDLE (CPU 0)
E (12800280) task_wdt: Tasks currently running:
E (12800280) task_wdt: CPU 0: wifi
E (12800280) task_wdt: wifi             X       23      3928    11      0
esp_timer       R       22      2488    2       0
BTU_TASK        R       20      2432    15      0
tiT             R       18      2228    9       0
PLRV            R       12      1508    18      0
BleTask         R       12      3388    16      0
Tmr Svc         R       1       2236    5       0
IDLE            R       0       1588    4       0
sys_evt         B       20      604     10      0
AppTask         B       12      900     8       0
PLSD            B       12      588     17      0
BTC_TASK        B       19      1280    13      0
btController    B       23      1944    12      0
hciT            B       22      1608    14      0
UartTask        B       12      3244    7       0
[... 75 lines cut ...]
3fc93fb0: 0x3fcace9c 0xffffffff 0x3fcace9c 0x3fcace9c 0x00000000 0x3fc93fc8 0xffffffff 0x3fc93fc8
3fc93fd0: 0x3fc93fc8 0x00000000 0x3fc93fdc 0xffffffff 0x3fc93fdc 0x3fc93fdc 0x00000000 0x3fc93ff0
3fc93ff0: 0xffffffff 0x3fc93ff0 0x3fc93ff0 0x00000000 0x3fc94004 0xffffffff 0x3fc94004 0x3fc94004
3fc94010: 0x00000007 0x3fcaa2e8 0xffffffff 0x3fca08bc 0x3fcbb160 0x00000000 0x3fc9402c 0xffffffff
3fc94030: 0x3fc9402c 0x3fc9402c 0x00000001 0x00000789 0x3fcbe920 0x3fc9a678 0x00000000 0x00000000
3fc94050: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94070: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc940b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc940d0: 0x00000000 0x00000000 0x00000001 0x3fc940e0 0xffffffff 0x3fcbe924 0x3fcbe924 0x00000000
3fc940f0: 0x3fc940f4 0xffffffff 0x3fc940f4 0x3fc940f4 0x3fc94038 0x3fc94048 0x3fc940d8 0x3fc94038

ELF file SHA256: d248eaa63d177410
ELF file SHA256: d248eaa63d177410

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038062a
0x4038062a: esp_restart_noos_dig at M:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_system/esp_system.c:44 (discriminator 1)

## Other items if possible

sdkconfig file

nishanth-radja, 2022-06-29:
@xueren-git can you pls give us the following.
1.SDK config
2.Are you using Nimble or bluedroid host?
3.After how long you are seeing the crash?
4.I see that you are mentioned gattc, Is there a active connection and dut continues to perform ble scan? 
5.Can you attach you application which you are using.

xueren-git, 2022-06-29:
Config Dual mode, but my application code only use ble.
Time is no fixed, sometimes more than ten hours, sometimes an hour or two.
esp_ble_gap_start_scanning(0), ble scanning all the time, and no ble device connection.
Wifi config sta mode, an connection working, send several bytes every 10 seconds.

//===Ble init ===
bool BleStart(void)
{
	esp_err_t err;
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();

	LOG_INFO("Start ...");
	gBleScaning = false;

	err = esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
	ERROR_ESP_RETURN(err, false, "Bt mem release ERROR");

	err = esp_bt_controller_init(&bt_cfg);
	ERROR_ESP_RETURN(err, false, "Bt controller init ERROR");

	err = esp_bt_controller_enable(ESP_BT_MODE_BLE);
	ERROR_ESP_RETURN(err, false, "Bt controller enable ERROR");

	err = esp_bluedroid_init();
	ERROR_ESP_RETURN(err, false, "Bluedroid init ERROR");

	err = esp_bluedroid_enable();
	ERROR_ESP_RETURN(err, false, "Bluetooth enable ERROR");

	err = esp_ble_gap_register_callback(esp_gap_cb);
	ERROR_ESP_RETURN(err, false, "Gap callback init ERROR");
	err = esp_ble_gattc_register_callback(esp_gattc_cb);
	ERROR_ESP_RETURN(err, false, "[Gattc] register callback ERROR");

	#ifdef CONFIG_BLE_MUT
	err = esp_ble_gatt_set_local_mtu(CONFIG_BLE_MUT);
	LOG_ERROR_ESP_IF(err, "Set local mtu ERROR");
	LOG_INFO("Set local mut %d", CONFIG_BLE_MUT);
	#endif

	esp_ble_gap_config_local_privacy(true);
	esp_ble_auth_req_t		auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;	 // bonding with peer device after authentication
	esp_ble_io_cap_t		iocap = ESP_IO_CAP_IN;
	uint8_t key_size	= 16;	//the key size should be 7~16 bytes
	uint8_t init_key	= ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	uint8_t rsp_key		= ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	uint8_t oob_support = ESP_BLE_OOB_DISABLE;
	esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));

	if(gBleTask == INVALID_TASK)
	{
		gBleTask = task_create("BleTask", BLE_TASK_PRIORIYT, BLE_TASK_SIZE, 0, BleTaskFun, 0);
		if(gBleTask == INVALID_TASK)
		{
			LOG_ERROR("Task create ERROR");
			return false;
		}
	}
	
	LOG_INFO("Start ok");
	return true;
}

//==== Ble scan param set ====
static bool Scan_SetParam(void)
[... 284 lines cut ...]
					{
						Profile_Request(pProfile);
					}
				}
				break;

			case ESP_GATTC_CLOSE_EVT:
				pProfile->bIsOpen = 0;
				break;

			case ESP_GATTC_DISCONNECT_EVT:
				pProfile->bIsOpen = 0;
				needClose = true;
				break;

			default:
				break;
		}

		if(needClose)
		{
			Profile_Close(pProfile);
		}
	}
}

xueren-git, 2022-06-29:
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_RISCV=y
CONFIG_IDF_TARGET="esp32c3"
CONFIG_IDF_TARGET_ESP32C3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0005

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="riscv32-esp-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

#
# Security features
#
[... 1737 lines cut ...]
CONFIG_TCP_RECVMBOX_SIZE=20
CONFIG_TCP_QUEUE_OOSEQ=y
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=20
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

brunofgc, 2022-07-01:
> ----------------------------- Delete above -----------------------------
>
> ## Environment
> * Development Kit:      ESP32-DevKitC
> * Kit version DevKitC: v1
> * Module or chip used:  ESP32-C3
> * IDF [...]

Hi there. I'm facing the same issue but in my case I'm not connecting to a BT Device. Only doing the "Advertisement Scan".
I have another big concern, the processor is not "reseting" after watchdog get triggered. It means my device get stuck and doesn't recover from this state.

Could you please try help me on that? Have you ever seen this before?

Please See bellow the output:

DIAG0 1000e3a0
DIAG1 120b0000
BB DIAG0: 000a5384
                            BB DIAG1: ca000c82
                            BB DIAG2: 00000000
                            BB DIAG3: 8e89bed6
                            BB DIAG4: 00000000
                            BB DIAG5: 00000000
assert rwble.c 261, param 00020000 00000000
E (2411792) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (2411792) task_wdt:  - IDLE (CPU 0)
E (2411792) task_wdt: Tasks currently running:
E (2411792) task_wdt: CPU 0: wifi

xueren-git, 2022-07-04:
> Hi there. I'm facing the same issue but in my case I'm not connecting to a BT Device. Only doing the "Advertisement Scan". I have another big concern, the processor is not "reseting" after watchdog [...]

Hi. Whether the Component config->ESP System Settings->Invoke panic handler on Task Watchdog timeout Selected? If it's seleted, when happend "Task watchdog got triggered", the "components\esp_system\task_wdt.c" task_wdt_isr() will run "abort()"  restart.
if (twdt_config->panic){     //Trigger Panic if configured to do so
        ESP_EARLY_LOGE(TAG, "Aborting.");
        portEXIT_CRITICAL_ISR(&twdt_spinlock);
        esp_reset_reason_set_hint(ESP_RST_TASK_WDT);
        abort();
    }

xueren-git, 2022-07-04:
> [sdkconfig.txt](https://github.com/espressif/esp-idf/files/9008553/sdkconfig.txt)

Hi, my application menucofig Component config->Esp System Settings->Invoke panic handler on Task Watchdog timeout selected. When wacthdog timeout happend, components\esp_system\task_wdt.c task_wdt_isr() will run into "abort" restart.
if (twdt_config->panic){     //Trigger Panic if configured to do so
        ESP_EARLY_LOGE(TAG, "Aborting.");
        portEXIT_CRITICAL_ISR(&twdt_spinlock);
        esp_reset_reason_set_hint(ESP_RST_TASK_WDT);
        abort();
    }

xueren-git, 2022-07-04:
> Hi there. I'm facing the same issue but in my case I'm not connecting to a BT Device. Only doing the "Advertisement Scan". I have another big concern, the processor is not "reseting" after watchdog [...]

Hi, my menucofig Component config->Esp System Settings->Invoke panic handler on Task Watchdog timeout was selected. When wacthdog timeout happend, components\esp_system\task_wdt.c task_wdt_isr()  will run into "abort()" restart.

if (twdt_config->panic){ //Trigger Panic if configured to do so
ESP_EARLY_LOGE(TAG, "Aborting.");
portEXIT_CRITICAL_ISR(&twdt_spinlock);
esp_reset_reason_set_hint(ESP_RST_TASK_WDT);
abort();
}

nishanth-radja, 2022-07-04:
Hi @xueren-git, I am trying to repro the issue. Are you using  a wpa2-psk connection for wifi and the data packet which are sent every 10 seconds,Do you use TCP or UDP?

xueren-git, 2022-07-04:
Yes, it's. A WPA2-PSK connection for wifi, an TCP socket connection , and Using the SNTP server: pool.ntp.org. Gattc ble scanning all the time, but no ble device connection. The complete log file are attached which abort several times.

xueren-git, 2022-07-04:
> Hi @xueren-git, I am trying to repro the issue. Are you using a wpa2-psk connection for wifi and the data packet which are sent every 10 seconds,Do you use TCP or UDP?

The tcp socket sent a few byts to server, and recv the server ack data every 10 seconds.

xueren-git, 2022-07-07:
I try to remove gattc security code from my application or do not initialize WiFi, and the issue would not happen. However, I need to use that. And I try the sdk example "gatt_security_client",  the issue would not happen. But if I add WiFi connection in this example, the issue would happe.

xueren-git, 2022-07-07:
[example_ble_sec_gattc_demo_wifi.txt](https://github.com/espressif/esp-idf/files/9059329/example_ble_sec_gattc_demo_wifi.txt)

xueren-git, 2022-07-07:
Hi there. I try to remove gattc security code from my application or do not initialize WiFi, and the issue would not happen. However, I need to use that. And I try the sdk example "gatt_security_client", the issue would not happen. But if I add WiFi connection in this example, the issue would happe.

xueren-git, 2022-07-08:
I found that the cause of the problem was the use of both esp_ble_gap_config_local_privacy(true) and wifi connection.

closed 2022-07-27 by xueren-git

xueren-git, 2022-10-11:
Thank you for your reply.
1. sdkconfig file in email attachment.
2. Config Dual mode, but my application code only use ble.
3. Time is no fixed, sometimes more than ten hours, sometimes an hour or two.
4. esp_ble_gap_start_scanning(0), ble scanning all the time, and no ble device connection.
5. Partial code：

Wifi config sta mode, an connection working, send several bytes every 10 seconds.

//===Ble init ===
bool BleStart(void)
{
	esp_err_t err;
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();

	LOG_INFO("Start ...");
	gBleScaning = false;

	err = esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
	ERROR_ESP_RETURN(err, false, "Bt mem release ERROR");

	err = esp_bt_controller_init(&amp;bt_cfg);
	ERROR_ESP_RETURN(err, false, "Bt controller init ERROR");

	err = esp_bt_controller_enable(ESP_BT_MODE_BLE);
	ERROR_ESP_RETURN(err, false, "Bt controller enable ERROR");

	err = esp_bluedroid_init();
	ERROR_ESP_RETURN(err, false, "Bluedroid init ERROR");

	err = esp_bluedroid_enable();
	ERROR_ESP_RETURN(err, false, "Bluetooth enable ERROR");

	err = esp_ble_gap_register_callback(esp_gap_cb);
	ERROR_ESP_RETURN(err, false, "Gap callback init ERROR");
	err = esp_ble_gattc_register_callback(esp_gattc_cb);
	ERROR_ESP_RETURN(err, false, "[Gattc] register callback ERROR");

	#ifdef CONFIG_BLE_MUT
	err = esp_ble_gatt_set_local_mtu(CONFIG_BLE_MUT);
	LOG_ERROR_ESP_IF(err, "Set local mtu ERROR");
	LOG_INFO("Set local mut %d", CONFIG_BLE_MUT);
	#endif

	esp_ble_gap_config_local_privacy(true);
	esp_ble_auth_req_t		auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;	 // bonding with peer device after authentication
	esp_ble_io_cap_t		iocap = ESP_IO_CAP_IN;
	uint8_t key_size	= 16;	//the key size should be 7~16 bytes
	uint8_t init_key	= ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	uint8_t rsp_key		= ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
	uint8_t oob_support = ESP_BLE_OOB_DISABLE;
	esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &amp;auth_req, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &amp;iocap, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &amp;key_size, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &amp;oob_support, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &amp;init_key, sizeof(uint8_t));
	esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &amp;rsp_key, sizeof(uint8_t));

	if(gBleTask == INVALID_TASK)
	{
		gBleTask = task_create("BleTask", BLE_TASK_PRIORIYT, BLE_TASK_SIZE, 0, BleTaskFun, 0);
		if(gBleTask == INVALID_TASK)
		{
			LOG_ERROR("Task create ERROR");
			return false;
		}
	}
	
	LOG_INFO("Start ok");
	return true;
[... 288 lines cut ...]
					{
						Profile_Request(pProfile);
					}
				}
				break;

			case ESP_GATTC_CLOSE_EVT:
				pProfile-&gt;bIsOpen = 0;
				break;

			case ESP_GATTC_DISCONNECT_EVT:
				pProfile-&gt;bIsOpen = 0;
				needClose = true;
				break;

			default:
				break;
		}

		if(needClose)
		{
			Profile_Close(pProfile);
		}
	}
}

smetlitzky, 2023-02-21:
Hi There, 
Was this issue ever resolved, we are using ESP32-C3 and only Scanning for BLE and then sending MQTT packets via WiFi but after a random set of time we get the same crash and it does not reset :(
Any help greatly appreciated.

Best Regards
Steve

13:32:25.620 > DIAG0 1000e4a0
13:32:25.620 > DIAG1 10000000
13:32:25.625 > BB DIAG0: 000a5384
13:32:25.625 >                             BB DIAG1: e2000c02
13:32:25.632 >                             BB DIAG2: 00000000
13:32:25.637 >                             BB DIAG3: 8e89bed6
13:32:25.637 >                             BB DIAG4: 00000000
13:32:25.644 >                             BB DIAG5: 00000000
13:32:25.652 > assert rwble.c 261, param 00020000 00000000

chegewara, 2023-02-21:
I didnt see this issue for quite long time.
If i may suggest something, maybe try to switch to nimble?
I cant say that my example is valid in this case, because the current firmware i am working on for over a year is on esp32 (not C3), but it is using mqtt with wifi and ble scan very intensive and its very stable, and that `rwble.c` assert is here for 3 years, maybe longer.
Of course it also may be memory issue in your app, which would be good to check.

smetlitzky, 2023-02-21:
Thanks very much for your reply. Yes we are using the nimble library but still we are having this problem.

Best Regards
Steve

AxelLin, 2023-02-21:
> Thanks very much for your reply. Yes we are using the nimble library but still we are having this problem. Best Regards Steve

Which SoC and  esp-idf version are you using? (git describe --tags)

AxelLin, 2023-02-21:
> Hi There, Was this issue ever resolved??, we are using ESP32-C3 and only Scanning for BLE and then sending MQTT packets via WiFi but after a random set of time we get the same crash and it does not reset :( Any help greatly appreciated.

Which esp-idf version are you using? (git describe --tags)

xueren-git, 2023-02-21:
hi，i used chip esp32-c3 and IDF version : esp-idf-v4.4.1. I found that the cause of the problem was the use of both esp_ble_gap_config_local_privacy(true) and wifi connection. Didn't call esp_ble_gap_config_local_privacy(true) can be solved.

AxelLin, 2023-02-21:
> hi，i used chip esp32-c3 and IDF version : esp-idf-v4.4.1.&nbsp;I found that the cause of the problem was the use of both esp_ble_gap_config_local_privacy(true) and wifi connection. Didn't call&nbsp;esp_ble_gap_config_local_privacy(true) can&nbsp;be solved.

So you just prove this issue can be easily reproduced by calling esp_ble_gap_config_local_privacy(true) with wifi connection.

smetlitzky, 2023-02-21:
Hi and thanks very much for responding.

We are using the nimble library using arduino and platformio and I don’t think we have access to esp_ble_gap_config_local_privacy(true). Unless maybe it is set as default inside the library ??

Best Regards
Steve

xueren-git, 2023-02-21:
Hi, the api esp_ble_gap_config_local_privacy() in include
components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h .
I didn't use the IDE, i used ESP-IDF 4.4 CMD. I guess maybe arduino and platformio didn't include this header file.

xueren-git, 2023-02-21:
It's default esp_ble_gap_config_local_privacy(false)

wmy-espressif, 2023-02-21:
Hi @smetlitzky  @xueren-git , this issue is resolved in git commit(v4.4.2-237-g8e6af17ee5) for branch release/v4.4, and TAG v4.4.3 includes this bugfix.

This issue is related to a Bluetooth hardware error when using address resolution function.

chegewara, 2023-02-21:
Nice to finally see the real reason of this issue, which actually is confirming what i suspected (hardware issue).
This also is the reason why it popped:
`IDF version : esp-idf-v4.4.1`
assuming its finally fixed in `v4.4.2-237`
I also believe this is bluedroid function and nimble should have its own version:
`esp_ble_gap_config_local_privacy`

Thanks

LNqueen, 2023-02-27:
Hello, if I use the arduino and platform.io, how can I fix this issue? I can't find the esp_ble_gap_config_local_privacy() function, and I used the Nimble module, thanks a lot.