espressif/esp-idf #9242: Panic when testing wifi_prov_mgr example code

AxelLin, 2022-06-28:
## Environment

- Module or chip used: Customized ESP32-C3-mini-1 board
- IDF version: v5.0-dev-3654-gc2ccc383da
- Build System: idf.py
- Compiler version: riscv32-esp-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

My example configuration:

CONFIG_EXAMPLE_PROV_TRANSPORT_BLE=y
CONFIG_EXAMPLE_PROV_SECURITY_VERSION_1=y
CONFIG_EXAMPLE_PROV_TRANSPORT=1
CONFIG_EXAMPLE_RESET_PROVISIONED=y
CONFIG_EXAMPLE_RESET_PROV_MGR_ON_FAILURE=y
CONFIG_EXAMPLE_PROV_MGR_MAX_RETRY_CNT=5

Hit below panic when scan the QR code for BLE provision. (100% reproducable)
```
I (685) wifi:wifi firmware version: 3aa4137
I (685) wifi:wifi certification version: v7.0
I (685) wifi:config NVS flash: enabled
I (685) wifi:config nano formating: disabled
I (685) wifi:Init data frame dynamic rx buffer num: 32
I (685) wifi:Init management frame dynamic rx buffer num: 32
I (685) wifi:Init management short buffer num: 32
I (685) wifi:Init dynamic tx buffer num: 32
I (685) wifi:Init static tx FG buffer num: 2
I (685) wifi:Init static rx buffer size: 1600
I (685) wifi:Init static rx buffer num: 10
I (685) wifi:Init dynamic rx buffer num: 32
I (685) wifi_init: rx ba win: 6
I (685) wifi_init: tcpip mbox: 32
I (685) wifi_init: udp mbox: 6
I (685) wifi_init: tcp mbox: 6
I (685) wifi_init: tcp tx win: 5744
I (685) wifi_init: tcp rx win: 5744
I (685) wifi_init: tcp mss: 1440
I (685) wifi_init: WiFi IRAM OP enabled
I (685) wifi_init: WiFi RX IRAM OP enabled
D (685) wifi_prov_mgr: execute_event_cb : 0
W (685) BTDM_INIT: esp_bt_mem_release not implemented, return OK
I (685) wifi_prov_scheme_ble: BT memory released
D (685) event: running post WIFI_PROV_EVENT:0 with handler 0x420053ca and context 0x3fc9f9d4 on loop 0x3fc9ea60
0x420053ca: event_handler at /home/axel/esp/esp-idf/examples/provisioning/wifi_prov_mgr/main/app_main.c:117

D (685) nvs: nvs_set opmode 1 2
D (695) nvs: nvs_set_blob sta.ssid 36
D (695) nvs: nvs_set sta.authmode 1 1
D (695) nvs: nvs_set_blob sta.pswd 65
D (695) nvs: nvs_set_blob sta.pmk 32
D (695) nvs: nvs_set sta.chan 1 0
D (695) nvs: nvs_set auto.conn 1 1
D (695) nvs: nvs_set bssid.set 1 0
D (695) nvs: nvs_set_blob sta.bssid 6
D (695) nvs: nvs_set sta.lis_intval 2 3
D (695) nvs: nvs_set sta.phym 1 3
D (695) nvs: nvs_set sta.phybw 1 2
D (695) nvs: nvs_set_blob sta.apsw 2
D (695) nvs: nvs_set_blob sta.apinfo 700
D (695) nvs: nvs_set sta.scan_method 1 0
D (695) nvs: nvs_set sta.sort_method 1 0
D (695) nvs: nvs_set sta.minrssi 1 -127
D (695) nvs: nvs_set sta.minauth 1 0
D (695) nvs: nvs_set sta.pmf_e 1 1
D (695) nvs: nvs_set sta.pmf_r 1 0
D (695) nvs: nvs_set sta.btm_e 1 0
[... 136 lines cut ...]
uxListRemove (pxItemToRemove=pxItemToRemove@entry=0x19) at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/list.c:197
197	    List_t * const pxList = pxItemToRemove->pxContainer;
#0  uxListRemove (pxItemToRemove=pxItemToRemove@entry=0x19) at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/list.c:197
#1  0x4038a298 in xTaskRemoveFromEventList (pxEventList=pxEventList@entry=0x3fcb5cf0) at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3647
#2  0x40388f6c in xQueueGenericSend (xQueue=xQueue@entry=0x3fcb5ccc, pvItemToQueue=pvItemToQueue@entry=0x0, xTicksToWait=<optimized out>, xTicksToWait@entry=0, xCopyPosition=xCopyPosition@entry=0) at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:890
#3  0x4038908a in xQueueGiveMutexRecursive (xMutex=0x3fcb5ccc) at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:692
#4  0x40382cb4 in npl_freertos_mutex_release (mu=<optimized out>) at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:518
#5  0x42019f5c in ble_sm_rx (chan=0x3fc942b0 <ble_l2cap_chan_mem+136>) at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_sm.c:2658
#6  0x4200f260 in ble_hs_hci_evt_acl_process (om=0x0) at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:946
#7  0x4200e7ce in ble_hs_process_rx_data_queue () at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:248
#8  0x40381ff8 in ble_npl_event_run (ev=0x3fc94004 <ble_hs_rx_q+8>) at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
#9  nimble_port_run () at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:181
#10 0x4200b6e6 in nimble_host_task (param=<optimized out>) at /home/axel/esp/esp-idf/components/protocomm/src/transports/protocomm_nimble.c:473
#11 0x4038b20a in prvTaskExitError () at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:132
Backtrace stopped: frame did not save the PC
ELF file SHA256: da19978a79e592dc

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038065c
0x4038065c: esp_restart_noos_dig at /home/axel/esp/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)

```

AxelLin, 2022-06-28:
BTW, https://github.com/espressif/esp-idf/commit/6222d4398490d3f044db0af6c4b748c5a946cefc#diff-2fee4db941ef083026785c45f69db680f8b60a002fb6dc1085875e728729e3c2R27 has a typo.

AxelLin, 2022-06-28:
After increase CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE, this issue is no longer happen.
(So the default stack size is too small?)

closed 2022-06-28 by AxelLin

rahult-github, 2022-06-28:
Hi @AxelLin  , thanks for reporting. Yes, a fix has already been identified and is been worked upon. As a temporary fix, increasing stack size can be tried till the actual fix gets pushed.

AxelLin, 2022-06-30:
@rahult-github 

When I test using esp_prov.py it always failed as below.
Is there anything wrong with below commands?

// test protocomm_security1
$ python esp_prov.py --verbose --sec_ver 1 --transport ble --service_name PROV_67846C --pop abcd1234
checking if adapter is powered on
bluetooth adapter powered on
starting discovery
start discovery successful
Connecting...
device is connected
Connected
Getting Services...

==== Starting Session ====
++++ Client Public Key: a0493e81461f897c110d12ee676549befb219bee31e944d37a360dec38fcf325 ++++
Failed to write value to characteristic 021aff51-0382-4aea-bff4-6b3f1c5adfb4: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Failed to establish session. Ensure that security scheme and proof of possession are correct
---- Error in establishing session ----
device disconnected

// test protocomm_security2
$ python esp_prov.py --verbose --transport ble --service_name PROV_67846C --sec_ver 2 --sec2_username testuser --sec2_pwd testpassword
checking if adapter is powered on
bluetooth adapter powered on
starting discovery
start discovery successful
Connecting...
device is connected
Connected
Getting Services...

==== Starting Session ====
++++ Client Public Key: 0xbf9ca2d735c1e59e7286f043c7bd1f4239f041e52b65ef7d0b0987762908c47da97e92df2f9771711553aa4a572035521e094ad236ef2fc74af24f6d6af06d07f9386c57fc320ec412ecaaa2936dd546b58dd45852ba0b9477c0ee5b352f58beb8cf9e32e2213116e5b18c9d4c9bf08a6ff287b2650255ca83ae8aa8059e2bb6e59866d86e35b93f4b690b376da6a0b85de42b24495c539ade9bda50b5e5b326a644f00d816668d597e9771b15a044f9deaa17fd31e6e32213da2057513e7e3d33c7f3aa5638034119ed030d5c0b61d193282b66941e9bca39d4535d5105769efbf848244d6f712610e9a25595cc1612f3d21457a69ec02909b2175e3f8ede21507523e86cb92afcb34dcf1bf2d5c3a78c0af842e326c90c29e7b7d43cf33676f8ea4b4450a6c53dd622baf70c22c01340a9e905be709a46d8ec46e40d34bccd6d3139d61570c70cab23b6db20e2c13fff231b70cd78ca4381a1e3dea6d75c610e26cd1f2ff58b70253142dbf6810b4d6ebe9ceb4ad9119ff6b2813049b1b8ed ++++
Failed to write value to characteristic 021aff51-0382-4aea-bff4-6b3f1c5adfb4: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Failed to establish session. Ensure that security scheme and proof of possession are correct
---- Error in establishing session ----
device disconnected

laukik-hase, 2022-06-30:
Hello, @AxelLin!

The issue you are facing is related to pairing. When using the `wifi_prov_mgr` example with the default configuration and the `esp_prov` tool, you will get a Bluetooth pairing request on the host side. E.g., Linux shows a prompt as follows -

[image]

Only on accepting the request will the provisioning advance or it will fail with the error logs you gave. You can disable the config `CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION` -> `Force Link Encryption during characteristic Read / Write` to disable the pairing prompt and allow the host to connect to the target directly.

hayschan, 2022-12-19:
Where did you set `CONFIG_EXAMPLE_PROV_SECURITY_VERSION_1=y`? @AxelLin 

```yaml
CONFIG_EXAMPLE_PROV_TRANSPORT_BLE=y
CONFIG_EXAMPLE_PROV_SECURITY_VERSION_1=y
CONFIG_EXAMPLE_PROV_TRANSPORT=1
CONFIG_EXAMPLE_RESET_PROVISIONED=y
CONFIG_EXAMPLE_RESET_PROV_MGR_ON_FAILURE=y
CONFIG_EXAMPLE_PROV_MGR_MAX_RETRY_CNT=5
```

I see that five of the above six settings could be set by `make menuconfig`.

[image]

The choices are written in `Kconfig.projbuild`

```yaml
menu "Example Configuration"

    choice EXAMPLE_PROV_TRANSPORT
        bool "Provisioning Transport"
        default EXAMPLE_PROV_TRANSPORT_SOFTAP if IDF_TARGET_ESP32S2
        default EXAMPLE_PROV_TRANSPORT_BLE
        help
            Wi-Fi provisioning component offers both, SoftAP and BLE transports. Choose any one.

        config EXAMPLE_PROV_TRANSPORT_BLE
            bool "BLE"
            select BT_ENABLED
            depends on !IDF_TARGET_ESP32S2
        config EXAMPLE_PROV_TRANSPORT_SOFTAP
            bool "Soft AP"
    endchoice

    config EXAMPLE_PROV_TRANSPORT
        int
        default 1 if EXAMPLE_PROV_TRANSPORT_BLE
        default 2 if EXAMPLE_PROV_TRANSPORT_SOFTAP

    config EXAMPLE_RESET_PROVISIONED
        bool
        default n
        prompt "Reset provisioned status of the device"
        help
            This erases the NVS to reset provisioned status of the device on every reboot.
            Provisioned status is determined by the Wi-Fi STA configuration, saved on the NVS.

    config EXAMPLE_RESET_PROV_MGR_ON_FAILURE
        bool
        default y
        prompt "Reset provisioned credentials and state machine after session failure"
        help
            Enable reseting provisioned credentials and state machine after session failure.
            This will restart the provisioning service after retries are exhausted.

    config EXAMPLE_PROV_MGR_MAX_RETRY_CNT
        int
        default 5
        prompt "Max retries before reseting provisioning state machine"
        depends on EXAMPLE_RESET_PROV_MGR_ON_FAILURE
        help
            Set the Maximum retry to avoid reconnecting to an inexistent AP or if credentials
            are misconfigured. Provisioned credentials are erased and internal state machine
            is reset after this threshold is reached.

    config EXAMPLE_PROV_SHOW_QR
        bool "Show provisioning QR code"
        default y
        help
            Show the QR code for provisioning.

    config EXAMPLE_PROV_USING_BLUEDROID
        bool
        depends on (BT_BLUEDROID_ENABLED && (IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3))
        select BT_BLE_42_FEATURES_SUPPORTED
        default y
        help
            This enables BLE 4.2 features for Bluedroid.
endmenu
```

However, I don't see where I could specify the security version. Do I specify it in the [`sdkconfig.defaults`]( https://github.com/espressif/esp-idf/blob/edd815af2e7d541b25ff3a74c59a7fe0612df42d/examples/provisioning/wifi_prov_mgr/sdkconfig.defaults)? Or do I specify it in [`sdkconfig.ci.security1`](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/wifi_prov_mgr/sdkconfig.ci.security1)?