espressif/esp-idf #9244: ESP32S3 efuses real for production won't work, flash crypted, secure boot v2 enabled, .bin signed, with efuses virtual work

MikyZ72, 2022-06-28:
## Environment

- Development Kit:      [custom board]
- Kit version     [custom board]
- Module or chip used:  [**ESP32-S3-WROOM-1-N16R8**]
- IDF version :    [**ESP-IDF v4.4.1-405-g6c5fb29c2c**]
- Build System:         [Espressif IDE (Eclipse IDE) Version: 2.4.2 Build id: 20220308-1406 CMake Ninja]
- Compiler version:   [xtensa-esp32s3-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0]
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: Yes [Espressif IDE (Eclipse IDE) Version: 2.4.2 Build id: 20220308-1406]
- Power Supply:         [external 3.3V from custom board]

## Problem Description

Hello guys, 
If i flash all .bin system for production (ota, ota_data, factory, partition table, bootloader ecc) with efuses virtual enabled in partition emul_efuse, all work fine.
If i disable virtual efuses for real burn efuses, start the problem, i think to see 2 problems:
- stop working partition ota: `esp_partition_get_sha256` on ota return code 0x2002 
- partition storage (FAT) at all power on the `esp_vfs_fat_spiflash_mount` format the FAT like it lose the last formatted procedure on power off

I have made many test, sdkconfig and different flash procedures and i have briked 10 module, till now i dont have found solution!

### Project

I have 2 project in Espressif IDE (Eclipse):
- one for my loader (bootloader, partition table and factory app that is my loader.bin) and i flash later all bin files created
- one for my firmware (ota app that is my firmware.bin) and i flash later only the firmware.bin file

The 2 project have the same sdkconfig and the same custom partition.csv file (they are ugual and i keep ugual)

In both there are enabled Flash encryption AES-256, Secure Boot V2 with sign binary during build with my generated key (openssl), ROM download mode is enabled (insecure for now).
Sign key is the same for the loader project and firmware project
I dont encrypt NVS data

**SDKconfig image encryption**
[image]

Bootloader of loader project is customized for run all the time at power on Factory partition and for run all the time ota partition from deep sleep wakeup.
Bootloader of firmware project it's the original from Espressif, isn't customized, but i dont use it.
When bootloader run my loader app from factory partition, after some check i launch deep sleep for hundred ms (100) and at wake up bootloader run ota partition (i use only 2 partition for update: 1 factory and 1 ota, this for don't lose flash space of 3 partition like 1 factory and 2 ota as a espressif default).
Partition table offset is set to **"0xb000"** for enlarge bootloader space (because crypt and info logs) as a documentation explain.

### Partitions
```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0xc000,16K,
otadata,data,ota,0x10000,8K,
phy_init,data,phy,0x12000,8K,
emul_efuse,data,efuse,0x14000,8K,
loader,app,factory,0x20000,1M,
app,app,ota_0,0x120000,1920K,
storage,data,fat,0x300000,13M,encrypted
```

### Write to flash

After build, i flash all binary with manual command from prompt:

`esptool.py -p COM7 --chip esp32s3 erase_flash`

`esptool.py -p COM7 --chip esp32s3 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x20000 loader.bin 0xb000 partition-table.bin 0x10000 ota_data_initial.bin 0x120000 firmware.bin`

`esptool.py -p COM7 --chip esp32s3 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader.bin`

In all command i dont reset after all bin file are flashed and Bootloader as the last as documentation mention because at restart it will start to burn the efuses if all crypt process is fine.
It will reset when i start monitor:
`idf.py -p COM7 monitor`

[... 469 lines cut ...]
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
EXT_PHY_ENABLE (BLOCK0)                            Enables external USB PHY                           = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enables BTLC GPIO                                  = 0 R/W (0b00)
DIS_USB_JTAG (BLOCK0)                              Disable usb_serial_jtag-to-jtag function           = True R/W (0b1)
DIS_USB_SERIAL_JTAG (BLOCK0)                       Disable usb_serial_jtag module                     = False R/W (0b0)
USB_PHY_SEL (BLOCK0)                               Select internal/external PHY for USB OTGand usb_se = False R/W (0b0)
                                                   rial_jtag

Vdd_Spi Config fuses:
VDD_SPI_XPD (BLOCK0)                               The VDD_SPI regulator is powered on                = True R/W (0b1)
VDD_SPI_TIEH (BLOCK0)                              The VDD_SPI power supply voltage at reset
   = Connect to VDD_RTC_IO R/W (0b1)
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD_SPI R/W (0b1)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.
```

## Other items if possible
- sdkconfig file:   [sdkconfig.txt](https://github.com/espressif/esp-idf/files/9001174/sdkconfig.txt)
- partition file:   [partitions.csv](https://github.com/espressif/esp-idf/files/9001178/partitions.csv)

mahavirj, 2022-07-04:
@MikyZ72 

From eFuse summary, both secure boot and flash encryption schemes have been correctly configured. Following bootloader log confirms this.

> I (347) secure_boot_v2: secure boot v2 is already enabled, continuing..
I (354) boot: Checking flash encryption...
I (359) flash_encrypt: flash encryption is enabled (0 plaintext flashes left)

Moreover, bootloader is also successful in verifying firmware (its signature and hence integrity) and then it successfully hands over control to it.

> Due to the ota is invalid my loader dont deep sleep and stay inside itself, wait for ota update via bluetooth with our protocol.
At every power on i see that loader formatting every time the FAT, this is not correct.

I have few questions here:

1. If you try ESP-IDF provided `fatfs` example, do you still run into same issue?
2. Can you please highlight on your changes or maybe share your sample project?

CC @igrr

MikyZ72, 2022-07-05:
Good morning @mahavirj ,
many thanks for your reply... 

After a hundred test we find that problem is with PSRAM enabled, encryption and in specific esp_partition_write stop work.
In our project if we disable PSRAM all work fine.

We have also try Espressif example flash_encryption and with PSRAM disabled work fine, if we enable it, also the example stop to work.
Please look example wrong output, the readed sequence of byte are not ugual to what it have just writed and the readed crypted byte are all 0xFF:

```Example to check Flash Encryption status
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision 0, 16MB external flash
FLASH_CRYPT_CNT eFuse value is 1
Flash encryption feature is enabled in DEVELOPMENT mode
Erasing partition "storage" (0x1000 bytes)
Writing data with esp_partition_write:
I (483) example: 0x3fcf40e0   00 01 02 03 04 05 06 07  08 09 0a 0b 0c 0d 0e 0f  |................|
I (493) example: 0x3fcf40f0   10 11 12 13 14 15 16 17  18 19 1a 1b 1c 1d 1e 1f  |................|
Reading with esp_partition_read:
I (503) example: 0x3fcf40c0   c7 85 a3 a7 01 03 02 c7  33 74 7b 1f d3 c6 1a f8  |........3t{.....|
I (513) example: 0x3fcf40d0   97 0d 13 fa 8f 5e 0b 66  92 ae 0c a4 e9 fa bc bb  |.....^.f........|
Reading with spi_flash_read:
I (523) example: 0x3fcf40c0   ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
I (533) example: 0x3fcf40d0   ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
I (553) nvs: NVS partition "nvs" is encrypted.```

mahavirj, 2022-07-05:
@MikyZ72 

> We have also try Espressif example flash_encryption and with PSRAM disabled work fine, if we enable it, also the example stop to work.

Thanks for confirming! Can you please share your `sdkconfig` in case you made any modifications to default configuration? Also you are trying this on `ESP32-S3-WROOM-1-N16R8` (just to double confirm)?

MikyZ72, 2022-07-06:
Hi @mahavirj ,
here the sdkconfig of flash encryption example with PSRAM enabled

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9056147/sdkconfig.txt)

Yes i confirm that module is ESP32-S3-WROOM-1-N16R8

Emill, 2022-07-07:
I did some more debugging.

First I tested the latest release/v5.0 branch to see if that would help. I had to migrate to the new `esp_flash_` functions, but that did not change the end result; it is still broken.

I then found out that the driver writes the correct data, but instead of writing to `addr`, it seems it writes to `(addr << 8)` instead (modulo 16 MB). After copying 64 bytes of data from address 0x00C000 in my flash dump (see my previous post) to address 0x1100C0 in the flash dump, and then decrypting the whole flash dump using espsecure.py, the data at address 0x1100C0 is then correctly decrypted (i.e. `5A 5A 5A ...`). This confirms that the AES-XTS hardware encrypts for the correct address, but the SPI1 peripheral writes the result to the wrong address `(addr << 8)`.

The reason the flash dump contains 64 byte "garbage" every 16 384 byte blocks is due to the `spi_flash_chip_generic_write_encrypted` function splits up the data in 64 byte blocks and then writes them individually. 64 << 8 is 16 384, so that's where that offset comes from.

By using a debugger and setting a breakpoint at the beginning of `spimem_flash_ll_user_start`, at this point `dev->user1.usr_addr_bitlen` contains the correct value `24 - 1`. Also `dev->addr` contains the correct value `0x00110000` (when writing to that address). I however noticed that `dev->cache_fctrl.usr_cmd_4byte` contained 1, which has the header file documentation "Set this bit to enable SPI1 transfer with 32 bits address. The value of `SPI_MEM_USR_ADDR_BITLEN` should be 31.". **This is the cause of the issue** which causes SPI1 to write to the incorrect flash address.

I saw that the octal PSRAM uses 32-bit address. After some investigation I found that the `esp_rom_opiflash_exec_cmd` function sets this value to 1 without revertinig it. That function is called for SPI1 during PSRAM initialization. It seems PSRAM only uses SPI0 later on, so it should be safe to revert it after PSRAM initialization is complete. Flash Encryption write operation uses SPI1.

So for now I added this line:

```C
SET_PERI_REG_BITS(SPI_MEM_CACHE_FCTRL_REG(1), SPI_MEM_CACHE_USR_CMD_4BYTE_V, 0, SPI_MEM_CACHE_USR_CMD_4BYTE_S);
```

to the function `spi_flash_set_rom_required_regs` (outside the `#if` block). This seems to solve the issue for the N16R8 module. I'm not sure if it's a correct solution for modules with larger flashes though.

Another alternative might be to set `dev->cache_fctrl.usr_cmd_4byte = bitlen == 32 ? 1 : 0;` in `spimem_flash_ll_set_usr_address`. But then it's done unnecessarily often maybe.

MikyZ72, 2022-07-09:
Hello Emill,
thank for the big and complete information, but we are in production and we have exchange our code for deactivate PSRAM and dont use more this feature, now we are not able to wait some more fix or more time, we must sell the product (thousands).
Now we sell product with code that don't respect our first specifications, and we have pay the module with PSRAM presence for nothing, plus we dont now if in the future (when there is the stable solution) we are able to re-enable PSRAM and update the product via ota without problem.
This bug i think don't must be happen in Espressif code/company, mostly in the encryption parts, now i feel and i'm afraid that all code isn't tested in good way and maybe we have many problem with the product on our customer.

moefear85, 2022-07-09:
@MikyZ72 

> This bug i think don't must be happen in Espressif code/company, mostly in the encryption parts, now i feel and i'm afraid that all code isn't tested in good way and maybe we have many problem with the product on our customer.

I personally share your frustration and have often wondered if espressif intends to sell only to hobbyists. But on the other hand, there is a huge price difference between esp products and other companies. No other company offers an mcu with integrated wifi/bt/usb/psram for such a low price (which I find attractive, like probably yourself too). I try to remind myself about that everytime I grind my teeth wondering why something isn't working as I hoped.

mythbuster5, 2022-07-18:
@Emill Hi, thanks for your fix, and your suggestion is great.

Take stability in to consideration, two parts to be modified. So, you can pick this commit https://github.com/espressif/esp-idf/commit/8538153616af1b651af990c69ebc1b4cb2835c53 and things will be ok.

closed 2022-07-19 by espressif-bot

SolidStateLEDLighting, 2023-05-29:
Has this issue ever been solved?   I have this same problem with the Esp32s3 

Where I enable Support for external, SPI-connected RAM and select Octal Mode PSRAM.

And now after a lot of development, we are trying to secure our work product and can't get Flash Encryption to work without receiving      Failed to read NVS security cfg: [0x1117] (ESP_ERR_NVS_CORRUPT_KEY_PART)