espressif/esp-idf #9245: Unable to compile ot_br if option CONFIG_OPENTHREAD_ESP_LIB_FROM_INTERNAL_SRC=y

fcgdam, 2022-06-28:
Before submitting a new issue, please follow the checklist and try to find the answer.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:  ESP32-DOIT DevKit v1
- Kit version (for WroverKit/PicoKit/DevKitC): V1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3654-gc2ccc383da
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:    Linux (Arch Linux fully updated)
- Using an IDE?: No
- Power Supply:       USB
- CMAKE version: cmake version 3.23.2
 
## Problem Description

Trying/testing the new releases from the openthread repository. For that I need to regenerate the libopenthread_*.a files.
For doing so, on the ot-br project ran: idf.py menuconfig and enabled "**Build esp_openthread libraries from source**" option by going to Components->Openthread and selecting the above option. This adds **CONFIG_OPENTHREAD_ESP_LIB_FROM_INTERNAL_SRC=y** to the **sdkconfig** file.

### Expected Behavior
Example is compiled and the Openthread libs under **components/openthread/lib/esp32** are regenerated.

### Actual Behavior

Compilation fails when executing the **idf.py build** command:
`
.....
-- Adding linker script /opt/ESP32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /opt/ESP32/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
CMake Error at /opt/ESP32/esp-idf/tools/cmake/component.cmake:380 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  /opt/ESP32/esp-idf/components/openthread/CMakeLists.txt:169 (idf_component_get_property)

CMake Error at /opt/ESP32/esp-idf/tools/cmake/component.cmake:380 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  /opt/ESP32/esp-idf/components/openthread/CMakeLists.txt:183 (idf_component_get_property)

-- Components: app_trace app_update asio
...
....
....
-- Configuring incomplete, errors occurred!
See also "/opt/ESP32/esp-idf/examples/openthread/ot_br/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 

/usr/bin/cmake --regenerate-during-build -S/opt/ESP32/esp-idf/examples/openthread/ot_br -B/opt/ESP32/esp-idf/examples/openthread/ot_br/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1

`

### Steps to reproduce

1. Run **idf.py menuconfig** on the ot_br example from esp-idf.
2. Goto **Component config->OpenThread** and enable **Build esp_openthread libraries from source**
[image]
3. Save and exist the menuconfig.
4. Try to compile with **idf.py build** 
5. Fails to compile.
6. To recover, edit the **sdkconfig** file and change the **CONFIG_OPENTHREAD_ESP_LIB_FROM_INTERNAL_SRC=y** to **=n**.

```
#
# Openthread
#
CONFIG_OPENTHREAD_ENABLED=y
# CONFIG_OPENTHREAD_RADIO_NATIVE is not set
CONFIG_OPENTHREAD_RADIO_SPINEL_UART=y
CONFIG_OPENTHREAD_FTD=y
# CONFIG_OPENTHREAD_MTD is not set
# CONFIG_OPENTHREAD_RADIO is not set
CONFIG_OPENTHREAD_DIAG=y
# CONFIG_OPENTHREAD_COMMISSIONER is not set
# CONFIG_OPENTHREAD_JOINER is not set
# CONFIG_OPENTHREAD_SRP_CLIENT is not set
CONFIG_OPENTHREAD_BORDER_ROUTER=y
CONFIG_OPENTHREAD_ESP_LIB_FROM_INTERNAL_SRC=y
CONFIG_OPENTHREAD_NUM_MESSAGE_BUFFERS=65
# end of OpenThread

```

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9002340/sdkconfig.txt)

chshu, 2022-06-29:
@fcgdam The `CONFIG_OPENTHREAD_ESP_LIB_FROM_INTERNAL_SRC` is not public available, only for internal use, so please don't enable this option from your side.

Currently only the ESP Openthread port and Border Router implementation are released in library, the Openthread core stack is still built from source.

fcgdam, 2022-06-29:
Ok, fair enough. So is not possible to recompile libopenthread*.a files under the lib/esp32 folder? I'm asking this because any change on the Openthread source is not mirrored on the used openthread stack in applications. What I mean is that if I add some debug log output to openthread sources, those changes aren't used since the ot_br code links to the library, not to the compiled code.

chshu, 2022-06-29:
Actually any changes to openthread sources, will be built into the final binary. 

Attach a stack layer diagram for your reference, only the ESP OpenThread Port and Border Router are in library, the Openthread itself is built from source code, so you can make any changes you want, as long as no changes to the interface(APIs). 

[image]

fcgdam, 2022-06-29:
Ok, thanks. The issue that I have right now is that I only see the ot_br sending IPV6 router solicitation messages (check by using Wireshark) and no Router Advertising messages. Hence the OMR (Off-Mesh Prefix) is not know by client computers. Nevertheless this seemed to work on previous versions of esp-idf (April/May I'll have to check) where OMR was added to host computers routing table automatically. So I was trying to add debug messages to the Openthread code to see if the Router ADV code is called.

The fact is if I add garbage code or lines to the Openthread code under **components/openthread/openthread/**, ot_br compiles fine... hence it does not compile the Openthread from sources, at least from disk and on my setup...

closed 2022-07-01 by fcgdam