espressif/esp-idf #9246: Conflict using TWAI and W5500 SPI Ethernet

backemf-esa, 2022-06-28:
## Environment

- Development Kit: [none]
-  Module or chip used: [ESP32-WROOM-32E]
- IDF version: [4.4]
- Operating System: [Windows]
- Using an IDE?: [VSCode]
- Power Supply: [external 3.3V]

We are using a configuration with a ESP32-WROOM-32E, Wiznet W5500 ethernet controller, and a MCP2551 CAN transceiver w/ level shifters. We are using GPIO4 for CAN TX and GPIO27 for CAN RX. For connection to the W5500 we are using VSPI: GPIO5(CS), GPIO18(CLK), GPIO19(MISO), GPIO23(MOSI), GPIO21(INTN), GPIO22(RST)

We are sending commands to a network of controllers.  The first node consumes a command on wired ethernet then sends commands to additional nodes over CAN. The controllers are identical hardware. We are able to send/receive CAN messages as long the ethernet does not have a cable plugged into it. So, our hardware works using CAN if it is node 2, 3, etc., but will not work if it is node 1. Our board is able to to CAN or Ethernet, but not both. We were able to verify this by using an Olimex ESP32-POE dev kit as node 1 to get our complete system to work. It uses an ethernet controller that uses the RMII interface, whereas the W5500 interfaces to the ESP32 over SPI.

When we use ethernet and CAN together we get undecodable CAN messages (see image below). If the ethernet is not plugged in the pulses look as they should and we are able to decode.

[image]

atanisoft, 2022-06-29:
It sounds likely that you are having interrupt starvation issues. Can you test initializing CAN on core 1 and ETH on core 0?

kostaond, 2022-08-03:
@backemf-esa, just for sure, you are able to run CAN and Ethernet when internal EMAC connected to PHY via RMII but it does not work when you use SPI Ethernet device (W5500).

> When we use ethernet and CAN together we get undecodable CAN messages (see image below). If the ethernet is not plugged in the pulses look as they should and we are able to decode.

Could you please provide decoding of colors? And at which point the signal was measured? On ESP32 GPIOs associated with CAN?

Since it seems you have problems with logic levels (based on the figure) it looks like some problem with hardware. It looks like there is something pushing the signals up and the IO driver is not able to pull it down.

kostaond, 2022-08-19:
@backemf-esa, just a gentle reminder. Could you please provide more info?

MacWyznawca, 2022-09-09:
I have "reverted" issue. TWAI works stable, but when TWAI start transmitted data with other TWAI node, SPI W5500 slowdown and stop responding. After TWAI deinstalled on other node, Ethernet start working without needing disconnected cable TWAI or power down node. 
ESP32S3 and ESP-IDF 5.0b1