espressif/esp-idf #9260: Riscv ULP hangs on ulp_riscv_config_and_run in ESP32S3

Belal-Elshinnawey, 2022-06-30:
## Environment

- Development Kit:      Standalone ESP32-S3 with 2MB flash.
- IDF version : v5.0-dev-3654-gc2ccc383da
- Build System: idf.py
- Compiler version: xtensa-esp32s3-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No, IDE only for Dev, Everything else is manually by CMD
- Power Supply:        Battery 3.7

## Problem Description
I'm experimenting with the GPIO interrupt example for ESP32S3 ulp riscv. 
Using the same GPIO 0, the problem exists in both RTC IO and regular GPIO. 
What I am trying to do is have the ESP go to deepsleep and only wakeup the ULP from interrupt. The ULP can decide if it wants to wakeup The main processor or not.

The problem is If the ESP resets, from Power, It does not init the ULP or go to Deepsleep (Current is 20mA). However, If I trigger the interrupt that causes the ULP to wake up the ESP(press the button), the main processor resets but manages to get the ULP to start, and manages to inter deepsleep afterwords. 

### Expected Behavior
The ESP should be able to Init the RISCV processor and continue the program as normal when Power reset is triggered.
### Actual Behavior
The ESP hangs and stops responding when  ulp_riscv_config_and_run(&cfg) is called. and does not execute any codes after it. Unless if the interrupt is triggered, then it works normally.
### Steps to reproduce

1. get  the https://github.com/espressif/esp-idf/tree/master/examples/system/ulp_riscv/gpio_interrupt from the ESP repo
2. set target as ESP32S3, and build flash it.
3. wait for the ESP to start ULP and go to sleep.
4. the ESP will stop at ulp_riscv_config_and_run
5. pull the GPIO0 pin low and the ESP will restart and work normally.
6. if you reset the esp via power it will do the same again

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
#include <stdio.h>
#include "esp_sleep.h"
#include "soc/sens_reg.h"
#include "driver/gpio.h"
#include "driver/rtc_io.h"
#include "ulp_riscv.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#define WAKEUP_PIN 0
extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");
static void init_ulp_program(void);
static void wakeup_gpio_init(void)
{
    rtc_gpio_init(WAKEUP_PIN);
    rtc_gpio_set_direction(WAKEUP_PIN, RTC_GPIO_MODE_INPUT_ONLY);
    rtc_gpio_pullup_dis(WAKEUP_PIN);
    rtc_gpio_pulldown_dis(WAKEUP_PIN);
    rtc_gpio_wakeup_enable(WAKEUP_PIN, GPIO_INTR_LOW_LEVEL);
}
void app_main(void)
{
    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();

    /* not a wakeup from ULP, load the firmware */
    if (cause != ESP_SLEEP_WAKEUP_ULP) {
        printf("Not a ULP-RISC-V wakeup, initializing it! \n");
        wakeup_gpio_init();
        init_ulp_program();
    }
    if (cause == ESP_SLEEP_WAKEUP_ULP) {
        printf("ULP-RISC-V woke up the main CPU! \n");
    }
    printf("Entering in deep sleep\n\n");
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
    ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup());
[... 309 lines cut ...]
V (671) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (678) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC02
D (686) intr_alloc: Connected src 59 to int 3 (cpu 0)
I (691) sleep: Configure to isolate all GPIO pins in sleep state
I (698) sleep: Enable automatic switching of GPIO sleep configuration
V (716) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (716) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (720) intr_alloc: Connected src 79 to int 9 (cpu 0)
I (725) cpu_start: Starting scheduler on PRO CPU.
V (730) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (730) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x402
D (730) intr_alloc: Connected src 57 to int 12 (cpu 0)
V (730) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (740) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (740) intr_alloc: Connected src 80 to int 2 (cpu 1)
I (750) cpu_start: Starting scheduler on APP CPU.
V (750) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (760) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x402
D (770) intr_alloc: Connected src 58 to int 3 (cpu 1)
D (770) heap_init: New heap initialised at 0x3fce0000
V (780) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (790) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (790) intr_alloc: Connected src 52 to int 13 (cpu 0)
Not a ULP-RISC-V wakeup, initializing it!
Entering in deep sleep  <<==== Enters Deep sleep Normally, and power is in uA range after  ULP is started.

ESP-Marius, 2022-06-30:
Hi, @Belal-Elshinnawey 

I haven't been able to reproduce it. Atleast with my S3 dev-kitC-1, the board always enters deep sleep like it should even on the first power on cycle.

Can you think of anything else you've changed, e.g. menuconfig options? If you just do a clean build from scratch using the example without any modifications does it still happen?

There shouldn't really be any places where the SW can get stuck inside `ulp_riscv_config_and_run`, as it is just setting some registers :thinking: . Are you able to add some more debug printing/check with a JTAG to help us narrow down where you actually get stuck?

Belal-Elshinnawey, 2022-07-01:
I tried what you said, I tried with the same devboard that you have also, I get the same behavior.

ESP-Marius, 2022-07-01:
How are you connecting to the board? Is it over the USB marked UART or the one marked USB?

Belal-Elshinnawey, 2022-07-01:
UART, the USB interface doesn't work, 
Also I have additional outputs on my my Devkit when I trigger the interrupt for the first time to get the example to work:

Backtrace:0x40376c37:0x3fc924100x40376aca:0x3fc92430 0x40378ae3:0x3fcf4630 0x42001f02:0x3fcf4650 0x4037b749:0x3fcf4670 0x4037cb65:0x3fcf4690
0x40376c37: systimer_ll_clear_alarm_int at C:/Users/belal/esp/esp-idf/components/hal/esp32s3/include/hal/systimer_ll.h:165
 (inlined by)  at C:/Users/belal/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_systick.c:130

0x40376aca: _xt_lowint1 at C:/Users/belal/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1117

0x40378ae3: xt_utils_wait_for_intr at C:/Users/belal/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/belal/esp/esp-idf/components/esp_hw_support/cpu.c:100

0x42001f02: cpu_ll_waiti at C:/Users/belal/esp/esp-idf/components/esp_hw_support/include/hal/cpu_ll.h:111
 (inlined by) esp_vApplicationIdleHook at C:/Users/belal/esp/esp-idf/components/esp_system/freertos_hooks.c:58

0x4037b749: prvIdleTask at C:/Users/belal/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3969 (discriminator 1)
0x4037cb65: vPortTaskWrapper at C:/Users/belal/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

/////Device reboots and starts working normally.////////

ESP-Marius, 2022-07-04:
Hmm, still not able to reproduce this is any way :thinking: 

A few suggestions:
 1. Can you try to use e.g. `esp_rom_printf()` to figure out exactly where in `ulp_riscv_config_and_run` it gets stuck? (Just add a few prints and see which is the first one that never gets output)
 2. Does this still happen if you remove `ulp_riscv_wakeup_main_processor()` from the ULP-Riscv main.c?

Belal-Elshinnawey, 2022-07-24:
So, it turns out that I somehow managed to brick multiple boards. the ulp is working as expected, I ordered few extra boards and with brand new esp. I don't know what I did to destroy the old ones, but what i know is it wasn't hanging in the  ulp_riscv_config_and_run, it was hanging when the esp_deep_sleep_start is called. it would sleep until any condition is met, say the timer is reached, or if I'm using the interrupt example, when the pin is pressed, the esp would crash and restart a normal restart with a wakeup cause different then an ulp wakeup. new board, new esp fixed it, I hope I don't brick this one too. 

Cheers!

ESP-Marius, 2022-07-28:
@Belal-Elshinnawey Good to hear that it resolved itself somehow.

I'll close this issue for now then, but please feel free to re-open (or open a new one) if anything similar happens again.

closed 2022-07-28 by ESP-Marius