espressif/esp-idf #9261: Crash in esp_wifi_set_ant_gpio()

BaaBaaGoat, 2022-06-30:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-C3FH4 on dev board dev board https://wiki.luatos.com/chips/esp32c3/hardware.html
- IDF version (run ``git describe --tags`` to find it):v4.4
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):8.4.0
- Operating System:     Windows
- (Windows only) environment type:ESP Command Prompt.
- Using an IDE?: Yes (Espressif-IDE 2.5.0)
- Power Supply:         USB

## Problem Description
an **Store access fault** was thrown when calling esp_wifi_set_ant_gpio,
### Expected Behavior
antenna controlling GPIO is successfully configured.
### Actual Behavior
an **Guru Meditation Error: Core  0 panic'ed (Store access fault)** was thrown during call to API esp_wifi_set_ant_gpio()
### Steps to reproduce

1. create an IDF project by using example template wifi/scan
[image]
2. add this code snippet after esp_wifi_init(line 119 in scan.c):
```c
    gpio_reset_pin(18);
    gpio_set_direction(18, GPIO_MODE_OUTPUT);
    gpio_reset_pin(19);
    gpio_set_direction(19, GPIO_MODE_OUTPUT);
	ESP_ERROR_CHECK(gpio_set_level(18,0));
	ESP_ERROR_CHECK(gpio_set_level(19,0));
    wifi_ant_gpio_config_t gpiocfg = {{
        { .gpio_select = 1, .gpio_num = 18 },
        { .gpio_select = 1, .gpio_num = 19 }
    }};
    ESP_ERROR_CHECK(esp_wifi_set_ant_gpio(&gpiocfg));
```
[image]
3. build&flash&monitor.error occur.
[image]
// If possible, attach a picture of your setup/wiring here.
using dev board https://wiki.luatos.com/chips/esp32c3/hardware.html,powered by USB type c.
no external wiring/setup made.

### Code to reproduce this issue

```cpp
    gpio_reset_pin(18);
    gpio_set_direction(18, GPIO_MODE_OUTPUT);
    gpio_reset_pin(19);
    gpio_set_direction(19, GPIO_MODE_OUTPUT);
	ESP_ERROR_CHECK(gpio_set_level(18,0));
	ESP_ERROR_CHECK(gpio_set_level(19,0));
    wifi_ant_gpio_config_t gpiocfg = {{
        { .gpio_select = 1, .gpio_num = 18 },
        { .gpio_select = 1, .gpio_num = 19 }
    }};
    ESP_ERROR_CHECK(esp_wifi_set_ant_gpio(&gpiocfg));
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x4203bc00  RA      : 0x4203a2ca  SP      : 0x3fc9f060  GP      : 0x3fc8ec00
0x4203bc00: wifi_set_ant_gpio at ??:?

0x4203a2ca: ieee80211_ioctl_process at ??:?

TP      : 0x3fc8b97c  T0      : 0x420126fc  T1      : 0x420130a0  T2      : 0x42012270
0x420126fc: aes_decrypt_init at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:326

0x420130a0: ccmp_decrypt at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/ccmp.c:141

0x42012270: esp_aes_decrypt at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/crypto_ops.c:35

S0/FP   : 0x3fc994f4  S1      : 0xffffffff  A0      : 0x00000001  A1      : 0x00000004
A2      : 0x000000d8  A3      : 0x0ffd114c  A4      : 0x00000000  A5      : 0x3ff44578
A6      : 0x4201226c  A7      : 0x42013158  S2      : 0x3fc93dbc  S3      : 0x3ff1b5a8
0x4201226c: esp_aes_gmac at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/crypto_ops.c:47

0x42013158: ccmp_encrypt at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/ccmp.c:193

S4      : 0x3fce0000  S5      : 0x3fce0000  S6      : 0x3fce0000  S7      : 0x3fce0000
S8      : 0x3ff1b000  S9      : 0x3fce0000  S10     : 0x3fcdf934  S11     : 0x00000000
T3      : 0x42012b54  T4      : 0x4201279c  T5      : 0x42012732  T6      : 0x42012724
0x42012b54: omac1_aes_128 at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/aes-omac1.c:153

0x4201279c: aes_128_cbc_decrypt at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:363

0x42012732: aes_128_cbc_encrypt at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:341

0x42012724: aes_decrypt_deinit at C:/Users/exote/.espressif/frameworks/esp-idf-v4.4/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:336

MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x3ff44578
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000
```
[idfMonitorLogs.log](https://github.com/espressif/esp-idf/files/9016438/idfMonitorLogs.log)

## Other items if possible
- [[sdkconfig.zip](https://github.com/espressif/esp-idf/files/9016449/sdkconfig.zip) ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ [scan.zip](https://github.com/espressif/esp-idf/files/9016447/scan.zip)] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ included in idfMonitorLogs.log ] coredump (This provides stacks of tasks.)
- full project archive [AFETest.zip](https://github.com/espressif/esp-idf/files/9016451/AFETest.zip)

BaaBaaGoat, 2022-06-30:
https://github.com/MaxwellAlan/esp32_dual_ant_demo also tested,same error.
log:[idfMonitorLogs.log](https://github.com/espressif/esp-idf/files/9016699/idfMonitorLogs.log)
code(include ELF):[AFETest.zip](https://github.com/espressif/esp-idf/files/9016705/AFETest.zip)

nopnop2002, 2022-06-30:
```
    wifi_ant_gpio_config_t config;
    config.gpio_cfg[0].gpio_select = 1;
    config.gpio_cfg[0].gpio_num = 18;
    config.gpio_cfg[1].gpio_select = 1;
    config.gpio_cfg[1].gpio_num = 19;
    config.gpio_cfg[2].gpio_select = 0;
    config.gpio_cfg[3].gpio_select = 0;
    ESP_ERROR_CHECK(esp_wifi_set_ant_gpio(&config));
```

BaaBaaGoat, 2022-06-30:
> ```
>     wifi_ant_gpio_config_t config;
>     config.gpio_cfg[0].gpio_select = 1;
>     config.gpio_cfg[0].gpio_num = 18;
>     config.gpio_cfg[1].gpio_select = 1;
>     config.gpio_cfg[1].gpio_num = 19; [...]
also tried ,same error.
also tried 
```c
    wifi_ant_gpio_config_t config ;
    ESP_ERROR_CHECK(esp_wifi_get_ant_gpio(&config));
    config.gpio_cfg[0].gpio_select =1;
    config.gpio_cfg[0].gpio_num =18;
    ESP_ERROR_CHECK(esp_wifi_set_ant_gpio(&config));
```c
 ,same error.

```c
    wifi_ant_gpio_config_t config ;
    ESP_ERROR_CHECK(esp_wifi_get_ant_gpio(&config));
    ESP_ERROR_CHECK(esp_wifi_set_ant_gpio(&config));
```c
 gives no error(&no effect).

BaaBaaGoat, 2022-06-30:
thi bug seems just affect ESP32-C3.same code works on ESP32.

Alvin1Zhang, 2022-08-12:
Thanks for reporting, sorry for the slow turnaround. Fix on master branch is available at https://github.com/espressif/esp-idf/commit/0e874812a0ea97e8e10dcc48472ed2099032e1be, we are also back porting to other release branches. 

Will update later when the fixes are synchronised here. Thanks.

Alvin1Zhang, 2022-11-21:
Thanks for reporting, the fix has been back ported to other release branches, and fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/9269a536ac123677b165ff531369dfd1338ef103, feel free to reopen.

closed 2022-11-21 by Alvin1Zhang