espressif/esp-idf #9263: Separate constraints file download breaks Docker image builds

ftab, 2022-06-30:
## Environment

- Development Kit:      all of em
- Module or chip used:  all of em
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-1426-ge899edd793
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: no
- Power Supply:         Unlimited power

## Problem Description

The commit here https://github.com/espressif/esp-idf/commit/b28d7e6850936a9e5bbf542fa1d55adf6f9ae2f3 caused esp-idf to start downloading a constraints file from the internet on every install / export.

This eventually broke my Docker build container which is a custom build of esp-adf that works around some odd issues and contains a specific checkout of esp-idf at the aforementioned e899edd793. (the build fails on recent masters but at the time of e899edd793 a bug was fixed that hadn't been backported to 4.4 at the time I made the container, so it was deemed necessary to venture into unstable territory)

At the time I built the container, everything was working. At some point between then and now, the constraints file changed. (somewhere between June 6-9 by the looks of where our job logs started all failing)

Now, all builds fail because the new constraints file is not satisfied. Rebuilding the Docker container should work, but that means the Dockerfile is not reproducible. It depends on a specific point in time and not just a specific git version.

I see a comment in the commit:

    # We need to constrain package dependencies as well. So all installed packages need to be checked.
    # For example package A requires package B. We have only A in our requirements. But the newest version of B could
    # broke at some time and in that case we add a constraint for B (on the server) but don't have to update the
    # requirement file (in the ESP-IDF repo).

I appreciate the desire to not have to do a whole commit-release cycle to fix something like that. Ironically enough, the fact that this file changed on the server **is the only thing that actually broke something**, at least so far, for me. Granted, this is an unreleased, unstable branch as of yet. But even so, if it is discovered that a newer version of a sub-dependency package does break something, I think it's best for it to be committed in the repo so that the same git checkout will build the same build environment every time. Ideally, dependencies are locked in at some way. I am not sure what sort of blackbox dependency hell auto update torture prompted this change, but the cure feels worse than the disease. Unfortunately the alternative does mean, in the event it happens, new checkout/installs of old versions that lack the necessary upper limits on dependency versions will of course break.

### Expected Behavior

Reproducible builds

### Actual Behavior

Build tool downloads an external file which is not a part of the repo which can break things when it changes outside of version control

### Steps to reproduce

1. go back in time to before https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt required idf-component-manager~=1.1
2. make a Docker image for an esp-idf build container based on any checkout from b28d7e6850936a9e5bbf542fa1d55adf6f9ae2f3 or newer. i.e. https://github.com/espressif/esp-idf/blob/master/tools/docker/Dockerfile with a specific `IDF_CHECKOUT_REF` 
3. go back forward in time
4. try to build using the Docker image produced in step 2

## CI Job Log

```
...
Downloading espidf.constraints.v5.0.txt to /root/.espressif/espidf.constraints.v5.0.txt.tmp
Done
The following Python requirements are not satisfied:
idf-component-manager~=1.1
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /opt/esp/python_env/idf5.0_py3.8_env
    Python interpreter used: /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Constraint file: /root/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /opt/esp/adf/esp-idf/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Cleaning up project directory and file based variables
00:01
ERROR: Job failed: exit code 1
```

ftab, 2022-06-30:
Perhaps something like pip-tools can help. It bakes the requirements.txt file similar to how a Gemfile.lock works for ruby gems if I understand correctly

https://github.com/jazzband/pip-tools

Or just an option to opt into the newer constraints file

dobairoland, 2022-06-30:
First of all, we are sorry for the inconvenience. The master branch currently is the work-in-progress v5.0 which is a major release and will contain breaking changes in comparison with v4.4.x. You cannot except to get the same results for all commits between the top of the v4.4 and the master branch. Once v5.0 will be released, the constraint file will freeze and will be modified only to solve potential bugs.

Back to your problem. I can help to work around the issue. I assume that you want to keep using `install.sh` and `export.sh`. (At the lower level script there are options to avoid the constraint file.) For the use case with `install.sh` and `export.sh` you can use the following.

You can modify your `/root/.espressif/espidf.constraints.v5.0.txt` file with requirements which suits your needs (compatibility with older master commits).  After that, you can add `touch /root/.espressif/espidf.constraints.v5.0.txt` to the beginning of your docker file before calling `install.sh` or `export.sh`. That will avoid the download and overwite with a newer version from the server. 

I hope this helps.

ftab, 2022-06-30:
Thanks for your attention to this issue. Sorry to trouble you.

I know that 5.0 is not released yet, but when it is, the same problem will apply. Everybody building with containers will run into it (built themselves or downloaded from the espressif docker hub)--eventually the image they're using will become out of sync with the constraints file on the server and builds will fail.

I should mention I did expect to be traveling some bumpy roads going off of the stable branch, and I usually don't bother anyone about what I run into since it's in development, and that's how it goes. This is a little different than the usual bleeding-edge branch excitement, if it's how it's going to be done once it is stable.

> Once v5.0 will be released, the constraint file will freeze and will be modified only to solve potential bugs.

Will it freeze, or will it be modified to solve potential bugs? Those are two different things.
* If the constraints file for that version ever does change (even to solve potential bugs), I suspect the build will still fail, the installed versions will be different than the ones stated in the constraints file downloaded at the time of building the image
* If the file will not change once released, why not commit to version control along with the rest of it?

The constraints format seems prone to causing confusion. It certainly confused me when I ran into it this month. Maybe that's just the covid that's taking a while to wear off.

I still think if you could try [pip-tools](https://github.com/jazzband/pip-tools) it might be a better way. They solve this for you by pinning the versions. You don't automatically get new features or fixes from future dependency updates, but that's the only downside. The major tradeoff is at least things are much less likely to break outside of your control.

e.g. if I move requirements.core.txt to requirements.core.in and then `pip-compile requirements.core.in -o requirements.core.txt --generate-hashes --allow-unsafe` it generates the below requirements.core.txt, as of right now (latest master)

```
#
# This file is autogenerated by pip-compile with python 3.8
# To update, run:
#
#    pip-compile --allow-unsafe --generate-hashes --output-file=requirements.core.txt requirements.core.in
#
bitstring==3.1.9 \
    --hash=sha256:0de167daa6a00c9386255a7cac931b45e6e24e0ad7ea64f1f92a64ac23ad4578 \
    --hash=sha256:a5848a3f63111785224dca8bb4c0a75b62ecdef56a042c8d6be74b16f7e860e7 \
    --hash=sha256:e3e340e58900a948787a05e8c08772f1ccbe133f6f41fe3f0fa19a18a22bbf4f
    # via esptool
certifi==2022.6.15 \
    --hash=sha256:84c85a9078b11105f04f3036a9482ae10e4621616db313fe045dd24743a0820d \
    --hash=sha256:fe86415d55e84719d75f8b69414f6438ac3547d2078ab91b67e779ef69378412
    # via requests
cffi==1.15.1 \
    --hash=sha256:00a9ed42e88df81ffae7a8ab6d9356b371399b91dbdf0c3cb1e84c03a13aceb5 \
    --hash=sha256:03425bdae262c76aad70202debd780501fabeaca237cdfddc008987c0e0f59ef \
    --hash=sha256:04ed324bda3cda42b9b695d51bb7d54b680b9719cfab04227cdd1e04e5de3104 \
    --hash=sha256:0e2642fe3142e4cc4af0799748233ad6da94c62a8bec3a6648bf8ee68b1c7426 \
    --hash=sha256:173379135477dc8cac4bc58f45db08ab45d228b3363adb7af79436135d028405 \
    --hash=sha256:198caafb44239b60e252492445da556afafc7d1e3ab7a1fb3f0584ef6d742375 \
    --hash=sha256:1e74c6b51a9ed6589199c787bf5f9875612ca4a8a0785fb2d4a84429badaf22a \
    --hash=sha256:2012c72d854c2d03e45d06ae57f40d78e5770d252f195b93f581acf3ba44496e \
    --hash=sha256:21157295583fe8943475029ed5abdcf71eb3911894724e360acff1d61c1d54bc \
    --hash=sha256:2470043b93ff09bf8fb1d46d1cb756ce6132c54826661a32d4e4d132e1977adf \
    --hash=sha256:285d29981935eb726a4399badae8f0ffdff4f5050eaa6d0cfc3f64b857b77185 \
    --hash=sha256:30d78fbc8ebf9c92c9b7823ee18eb92f2e6ef79b45ac84db507f52fbe3ec4497 \
    --hash=sha256:320dab6e7cb2eacdf0e658569d2575c4dad258c0fcc794f46215e1e39f90f2c3 \
    --hash=sha256:33ab79603146aace82c2427da5ca6e58f2b3f2fb5da893ceac0c42218a40be35 \
    --hash=sha256:3548db281cd7d2561c9ad9984681c95f7b0e38881201e157833a2342c30d5e8c \
    --hash=sha256:3799aecf2e17cf585d977b780ce79ff0dc9b78d799fc694221ce814c2c19db83 \
    --hash=sha256:39d39875251ca8f612b6f33e6b1195af86d1b3e60086068be9cc053aa4376e21 \
    --hash=sha256:3b926aa83d1edb5aa5b427b4053dc420ec295a08e40911296b9eb1b6170f6cca \
    --hash=sha256:3bcde07039e586f91b45c88f8583ea7cf7a0770df3a1649627bf598332cb6984 \
    --hash=sha256:3d08afd128ddaa624a48cf2b859afef385b720bb4b43df214f85616922e6a5ac \
    --hash=sha256:3eb6971dcff08619f8d91607cfc726518b6fa2a9eba42856be181c6d0d9515fd \
    --hash=sha256:40f4774f5a9d4f5e344f31a32b5096977b5d48560c5592e2f3d2c4374bd543ee \
    --hash=sha256:4289fc34b2f5316fbb762d75362931e351941fa95fa18789191b33fc4cf9504a \
    --hash=sha256:470c103ae716238bbe698d67ad020e1db9d9dba34fa5a899b5e21577e6d52ed2 \
    --hash=sha256:4f2c9f67e9821cad2e5f480bc8d83b8742896f1242dba247911072d4fa94c192 \
    --hash=sha256:50a74364d85fd319352182ef59c5c790484a336f6db772c1a9231f1c3ed0cbd7 \
    --hash=sha256:54a2db7b78338edd780e7ef7f9f6c442500fb0d41a5a4ea24fff1c929d5af585 \
    --hash=sha256:5635bd9cb9731e6d4a1132a498dd34f764034a8ce60cef4f5319c0541159392f \
    --hash=sha256:59c0b02d0a6c384d453fece7566d1c7e6b7bae4fc5874ef2ef46d56776d61c9e \
    --hash=sha256:5d598b938678ebf3c67377cdd45e09d431369c3b1a5b331058c338e201f12b27 \
    --hash=sha256:5df2768244d19ab7f60546d0c7c63ce1581f7af8b5de3eb3004b9b6fc8a9f84b \
[... cut ...]
I would be interested to know more about the option to skip the constraints file. Is there any chance the option could be made available to export.sh? Perhaps that is what belongs in the Docker image now.

I assume it would be modifying this line in export.sh...

```diff
-python "${IDF_PATH}/tools/idf_tools.py" check-python-dependencies || return 1
+python "${IDF_PATH}/tools/idf_tools.py" check-python-dependencies --no-constraints || return 1
```

or some way that passes it along as a command line switch (does export.sh have any other switches? could just pass the args straight through if not, then `. path/to/export.sh --no-constraints` just works)

closed 2022-07-19 by espressif-bot