espressif/esp-idf #9265: ESP ModBus remote host limitation(s)

ammaree, 2022-06-30:
## Environment
- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3493-ga18a9c5011
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: macOS
- Using an IDE?: Eclipse
- Power Supply: external 5V

## Problem Description
We need to  connect to a number of PLC's, some located behind a firewall at a remote site. The remote PLC's are accessed using NAT on the firewall, mapped to different port numbers say X, X+10 and X+20. Using the current library API's this is not possible, since only a single port address can be specified.

### Expected Behaviour
Mechanism allowing multiple different host configurations including hostname/address, port number etc.

### Actual Behaviour
Connections fail

### Steps to reproduce

### Code to reproduce this issue

## Debug Logs

## Other items if possible

alisitsyn, 2024-01-24:
Hello @ammaree,

Please you take a look to the [v2.0.0-beta release here](https://github.com/espressif/esp-modbus/discussions/45).
This version of stack allows to define several instances of modbus master and slaves and allows to define [extended network options](https://github.com/espressif/esp-modbus/blob/release/v2.0/test_apps/adapter_tests/main/test_modbus_adapter_comm.c#L189) for slaves. This is described also in the [documentation for this version](https://docs.espressif.com/projects/esp-modbus/en/v2.0.0-beta/esp32/port_initialization.html#master-communication-options). 
It is also supported to send request for CID to slave with unique UID using the [mbc_master_set_parameter_with() API function](https://docs.espressif.com/projects/esp-modbus/en/v2.0.0-beta/esp32/master_api_overview.html#master-communication)
Please review this way of definition of slave IP addresses and let me know if this would work for you.

ammaree, 2024-02-14:
Thanks for the response but moved on to [liblightmodbus](https://github.com/Jacajack/liblightmodbus-esp) 18 months ago so not an issue any more.

closed 2024-02-14 by ammaree