espressif/esp-idf #9267: "esp_hidh_dev_get_report" doesn't work as expected

selimrecep, 2022-07-01:
## Environment

- Development Kit:        ESP32-DevKitC
- Kit version                   v4
- Module or chip used:  ESP32-WROOM-32
- IDF version                  4.4.1 (ESP_IDF_VERSION)
- Build System:              idf.py (on platform.io)
- Compiler version:       toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
- Operating System:      Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Platform IO
- Power Supply: USB

## Problem Description

`esp_hidh_dev_get_report` can not send hid report request to the device, it keeps `dev (esp_hidh_dev_t*)`'s transport state at `TRANS_GET_REPORT`, therefore neither can receive report or make another get report request (ends up with `Pending previous tansaction TRANS_GET_REPORT done, try later!`) [even with 3 seconds wait time between the requests.] Also `esp_hidh_dev_feature_get` seems to have the same behaviour.

### Expected Behavior
`esp_hidh_dev_get_report` should send request and report should be sent through the config's callback function
### Actual Behavior
Hangs on `TRANS_GET_REPORT`

### Steps to reproduce

1. Connect to the device using the `esp_hid_host` example
2. With additional code, try to read hid feature from the device
3. ESP32 can't read from the device.

### Wiring
No wiring, bare USB connection :)

### Code to reproduce this issue
The full code
 [GIST](https://gist.github.com/selimrecep/a9973fd8b0c81e6be069816fcd1e6ce6)
The part doesn't work is here, 
```cpp
void getMacAddress(esp_hidh_dev_t* dev) {
    ESP_LOGI(TAG, "Reading 0x09\n");
    size_t size = sizeof(uint8_t) * (20);
    ESP_LOGI(TAG, "result: %d", esp_hidh_dev_get_report(dev, 0, 0x09, ESP_HID_REPORT_TYPE_FEATURE, size));
    ESP_LOGI(TAG, "got the mac address info");

    vTaskDelay(3000 / portTICK_PERIOD_MS);
}
```

## Debug Logs

```
Report Map Length: 279
    VENDOR FEATURE REPORT, ID: 240, Length:  63
    VENDOR FEATURE REPORT, ID: 242, Length:  15
    VENDOR FEATURE REPORT, ID: 241, Length:  63
    VENDOR FEATURE REPORT, ID: 131, Length:  63
    VENDOR FEATURE REPORT, ID: 130, Length:   9
    VENDOR FEATURE REPORT, ID: 129, Length:  63
    VENDOR FEATURE REPORT, ID: 128, Length:  63
    VENDOR FEATURE REPORT, ID:  34, Length:  63
    VENDOR FEATURE REPORT, ID:  32, Length:  63
    VENDOR FEATURE REPORT, ID:   9, Length:  19
    VENDOR FEATURE REPORT, ID:   8, Length:  47
    VENDOR FEATURE REPORT, ID:   5, Length:  40
    VENDOR  OUTPUT REPORT, ID:  57, Length: 546
    VENDOR  OUTPUT REPORT, ID:  56, Length: 525
    VENDOR  OUTPUT REPORT, ID:  55, Length: 461
    VENDOR  OUTPUT REPORT, ID:  54, Length: 397
    VENDOR  OUTPUT REPORT, ID:  53, Length: 333
    VENDOR  OUTPUT REPORT, ID:  52, Length: 269
    VENDOR  OUTPUT REPORT, ID:  51, Length: 205
    VENDOR  OUTPUT REPORT, ID:  50, Length: 141
    VENDOR  OUTPUT REPORT, ID:  49, Length:  77
    VENDOR   INPUT REPORT, ID:  49, Length:  77
   GAMEPAD   INPUT REPORT, ID:   1, Length:   9
```
The reports are successfully reported, also some get reports are sent by device without host request, succesfully
```
 (16754) event: running post ESP_HIDH_EVENTS:2 with handler 0x4013284c and context 0x3ffd73f4 on loop 0x3ffd7214␛[0m
␛[0;32mI (16754) ESP_HIDH_DEMO: DEV_: 0x3ffbb684␛[0m
D (16764) event: running post ESP_HIDH_EVENTS:2 with handler 0x400d3820 and context 0x3ffd7080 on loop 0x3ffd7214␛[0m
␛[0;32mI (16784) ESP_HIDH_DEMO: 4c:b9:9b:7f:fa:0d INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:␛[0m
␛[0;32mI (16794) ESP_HIDH_DEMO: 7e 83 81 83 08 00 00 00 00 ␛[0m
␛[0;32mI (16794) ESP_HIDH_DEMO: DEV I: 0x3ffbb684␛[0m
D (16804) event: running post ESP_HIDH_EVENTS:2 with handler 0x4013284c and context 0x3ffd73f4 on loop 0x3ffd7214␛[0m
D (16814) event: running post ESP_HIDH_EVENTS:2 with handler 0x400d3820 and context 0x3ffd7080 on loop 0x3ffd7214␛[0m
␛[0;32mI (16824) ESP_HIDH_DEMO: 4c:b9:9b:7f:fa:0d INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:␛[0m
␛[0;32mI (16834) ESP_HIDH_DEMO: 7e 83 81 83 08 00 04 00 00 ␛[0m
```

but at the point of reading get hid reports
```
␛[0;32mI (19774) ESP_HIDH_DEMO: Reading 0x09
␛[0m
␛[0;32mI (19774) ESP_HIDH_DEMO: result: 0␛[0m
␛[0;32mI (19774) ESP_HIDH_DEMO: got the mac address info␛[0m
␛[0;32mI (20984) ESP_HIDH_DEMO: hi there ;)␛[0m
␛[0;32mI (25774) ESP_HIDH_DEMO: Reading 0x20
␛[0m
␛[0;31mE (25774) BT_HIDH: Pending previous tansaction TRANS_GET_REPORT done, try later!␛[0m
␛[0;32mI (25774) ESP_HIDH_DEMO: l: -1␛[0m
␛[0;32mI (25774) ESP_HIDH_DEMO: got the firmware info␛[0m
␛[0;32mI (25984) ESP_HIDH_DEMO: hi there ;)␛[0m
␛[0;32mI (28784) ESP_HIDH_DEMO: Reading 0x05
␛[0m
␛[0;31mE (28784) BT_HIDH: Pending previous tansaction TRANS_GET_REPORT done, try later!␛[0m
␛[0;32mI (28784) ESP_HIDH_DEMO: l: -1␛[0m
␛[0;32mI (28784) ESP_HIDH_DEMO: got the calibration info␛[0m
```
I receive the errors

## Other items if possible

- [x] [sdkconfig.txt](https://github.com/espressif/esp-idf/files/9024402/sdkconfig.txt)  [renamed to .txt to upload]
- [x] [firmware.txt](https://github.com/espressif/esp-idf/files/9024414/firmware.txt) [renamed to .txt to upload]
- [ ] coredump

## Additional
I don't think I am calling the API wrong way even I might be, but as a side feature request to this issue, an example of HID Host getting(via requesting/reading) and sending HID reports would be useful. (I can open another github issue for that if needed)

moefear85, 2022-07-01:
This is probably related to this [issue](https://github.com/espressif/esp-idf/issues/7606)

selimrecep, 2022-07-01:
I am not getting errors related to parsing errors though, but thanks for sharing.

boblane1, 2022-07-05:
Hi @selimrecep，

Does the remote HID device is also implemented by ESP32?

It seems that the HID Device doesn't response to the request of ‘GET_REPORT’ which is sent by the HID Host.

selimrecep, 2022-07-05:
Thanks for response @boblane1 ,
Yes, I tried that function as I mentioned it in the post. In fact, if the size is wrong the function logs error with the size being wrong anyway, plus I already got the sizes from the `esp_hidh_dev_dump(param->open.dev, stdout);`. It reports all reports neatly. I also tried `esp_hidh_dev_feature_get` too.

And no, the remote HID device isn't ESP32, actually it is a Bluetooth Classic compatible device that is a gamepad, has touchpad and gyro accelerometer sensors...

But! I have good news, after spending unneccessary full of few days of investigation, it is 95% about timing. The client device sends 0x1 data for a short time, then.. somehow... it stops reporting. So yesterday I tried replacing the `ESP_LOG` functions with raw `printf` functions that is printing only few characters (yes, won't even print the reports itself)... Apparently now I can send and receive reports from and to the device.

Controller is sending data so fast that ESP_LOG functions (or maybe the UART itself, I don't know, it is on 115200 baud) slows down micro controller causing the bluetooth of ESP32 not be able to keep up with the controller's speed (and it makes sense since controller has accelerometer and gyroscope type of sensors...)

I haven't *fully* confirmed it is about timing, I still have several connection issues, I might open another issue for that, but after confirming it I will probably close the issue in few days. I hope this will be helpful for other people, considering nothing on Google showed up when I was seeking for a solution :)

boblane1, 2022-07-05:
I am glad to you have did these investigation. 

>Edit: Just for the sake of confirming it, if there is a type of function that I could use to confirm the ESP32 can't receive reports fast enough, I would like to try it.

For now,  IDF doesn't have a such function. But in most classical Bluetooth HID Device, the report latency is 10 ms.

moefear85, 2022-07-05:
if uart speed is an issue, it doesn't make a difference if you use esp_log or printf. So if esp_log isn't working, it is probably because you haven't set the log levels correctly.  both issues can be configured in sdkconfig.

for uart:
[esp system settings]->[Channel for Console Output] :: change it from "Default: UART0" to "Custom UART". Then a submenu is available and you can set "UART console baud rate" to any value. If your usb-uart adapter supports it, and the wires aren't noisy, you can safely go up to 2073600 (and even more). that's a 20x speed increase.

for missing logs:
make sure [Log Output]->[Maximum Log Verbosity] is set to "Verbose". Also, whatever tag you choose for  the calls to ESP_LOGx(...), first you need to set the log level for that specific tag before making any calls. Hence at the start of app_main(), add a line:
> esp_log_level_set(TAG,ESP_LOG_VERBOSE)
where TAG is whatever tag you are using.

selimrecep, 2022-07-05:
Hi,
I don't know if UART is the speed issue, all I know is replacing huge ESP LOG data functions with few bytes printing printf functions work, but I will investigate more on this so thank you for pointing out. 

And about missing logs, I don't have missing logs actually(unless you are talking about the timing issue), I also tried verbose too, didn't git anything useful.

I will also try the UART speed, I am sure it will be very useful for debugging process too, so thanks for that too.

selimrecep, 2022-07-09:
Addition: Not directly related to the issue here but since the issue gets affected from this, I experienced connection issues with the gamepad, for some reason enabling "Disable BT debug logs" on menu config made it much reliable. It could still be about timing considering gamepad reports about 800 Hertz, unless I got something wrong, I will close the issue after a further investigation

selimrecep, 2022-07-17:
Closing for now, no issues so far

closed 2022-07-17 by selimrecep