espressif/esp-idf #9268: 连接上ap，但无法得到ip地址

JustQY, 2022-07-01:
这个问题出现很久了，几乎每个版本都有，我这边当前用的idf4.4
如图看日志
[image]

有没有办法不用重启解决此问题，我试了关闭dhcp再启动也没用

MaxwellAlan, 2022-07-13:
Hi @JustQY
根节点的 RSSI 值 为何是-87dBm 这么低？
Thanks.

JustQY, 2022-07-14:
> 为何是-87dBm

我们这边是ac+ap模式，有可能连接到稍远的ap那儿去了。

也就是说如果信号差了就会出现这个情况，是吧

shenjun7, 2022-07-14:
信号好的情况下有获取不到ip的情况吗？网络通过`esp_mesh_set_type(MESH_ROOT)`设置了FIX-ROOT 固定根节点吗？
如果是自组织网络 root rssi 值很差可以调用`esp_mesh_waive_root()`重新选择信号更好的根节点．

JustQY, 2022-07-27:
> 信号好的情况下有获取不到ip的情况吗？网络通过`esp_mesh_set_type(MESH_ROOT)`设置了FIX-ROOT 固定根节点吗？ 如果是自组织网络 root rssi 值很差可以调用`esp_mesh_waive_root()`重新选择信号更好的根节点．

信号好的时候没注意，实际场景用时没出现这个情况，就在公司环境调试时遇到，不过如果家庭也是多ap的情况可能也会出现。
mesh方面没有做fix root，我现在做了超时2分钟重启，即通知说连上ap了，但2分钟后没得到ip就重启

JustQY, 2022-07-27:
好了，原因知道了，可以关闭了

Alvin1Zhang, 2022-07-27:
感谢反馈。

closed 2022-07-27 by Alvin1Zhang