espressif/esp-idf #9272: Boot loader problem

LuizFernandoDeveloper, 2022-07-02:
> Executing task: ninja  <

[11/619] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map
cmd.exe /C "cd /D C:\engenharia_de_software\projetos_embarcados\blink\build\bootloader && C:\Espressif\tools\cmake\3.20.3\bin\cmake.exe --build ."
[1/26] Building C object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_crc.c.obj
[2/26] Building C object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_tjpgd.c.obj
[3/26] Building ASM object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_longjmp.S.obj
[4/26] Building C object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_sys.c.obj
[5/26] Building C object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_uart.c.obj
[6/26] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj
[7/26] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log.c.obj
[8/26] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_mem.c.obj
FAILED: esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_mem.c.obj
C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe  -Iconfig -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader_support/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader_support/include_bootloader -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32/private_include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/efuse/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader/subproject/components/micro-ecc -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader/subproject/components/micro-ecc/micro-ecc -IC:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/include -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader/subproject=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v4.4.1=IDF -fstrict-volatile-bitfields 
-Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.1\" -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_mem.c.obj -MF esp-idf\bootloader_support\CMakeFiles\__idf_bootloader_support.dir\src\bootloader_mem.c.obj.d -o esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_mem.c.obj -c C:/Espressif/frameworks/esp-idf-v4.4.1/components/bootloader_support/src/bootloader_mem.c
C:/Espressif/frameworks/esp-idf-v4.4.1/components/hal/esp32/include/hal/cpu_ll.h: Assembler messages:
C:/Espressif/frameworks/esp-idf-v4.4.1/components/hal/esp32/include/hal/cpu_ll.h:65: Error: invalid register number (97) for 'wsr' instruction
[9/26] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_buffers.c.obj
[10/26] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/bootloader_start.c.obj
ninja: build stopped: subcommand failed.
[20/619] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/netif/slipif.c.obj
ninja: build stopped: subcommand failed.
O processo de terminal "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " foi terminado com o código de saída: 1.

LuizFernandoDeveloper, 2022-07-02:
Já  tentei de tudo !

Alvin1Zhang, 2022-07-04:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ESP-Marius, 2022-07-04:
@LuizFernandoDeveloper It seems like something is getting confused in your build. It's building with the ESP32-S2 toolchain (`xtensa-esp32s2-elf-gcc.exe `), but the files getting pulled in are for ESP32 (`esp32/include/hal/cpu_ll.h`).

Not sure how you would end up in such a situation though. If you try a clean build (build folder and sdkconfig deleted) on a unmodified IDF example do you still have problems?

If yes, then please attach all the output logged while building.

ESP-Marius, 2022-07-19:
Closing this issue due to lack of response.

Feel free to reopen the issue if you are still having problems.

closed 2022-07-19 by ESP-Marius